JuiceRescue/juicepassproxy #1: socket.gaierror

Juggler00, 2023-11-05:
Thank you for publishing this! I've "dockerized" this with the following Dockerfile
```
FROM python:latest

RUN git clone https://github.com/snicker/juicepassproxy

RUN pip install -r juicepassproxy/requirements.txt

CMD [ "python", "juicepassproxy/juicepassproxy.py", "-s 127.0.0.1:8051", "-d 54.161.185.130:8051", "-p 1883", "-H 192.168.2.10"]
```
When I try to bring the container up, I see the following error output from python:
```
juicebox            | Traceback (most recent call last):
juicebox            |   File "//juicepassproxy/juicepassproxy.py", line 205, in <module>
juicebox            |     main()
juicebox            |   File "//juicepassproxy/juicepassproxy.py", line 196, in main
juicebox            |     handler = JuiceboxMessageHandler(mqtt_settings=mqttsettings, 
juicebox            |               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
juicebox            |   File "//juicepassproxy/juicepassproxy.py", line 42, in __init__
juicebox            |     self._init_devices()
juicebox            |   File "//juicepassproxy/juicepassproxy.py", line 48, in _init_devices
juicebox            |     self._init_device_status(device_info)
juicebox            |   File "//juicepassproxy/juicepassproxy.py", line 61, in _init_device_status
juicebox            |     sensor = Sensor(settings)
juicebox            |              ^^^^^^^^^^^^^^^^
juicebox            |   File "/usr/local/lib/python3.12/site-packages/ha_mqtt_discoverable/__init__.py", line 652, in __init__
juicebox            |     self._connect_client()
juicebox            |   File "/usr/local/lib/python3.12/site-packages/ha_mqtt_discoverable/__init__.py", line 707, in _connect_client
juicebox            |     result = self.mqtt_client.connect(
juicebox            |              ^^^^^^^^^^^^^^^^^^^^^^^^^
juicebox            |   File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 914, in connect
juicebox            |     return self.reconnect()
juicebox            |            ^^^^^^^^^^^^^^^^
juicebox            |   File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 1044, in reconnect
juicebox            |     sock = self._create_socket_connection()
juicebox            |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
juicebox            |   File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
juicebox            |     return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
juicebox            |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
juicebox            |   File "/usr/local/lib/python3.12/socket.py", line 828, in create_connection
juicebox            |     for res in getaddrinfo(host, port, 0, SOCK_STREAM):
juicebox            |                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
juicebox            |   File "/usr/local/lib/python3.12/socket.py", line 963, in getaddrinfo
juicebox            |     for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
juicebox            |                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
juicebox            | socket.gaierror: [Errno -2] Name or service not known
```
Wondering if you have any thoughts on this.

Note that when I telnet into my Juicebox, I get the following for a list command:
```
list
! # Type  Info
# 0 FILE  webapp/index.html-1.4.0.24 (1995, 0)
# 1 UDPC  juicenet-udp-prod4-usa.enelx.com:8051 (12684)
```

snicker, 2023-11-05:
based on the stack track, the MQTT client library is failing to locate your MQTT server. This seems like an issue with how your network is configured in the docker environment or an incorrect IP for your MQTT server.

snicker, 2023-11-07:
hey @Juggler00  did you ever figure this out?

closed 2023-11-09 by Juggler00