JuiceRescue/juicepassproxy #3: Can't get MQTT working

Snuffy2, 2023-11-05:
I have it running in a Docker (see #2) and it appears to be receiving and passing along the UDP messages to the EnelX Server.

However, I don't see it sending anything via MQTT and see the error below when Debug is enabled:
```
juicepassproxy  | 2023-11-05 11:19:17,301 DEBUG      Starting UDP proxy...
juicepassproxy  | 2023-11-05 11:19:17,302 DEBUG      Src: 172.16.100.16:8047
juicepassproxy  | 2023-11-05 11:19:17,302 DEBUG      Dst: 54.161.147.91:8047
juicepassproxy  | 2023-11-05 11:19:17,303 DEBUG      Looping proxy (press Ctrl-Break to stop)...
juicepassproxy  | 2023-11-05 11:19:19,460 DEBUG      local : b'0910042001280643829222621623:v09u,s010,F10,u00053456,V2480,L00001786832,S01,T18,M0040,C0040,m0040,t29,i52,e-0001,f5999,r51,b000,B0000000!DLD:'
juicepassproxy  | 2023-11-05 11:19:19,460 DEBUG      failed to process basic message: list index out of range
juicepassproxy  | 2023-11-05 11:19:21,691 DEBUG      remote: b'CMD01119A0040M040C208S012!JHZ$'
```
I've confirmed my MQTT settings are correct (I use MQTT in HA for a number of other things) and I can ping HA from within the Docker container. Any ideas?

snicker, 2023-11-05:
the message coming from the Juicebox seems to be different than mine (or others from that previous thread), what model of Juicebox are you using?

Particularly, the issue seems to be that the mains frequency is the 15th element in your device and in mine it is the 13th. To fix this we may need to actually parse the message to support different devices instead of assuming the order in the CSV from the device.

Snuffy2, 2023-11-05:
I'm in the US and it is a JuiceBox 40 (NEMA 14-50P) Plug In. Model: 2JBO401RNA

Snuffy2, 2023-11-05:
A bit longer of a log if it helps. Happy to provide anything else to assist.
[juicepassproxy.log](https://github.com/snicker/juicepassproxy/files/13260258/juicepassproxy.log)

PR #4 (2023-11-07): Support fields in different order
This is now making MQTT work for me.

Fixes #3
closed 2023-11-07 by snicker