JuiceRescue/juicepassproxy #7: add local DNS server to reduce dependencies on pihole/adguard/nice routers

snicker, 2023-11-07:
problem:
currently juicepassproxy requires running your own configurable dns server on your network (pihole, adguard, or a router running a dns server with the ability to configure static hosts). juicepassproxy could run its own dns server to remove that need.

potential implementation:
[ ] use something like `dnsrewriteproxy` : https://pypi.org/project/dnsrewriteproxy/
[ ] given low level nature of the UDP proxy, probably just run each server in separate threads and join upon termination

snicker, 2023-11-07:
@Snuffy2 you may be interested in looking at `dnsrewriteproxy` as something to spin up as a separate process in the docker container to remove dependencies on external local DNS servers

Snuffy2, 2023-11-07:
If this works:
```
set wlan.static.dns
save
```
Then, yes, it would make a lot of sense to stand up a DNS in Docker for this. I think it could easily be added to the same docker container and not even require 2 separate containers. Then, it would really only require the JuiceBox IP in the config and I could run Telnet scripts to get the EnelX Server, the JuiceBox ID, and set the JuiceBox DNS to point to the Docker. I'll try it out and see how far I can get.

Snuffy2, 2023-11-08:
I stood up a separate DNS server on my network and then set `wlan.static.dns` to that local DNS server. I turned off all of my router settings to force all DNS requests to my router. However, looking at my router logs, the JuiceBox still seems to be going to my router for the DNS despite setting `wlan.static.dns`.

My guess is that the JuiceBox (like most other devices) uses the DNS server sent as part of DHCP and only uses the one set in `wlan.static.dns` if one isn't sent in DHCP. I wonder if @jon6fingrs isn't either isn't sending a DNS server or not running DHCP on the network where the JuiceBox is since he notes it is working for him but also that the device is blocked from the internet.

[image]

[image]

snicker, 2023-11-08:
it is possible that wlan.static.dns on the device is only used if all other static IP settings are defined. i am using a static IP for my Juicebox and have the DNS set directly to the machine running pihole

snicker, 2023-11-08:
expanding further that most other devices on my network are getting dhcp leases from my router, which is configured to use my pihole as the primary dns server and 1.1.1.1 as secondary (in case the device running pihole is down). pihole logs show direct requests from the juicebox and all other requests from my gateway. so i know it is working.

Snuffy2, 2023-11-08:
@snicker What other settings on the JuiceBox did you set via Telnet in addition to `wlan.static.dns`?

michaelwoods, 2023-11-08:
```shell
nlo juicenet-udp-prod3-usa.enelx.com
```
This network lookup command (found from `help all`) seems to resolve the DNS. It was responding with the public EnelX IP until I rebooted after fixing the network's DHCP DNS info and now resolves properly to my internal IP.

I don't think using the static setting works on its own.

devzwf, 2023-11-08:
can an IP be set to the juicebox via Telnet ?
because i think the DHCP DNS settings superseed the wlan.static.dns

Snuffy2, 2023-11-08:
I tried setting:
```
wlan.static.dns: 10.100.200.50
wlan.static.gateway: 10.100.50.1
wlan.static.ip: 10.100.50.30
wlan.static.netmask: 255.255.255.0
```

Then `save` then `rboot` and confirmed the values persisted on reboot. However, it is still also getting the DHCP address details. I tried setting `wlan.dhcp.enabled` to 0 and `save`. But it seems to reset to 1 every time I reboot.

Is there another setting in Telnet I need to set to force it to use the static details and not the DHCP details?

```
> get wlan
get wlan
wlan.dhcp.cache_enabled: 0
wlan.dhcp.enabled: 1
wlan.dhcp.hostname: gecko_os-#
wlan.dhcp.timeout: 15
wlan.hide_passkey: 0
wlan.info: state: up
wlan.network.dns: 10.100.50.1
wlan.network.gateway: 10.100.50.1
wlan.network.ip: 10.100.50.30
wlan.network.netmask: 255.255.255.0
wlan.network.status: 2
wlan.network.status_gpio: -1
wlan.static.dns: 10.100.200.50
wlan.static.gateway: 10.100.50.1
wlan.static.ip: 10.100.50.30
wlan.static.netmask: 255.255.255.0
[...]
```

devzwf, 2023-11-08:
is it still use the dhcp dns settings ? because that's the part we want to change 

Of course the easy way would be to have all you LAN using the adguard / pihole / whatever internal dns you have ....
but i have the feeling many people dont have that

Snuffy2, 2023-11-08:
Yes, sorry it is still using the DHCP DNS settings. 

Agreed, I have it working when I override the DNS from my router, but we were trying to do it without requiring that to make it easier for users.

Snuffy2, 2023-11-09:
Ok, I may have another solution that seems to be working for me!! 

It doesn't require router changes or standing up a DNS server. It doesn't persist on JuiceBox reboot, but I could script it to make these changes in Telnet on Docker start:

1. Start juicepassproxy via Docker
2. `telnet <juicebox_ip> 2000`
3. `list` and note which # is UDPC (probably # 2)
4. `stream_close 2` or whatever # was above
5. `udpc <juicepassproxy_ip> 8047` or whatever port you have set in the docker

Watch the Docker logs as it now receives UDP packets. Please try this and if it works for others too, I'll work to automate it in the Docker start.

Snuffy2, 2023-11-09:
Ok, I spoke way too soon. It works but it seems to revert back to the juicebox url after 1-2 min. Sorry

devzwf, 2023-11-09:
yeah probably as soon it make a dns call :)

devzwf, 2023-11-09:
well 
according to the doc :
`wlan.dhcp.enabled
Abbreviation

wl d e

Access

get/set

Description

Select whether DHCP is used to automatically obtain an IPv4 IP address on the wlan interface.

See [Networking and Security, DHCP Client](https://docs.silabs.com/gecko-os/4/standard/latest/networking-and-security#dhcp-client).

Arguments

1|true|on/0|false|off

where

0 or false or off: disabled
1 or true or on: enabled
Default

1

Platforms

WGM160P
Properties

network_restart to apply change
available in Safe Mode]
preserved by NVM Backup`

so we should be able to disable dhcp 
did you network restart when you tried ?

devzwf, 2023-11-09:
i dont have Juicebox , trying to help a friend

snicker, 2023-11-09:
looking back at my settings and i am fairly sure that even though i configured my juicebox with a static IP and DNS it has reverted to DHCP. might be a weird bug

snicker, 2023-11-09:
```
wlan.dhcp.cache_enabled: 0
wlan.dhcp.enabled: 1
wlan.dhcp.hostname: ZentriOS-#
wlan.dhcp.timeout: 15
(...)
wlan.network.dns: 192.168.1.1
wlan.network.gateway: 192.168.1.1
wlan.network.ip: 192.168.1.39
wlan.network.netmask: 255.255.255.0
(...)
wlan.static.dns: 192.168.1.183
wlan.static.gateway: 192.168.1.1
wlan.static.ip: 192.168.1.39
wlan.static.netmask: 255.255.255.0
```

MrDrew514, 2023-11-13:
Any news on this? If it's not possible to enable a static ip directly on the machine due to a bug, would it be possible to make it run a telnet command every x minute like udpc <juicepassproxy_ip> 8047 to update the juicebox sensors on home assistant?

Snuffy2, 2023-11-13:
It would have to update the udpc line every 30 sec if not more frequently. Not sure how realistic that would be and any negative repercussions to the os, container, network, etc

Snuffy2, 2023-11-14:
I'm willing to put something together to give it a try. I'll update soon.

snicker, 2023-11-14:
@Snuffy2 I'll also look into this- I am thinking it might be better suited to include this in the python application

MrDrew514, 2023-11-14:
I don't know if a config change would be more ressource hogging than a dns request itself. From what I can see in the logs, a 2 minutes delay could be enough between the changes since it seems to revert back after that delay. 

Thanks for taking this temporary workaround into consideration.

Snuffy2, 2023-11-14:
I've put a draft PR #19 together. I'll need to test it out a lot more but so far it seems to stay running, stay connected to telnet and reconnect if it disconnects, check the list every 10 seconds and update it to the IP address when it reverts to the EnelX Server.

Snuffy2, 2023-11-14:
> @Snuffy2 I'll also look into this- I am thinking it might be better suited to include this in the python application

I had initially tried doing it in python but `telnetlib` is deprecated in python 3.11+ and I didn't like any of the alternatives. I've had more luck with a telnet expect script so far. If you can get a python one working though I'm sure that would be more robust.

snicker, 2023-11-14:
well damn, I did not know about telnetlib being deprecated in 3.11+. but #20 is probably my preferred approach

devzwf, 2023-11-14:
did any of you tried a network_restart after changing the IP to static and move dhcp.enable to 0 ?
since according to the doc this is the only way to apply the change to  : wlan.dhcp.enabled
"network_restart to apply change" 

also i am going to integrate a dns rewiter in the image ....
i found this https://github.com/jc21/dnsrouter who seems to fill all the tick so far 

let me know

snicker, 2023-11-14:
@MrDrew514 and @devzwf I merged #20 - Please check the read me and test out this feature (not yet set up for the docker container)

you'll need to set the `update_udpc` and `juicebox_host` command line flags to use it

snicker, 2023-11-14:
in my testing, I might be missing a message or two with the interval set at 30s. we could tighten this up.

devzwf, 2023-11-14:
the interval could be a setting of some sort with some minimum / max to avoid abuse... 
@MrDrew514 let me know if you have issue with the test
i cannnot test this part sadly since i do not have the box 

i will let @Snuffy2 fix the entrypoint  script accordingly should be strait forward

MrDrew514, 2023-11-14:
> command line flags

I'll gladly test it once It's ready for docker

Snuffy2, 2023-11-14:
> did any of you tried a network_restart after changing the IP to static and move dhcp.enable to 0 ? since according to the doc this is the only way to apply the change to : wlan.dhcp.enabled [...]

Yes and it didn't work/stick for me. FWIW, I have a branch with a working DNS Rewriter using [async-dns](https://pypi.org/project/async-dns/): https://github.com/Snuffy2/juicepassproxy/tree/Run-local-DNS-and-point-to-it-using-Telnet But without a way to override the DNS on the JuiceBox it's not of use right now.

I'll work on updating the entrypoint script, likely tonight.

Snuffy2, 2023-11-14:
@snicker, I believe we need to add another optional argument to JPP when Update UDPC is True. We need to capture the IP address of the machine that JPP is running on. When running in Docker (bridge mode), this is different from the SRC IP. The JPP Host IP is the external/intranet address (ex. 192.160.0.100) [the address you set the UDPC line to] whereas the SRC IP is the Docker address (ex. 172.16.100.10) [the address pyproxy listens to for packets]. When not running in Docker (or similar), those are likely to be the same address.

Maybe add an optional `--jpp_host` argument:
* If `--update_udpc` is true and `--jpp_host` is defined, use `--jpp_host` for UDPC telnet.
* Else If `--update_udpc` is true and `--jpp_host` is not defined, use `--src` for UDPC telnet.

strawless, 2023-11-15:
> in my testing, I might be missing a message or two with the interval set at 30s. we could tighten this up.

I'm trying to catch up to this thread.  Are you able to avoid custom router DNS or iptables entries if you send a telnet command constantly to the juicebox?  I'm glad that others were not able to use the static.dns, I feel like I spent a hour trying to make it work last week after the homeassistant thread everyone kept saying it worked.

snicker, 2023-11-15:
@Snuffy2  added `--juicebox_proxy_host` in #27

snicker, 2023-11-15:
> I'm trying to catch up to this thread. Are you able to avoid custom router DNS or iptables entries if you send a telnet command constantly to the juicebox? I'm glad that others were not able to use [...]

@strawless  - yes, the updates from last night will allow you to use this without setting up a DNS server to redirect requests. check the readme, and use the `--update_udpc`, `--juicebox_host`, and optionally the `--juicepass_proxy_host` arguments to activate the updater that sends the telnet commands to the juicebox every 30s. It also only updates if required.

snicker, 2023-11-25:
with the v0.2.0 release, I am closing this since we have implemented this functionality to remove the dependency on a local DNS rerouting in the `update_udpc` functionality!

closed 2023-11-25 by snicker