JuiceRescue/juicepassproxy #8: Can you point me to the right direction? ERROR getting EnelX Server from Telnet.

MrDrew514, 2023-11-08:
First, thanks for the time and effort put in this project.

juicebox machine ip address is 192.168.2.18
juicepassproxy is running with all the correct mqtt infos
adguard instance is running on 192.168.2.5

I've been using putty to login the juicebox machine,
list command is showing udp-prod3-usa.enelx.com (nslookup showing 54.161.185.130)
I did set wlan.static.dns 192.168.2.5 (which is adguard ip)
save

AdGuard rewrite
[image]

Container logs
[image]

Am I doing something wrong? I'd really appreciate any help. Thank you.

devzwf, 2023-11-08:
use the FQDN or *.enelx.com  in adguard and not the IP

Snuffy2, 2023-11-08:
1. It is likely not working due to your Adguard setting (as noted by @devzwf). From your Adguard screenshot, change Domain from `54.161.185.130` to `juicenet-udp-prod3-usa.enelx.com`.

2.  As for the Telnet Error in the Docker log: It looks like it is falling back to the defaults and for now those should work ok, however I'd like to understand why it is Erroring to try to improve it.
    a. Can you share your Docker Compose
    b. Can you get to a bash prompt in the Docker Container? `docker compose exec juicepassproxy bash` If so:
    c. Can you connect to the JuiceBox by typing `telnet <juicebox_ip>`?
    d. What do you see if you enter `/juicepassproxy/telnet_get_server.expect <juicebox_ip>`?

devzwf, 2023-11-09:
the error was due to my dumbness ... ignore it , when i build the the image for unraid i forgot to include `telnet_get_server.expect`
it is now fixed

closed 2023-11-09 by MrDrew514