JuiceRescue/juicepassproxy #9: what is SRC

Juggler00, 2023-11-09:
Apologies if this is obvious, but I'm trying to understand what the SRC is. Is it the IP address of my docker host, or the IP assigned to the container? If it is the later, this address is not routable. If it is the former, I seem to be getting an error (see below). Appreciate any guidance.

Docker host IP: 192.168.2.10
MQTT server (also container): 192.168.2.10:1883

```
  juicepassproxy:
    build: ./juicepassproxy
    hostname: juicepassproxy
    container_name: juicepassproxy
    restart: unless-stopped
    logging:
      driver: json-file
    ports:
      - 8051:8051/udp
    networks:
      public:
    environment:
      - JUICEBOX_LOCAL_IP=192.168.3.33
      - SRC=192.168.2.10
      - MQTT_HOST=192.168.2.10
      - DEBUG=true
    volumes:
      - /etc/localtime:/etc/localtime:ro 
```

From debug:
```
juicepassproxy      | 11/08/23 19:25:34: COMMAND: python /juicepassproxy/juicepassproxy.py --src 192.168.2.10:8051 --dst 54.161.147.91:8051 --host 192.168.2.10 --port 1883 --discovery-prefix homeassistant --name JuiceBox --debug
...
juicepassproxy      | 2023-11-08 19:25:35,037 DEBUG      Starting UDP proxy...
juicepassproxy      | 2023-11-08 19:25:35,037 DEBUG      Src: 192.168.2.10:8051
juicepassproxy      | 2023-11-08 19:25:35,037 DEBUG      Dst: 54.161.147.91:8051
juicepassproxy      | Traceback (most recent call last):
juicepassproxy      |   File "/juicepassproxy/juicepassproxy.py", line 221, in <module>
juicepassproxy      |     main()
juicepassproxy      |   File "/juicepassproxy/juicepassproxy.py", line 218, in main
juicepassproxy      |     pyproxy.udp_proxy(args.src, args.dst)
juicepassproxy      |   File "/juicepassproxy/pyproxy/pyproxy.py", line 43, in udp_proxy
juicepassproxy      |     proxy_socket.bind(ip_to_tuple(src))
juicepassproxy      | OSError: [Errno 99] Cannot assign requested address
```

devzwf, 2023-11-09:
any reason you are using 8051  as port ?

snicker, 2023-11-09:
SRC should be the IP and port of the host machine

closed 2023-11-09 by snicker

Juggler00, 2023-11-09:
I used port 8051 as this is what was listed:
```
telnet 192.168.3.33 2000
Trying 192.168.3.33...
Connected to 192.168.3.33.
Escape character is '^]'.
[2023-11-08 | 19:06:17: Ready]
> list
list
! # Type  Info
# 0 FILE  webapp/index.html-1.4.0.24 (1995, 0)
# 1 UDPC  juicenet-udp-prod4-usa.enelx.com:8051 (19405)
```
Funny enough, overnight we had a power outage. Now when I log into the Jukebox:
```
telnet 192.168.3.33 2000
Trying 192.168.3.33...
Connected to 192.168.3.33.
Escape character is '^]'.
[2023-11-09 | 12:05:26: Ready]
> list
list
! # Type  Info
# 0 FILE  webapp/index.html-1.4.0.24 (1995, 0)
# 1 UDPC  jbv1.emotorwerks.com:8042 (1952)
```
It appears that it has chosen a different server and port!

devzwf, 2023-11-09:
That's interesting.
and make it more difficult to spoof the DNS....
the FW rules might be the best option filtering with the src IP for base the rule of.

devzwf, 2023-11-09:
jbv1.emotorwerks.com
- jbv1.emotorwerks.com is an alias for switcher-us-prod-cs.cloudapp.net.
- switcher-us-prod-cs.cloudapp.net has address 138.91.137.23

juicenet-udp-prod4-usa.enelx.com
- juicenet-udp-prod4-usa.enelx.com is an alias for juicenet-udp-prod4-usa.awspub.enel.com.
- juicenet-udp-prod4-usa.awspub.enel.com has address 54.161.185.130

juicenet-udp-prod3-usa.enelx.com
- juicenet-udp-prod3-usa.enelx.com is an alias for juicenet-udp-prod3-usa.awspub.enel.com.
- juicenet-udp-prod3-usa.awspub.enel.com has address 54.161.185.130

juicenet-udp-prod2-usa.enelx.com
- juicenet-udp-prod2-usa.enelx.com is an alias for juicenet-udp-prod2-usa.awspub.enel.com.
- juicenet-udp-prod2-usa.awspub.enel.com has address 54.161.147.91

Juggler00, 2023-11-09:
A couple of notes. I have added 5x DNS rewrite rules to my AdGuard Home:
jbv1.emotorwerks.com
juicenet-udp-prod1-usa.enelx.com
juicenet-udp-prod2-usa.enelx.com
juicenet-udp-prod3-usa.enelx.com
juicenet-udp-prod4-usa.enelx.com

It would be more elegant if I was able to use FW rules, but it does not appear the my Unifi router has the ability to create such a rule.

With regards to SRC, when I use the following docker-compose config, I get the error below. It appears that it is choking on the SRC IP:
```
  juicepassproxy:
    build: ./juicepassproxy
    hostname: juicepassproxy
    container_name: juicepassproxy
    restart: unless-stopped
    logging:
      driver: json-file
#     network_mode: host
    ports:
      - 8042:8042/udp
    networks:
      public:
    environment:
      - JUICEBOX_LOCAL_IP=192.168.3.33
      - SRC=192.168.2.10
      - MQTT_HOST=192.168.2.10
      - DEBUG=true
    volumes:
      - /etc/localtime:/etc/localtime:ro      
```

If instead I use host network_mode, it works. Not ideal, but I'll live with it unless someone has any pointers?
```
  juicepassproxy:
    build: ./juicepassproxy
    hostname: juicepassproxy
    container_name: juicepassproxy
    restart: unless-stopped
    logging:
      driver: json-file
    network_mode: host
    environment:
      - JUICEBOX_LOCAL_IP=192.168.3.33
      - SRC=192.168.2.10
      - MQTT_HOST=192.168.2.10
      - DEBUG=true
    volumes:
      - /etc/localtime:/etc/localtime:ro      
```

DEBUG OUTPUT:
```
juicepassproxy      | 11/09/23 12:00:37: SRC: 192.168.2.10
juicepassproxy      | 11/09/23 12:00:37: DST: 138.91.137.23
juicepassproxy      | 11/09/23 12:00:37: ENELX_SERVER: jbv1.emotorwerks.com
juicepassproxy      | 11/09/23 12:00:37: ENELX_PORT: 8042
juicepassproxy      | 11/09/23 12:00:37: MQTT_HOST: 192.168.2.10
juicepassproxy      | 11/09/23 12:00:37: MQTT_PORT: 1883
juicepassproxy      | 11/09/23 12:00:37: MQTT_DISCOVERY_PREFIX: homeassistant
juicepassproxy      | 11/09/23 12:00:37: DEVICE_NAME: JuiceBox
juicepassproxy      | 11/09/23 12:00:37: DEBUG: true
juicepassproxy      | 11/09/23 12:00:37: COMMAND: python /juicepassproxy/juicepassproxy.py --src 192.168.2.10:8042 --dst 138.91.137.23:8042 --host 192.168.2.10 --port 1883 --discovery-prefix homeassistant --name JuiceBox --debug
juicepassproxy      | 2023-11-09 12:00:38,172 INFO       config_topic: homeassistant/sensor/JuiceBox/JuiceBox-Status/config
juicepassproxy      | 2023-11-09 12:00:38,172 INFO       state_topic: hmd/sensor/JuiceBox/JuiceBox-Status/state
juicepassproxy      | 2023-11-09 12:00:38,172 DEBUG      Creating mqtt client(None) for {mqtt_settings.host}:{mqtt_settings.port}
juicepassproxy      | 2023-11-09 12:00:38,172 DEBUG      Connecting to 192.168.2.10:1883 without SSL
juicepassproxy      | 2023-11-09 12:00:38,176 INFO       config_topic: homeassistant/sensor/JuiceBox/JuiceBox-Current/config
juicepassproxy      | 2023-11-09 12:00:38,176 INFO       state_topic: hmd/sensor/JuiceBox/JuiceBox-Current/state
juicepassproxy      | 2023-11-09 12:00:38,176 DEBUG      Creating mqtt client(None) for {mqtt_settings.host}:{mqtt_settings.port}
[... 15 lines cut ...]
juicepassproxy      | 2023-11-09 12:00:38,182 DEBUG      Creating mqtt client(None) for {mqtt_settings.host}:{mqtt_settings.port}
juicepassproxy      | 2023-11-09 12:00:38,182 DEBUG      Connecting to 192.168.2.10:1883 without SSL
juicepassproxy      | 2023-11-09 12:00:38,184 INFO       config_topic: homeassistant/sensor/JuiceBox/JuiceBox-Voltage/config
juicepassproxy      | 2023-11-09 12:00:38,184 INFO       state_topic: hmd/sensor/JuiceBox/JuiceBox-Voltage/state
juicepassproxy      | 2023-11-09 12:00:38,184 DEBUG      Creating mqtt client(None) for {mqtt_settings.host}:{mqtt_settings.port}
juicepassproxy      | 2023-11-09 12:00:38,184 DEBUG      Connecting to 192.168.2.10:1883 without SSL
juicepassproxy      | 2023-11-09 12:00:38,185 DEBUG      Starting UDP proxy...
juicepassproxy      | 2023-11-09 12:00:38,185 DEBUG      Src: 192.168.2.10:8042
juicepassproxy      | 2023-11-09 12:00:38,186 DEBUG      Dst: 138.91.137.23:8042
juicepassproxy      | Traceback (most recent call last):
juicepassproxy      |   File "/juicepassproxy/juicepassproxy.py", line 221, in <module>
juicepassproxy      |     main()
juicepassproxy      |   File "/juicepassproxy/juicepassproxy.py", line 218, in main
juicepassproxy      |     pyproxy.udp_proxy(args.src, args.dst)
juicepassproxy      |   File "/juicepassproxy/pyproxy/pyproxy.py", line 43, in udp_proxy
juicepassproxy      |     proxy_socket.bind(ip_to_tuple(src))
juicepassproxy      | OSError: [Errno 99] Cannot assign requested address
juicepassproxy      | 2023-11-09 12:00:38,203 DEBUG      Shutting down MQTT client
juicepassproxy      | 2023-11-09 12:00:38,203 DEBUG      Shutting down MQTT client
juicepassproxy      | 2023-11-09 12:00:38,203 DEBUG      Shutting down MQTT client
juicepassproxy      | 2023-11-09 12:00:38,203 DEBUG      Shutting down MQTT client
juicepassproxy      | 2023-11-09 12:00:38,203 DEBUG      Shutting down MQTT client
juicepassproxy      | 2023-11-09 12:00:38,203 DEBUG      Shutting down MQTT client
juicepassproxy      | 2023-11-09 12:00:38,203 DEBUG      Shutting down MQTT client
```

devzwf, 2023-11-09:
juicenet-udp-prod1-usa.enelx.com do not exist ......

and  dont put SRC , let the container determine it , specially if you use bridge ..... the container will not have access to the IP you are entering as SRC.... ==> the error you get