JuiceRescue/juicepassproxy #11: update README

snicker, 2023-11-11:
per #10 - v0.1.0 published
https://github.com/snicker/juicepassproxy/releases/tag/v0.1.0

@Snuffy2 
> I can then test to ensure it works and will then make another PR to update the Readme with revised Docker Compose instructions.

all yours!

snicker, 2023-11-11:
@Snuffy2  - this is failing, likely because I do not have a token defined in the secrets of the repository

> Error: buildx failed with: ERROR: failed to solve: failed to push ghcr.io/snicker/juicepassproxy:0.1.0: unexpected status from POST request to https://ghcr.io/v2/snicker/juicepassproxy/blobs/uploads/: 403 Forbidden

I tried setting up a fine grained access token for deployments, but I must be doing something wrong since GH will not allow me to include a personal access token in the repo secrets.

If you can give me some instructions on setting this part up I can implement this weekend.

Snuffy2, 2023-11-11:
Hopefully, we can fix it easily with the Repo permissions.

Go into the Repo, Settings, Actions, General.

Change the settings per below:
[image]

Then try to rerun the Action.

Go into the Repo, Actions, Click the v0.1.0 run. Then click Re-run jobs and cross your fingers.

[image]

snicker, 2023-11-11:
Excellent- all working now! https://github.com/snicker/juicepassproxy/pkgs/container/juicepassproxy

PR #12 (2023-11-12): Update Readme for GitHub Container Registry
Fixes #11 

Also adds a line to the Dockerfile to explicitly set your repo to the container package.
closed 2023-11-12 by snicker