JuiceRescue/juicepassproxy #21: issue with mqtt client

devzwf, 2023-11-14:
getting 

Traceback (most recent call last):
  File "/juicepassproxy/juicepassproxy.py", line 309, in <module>
    main()
  File "/juicepassproxy/juicepassproxy.py", line 296, in main
    handler = JuiceboxMessageHandler(
              ^^^^^^^^^^^^^^^^^^^^^^^
  File "/juicepassproxy/juicepassproxy.py", line 46, in __init__
    self._init_devices()
  File "/juicepassproxy/juicepassproxy.py", line 66, in _init_devices
    self._init_device_energy_lifetime(device_info)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'JuiceboxMessageHandler' object has no attribute '_init_device_energy_lifetime'. Did you mean: '_init_device_power_lifetime'?
2023-11-14 00:07:54,373 DEBUG      Shutting down MQTT client
2023-11-14 00:07:54,374 DEBUG      Shutting down MQTT client
2023-11-14 00:07:54,374 DEBUG      Shutting down MQTT client
--------------------------------

since today ...

snicker, 2023-11-14:
sorry about that - likely from a merge just a few hours ago- fixed now @devzwf and edge should rebuild shortly

closed 2023-11-14 by snicker

devzwf, 2023-11-14:
no worries 
we are there for testing :)

commit 1bcda783 (2023-11-14): Merge pull request #22 from snicker/20231113_fix_21

fixes #21
commit 29a73a3a (2023-11-14): fixes #21