JuiceRescue/juicepassproxy #29: Docker variable typo error --juicebox_proxy_host

MrDrew514, 2023-11-15:
I'm getting this error when setting the jpp_host variable

juicepassproxy.py: error: unrecognized arguments: --juicebox_proxy_host

 It should be juicepassproxy_host instead of juiceboxproxy_host

snicker, 2023-11-15:
is this on `:edge`? I might recommend grabbing a more recent build of the container, I was working on a lot of stuff tonight and it's possible that there was a state on master that would not run. It should be stable right now.

MrDrew514, 2023-11-15:
yes :edge, should I use :latest?

snicker, 2023-11-15:
I think try deleting your current container image and re-downloading the :edge container and see if that is working

MrDrew514, 2023-11-15:
I just deleted and redownloaded the :edge, when udpc is set to true, the container crashes. It's running fine when set to false.

I'm still getting the same error below
```
usage: juicepassproxy.py [-h] -s SRC -d DST [--debug] [-u USER] [-P PASSWORD]
                         [-H HOST] [-p PORT] [-D DISCOVERY_PREFIX]
                         [--name DEVICE_NAME] [--juicebox_id JUICEBOX_ID]
                         [--update_udpc] [--juicebox_host JUICEBOX_HOST]
                         [--juicepass_proxy_host JUICEPASS_PROXY_HOST]
juicepassproxy.py: error: unrecognized arguments: --juicebox_proxy_host 192.168.2.202
```

MrDrew514, 2023-11-15:
Everything seem to be working now, great work! Thanks

closed 2023-11-15 by MrDrew514