JuiceRescue/juicepassproxy #32: udpc_update is working for a while then stops (stream_close)

MrDrew514, 2023-11-16:
Seems like the udpc_update is working fine for a few hours then it stop working. When I stop the container to log in telnet to check the list I get a bunch of udpc streams, I assume it reaches a limit then it's no longer streaming?

If I /stream_close (2-3-4-5-6-7) and then start up the container, it's working again.

Sorry if this has been fixed already.

Snuffy2, 2023-11-17:
I'm seeing something similar. From the logs, it looks likes at some point an extra UDPC line is added. Then the UDPC check is looking only at the last line and thus that line looks ok and doesn't update the UDPC line even though it isn't looking at line \#2. We probably need to make the logic a bit more robust to ensure it is checking the proper line and that it cleans up extraneous UDPC lines if they get added.

[image]

[image]

MrDrew514, 2023-11-17:
Here's what I'm getting. 

Cleaning up extra udpc lines would definitely help

[image]

[image]

snicker, 2023-11-17:
thanks @Snuffy2 looks good, i was slightly worried about a race condition where a new UDPC stream was added as it reset the original one. that might still be an issue but only for one cycle with your fix.

Snuffy2, 2023-11-17:
Looking at @MrDrew514's screenshot above, their "active" UDPC stream is \#6. In some testing I've also been seeing something similar. Where when we change the UDPC line a lot, sometimes the JuiceBox starts changing a different line instead of \#2 like normal. I'm adapting my update to try to handle that as well.

closed 2023-11-17 by Snuffy2

strawless, 2024-02-21:
Anyone experience this still?  I'm using it via Python and it lasts for like 4 days and then it will just stop.  The process is still running but I don't get a single printout.

j358, 2024-03-27:
I was getting this issue too. I noticed that every few hours the UDPC updater was getting stuck.
I made a modification to add timeouts to all telnet reads and that has fixed the issue so far for me. See pull request #53 

A timeout now looks like this:

> 2024-03-27 13:38:58,439 DEBUG      Writing '0.0' to hmd/sensor/JuiceBox/Power/state
2024-03-27 13:38:58,441 DEBUG      Publish result: (0, 4)
2024-03-27 13:39:10,253 DEBUG      JuiceboxUDPCUpdater check... 
2024-03-27 13:39:16,261 ERROR      Error in JuiceboxUDPCUpdater: timed out
Traceback (most recent call last):
  File "/juicepassproxy/juicepassproxy.py", line 327, in start
    with JuiceboxTelnet(self.juicebox_host, 2000, self.timeout) as tn:
  File "/juicepassproxy/juicebox_telnet.py", line 12, in __enter__
    self.connection = self.open()
                      ^^^^^^^^^^^
  File "/juicepassproxy/juicebox_telnet.py", line 22, in open
    self.connection = Telnet(host=self.host, port=self.port, timeout=self.timeout)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/telnetlib.py", line 221, in __init__
    self.open(host, port, timeout)
  File "/usr/local/lib/python3.12/telnetlib.py", line 238, in open
    self.sock = socket.create_connection((host, port), timeout)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/socket.py", line 852, in create_connection
    raise exceptions[0]
  File "/usr/local/lib/python3.12/socket.py", line 837, in create_connection
    sock.connect(sa)
TimeoutError: timed out

The UDPC updater then continues as normal.

PR #33 (2023-11-17): Close all UDPC streams present
Fixes #32