JuiceRescue/juicepassproxy #43: Charging session interruption

MrDrew514, 2023-12-03:
Since I updated the container, I've noticed that the charging session gets interrupted after a few minutes of being plugged in. Also when the charging is done, I've noticed that a charging session started during the night for about 5 to 10 minutes then it switched back to plugged in. The app is constantly sending me a connection error that it lost connection to the server then it gets back online after a minute or 2. I was wondering if you noticed something similar with your unit, and if it was due to the update_udpc closing the enelx stream.

[image]

MrDrew514, 2023-12-10:
Sorry for opening/closing the issue, it took some times before I figure out what's going on.

The udpc close/add script is also closing the default enelx stream which cause the charger to pause the ongoing session constantly when running. Is it possible to skip this specific stream and keep the checking for the others?

fearlesschicken, 2024-01-17:
I am seeing the same thing happen here:
Initial charging was going fine because juicepassproxy stopped working.  I restarted it afterwards and that's when we start seeing the problem:

[image]

dambrosioj, 2024-02-21:
Is this still an issue?  I am hesitant to set this up because of users documenting this.

MrDrew514, 2024-02-21:
Afaik yes, but shutting down the container stops the interruptions so don't be afraid of trying it yourself.

dambrosioj, 2024-02-21:
> Afaik yes, but shutting down the container stops the interruptions so don't be afraid of trying it yourself.

Thanks for the fast reply. Maybe I am misunderstanding but wouldn’t that mean the integration to home assistant stops when the container is down not redirecting the traffic?

MrDrew514, 2024-02-21:
The container collects the info of the charger and redirect it to home assistant via MQTT. The bug should occurs only if you are using the update_udpc which constantly check for the "stream" and update it to the correct ip directly on the machine. This step seems to be interrupting the ongoing session (Home assistant + physical charger, app won't resume).

dambrosioj, 2024-02-21:
> The container collects the info of the charger and redirect it to home assistant via MQTT. The bug should occurs only if you are using the update_udpc which constantly check for the "stream" and [...]

Ah that makes total sense. Thanks for that explanation.

dambrosioj, 2024-02-21:
> The container collects the info of the charger and redirect it to home assistant via MQTT. The bug should occurs only if you are using the update_udpc which constantly check for the "stream" and [...]

I started setting up the container today and now understand this a bit better.  So is the idea we can startup the container and set the update_udpc = true, it will collect all the data send that to mqtt, hardcode the dns to pass the udp port to homeassistant and then we just shut down the container.  That persists after reboot?

Snuffy2, 2024-06-06:
I'm going to close this for now with the release of v0.3.0 which will hopefully fix this. Please open a new issue if you are still having problems.

closed 2024-06-06 by Snuffy2