JuiceRescue/juicepassproxy #44: Multiple Juiceboxes

jeffml42, 2023-12-07:
Possible to support multiple juiceboxes?  I was thinking I could just install another docker instance but ports and MQTT name collision probably need to be worked out.

chevycc, 2024-03-18:
Can we use a different UDP port?

jeffml42, 2024-05-04:
I was able to designate an alternate local container port by using the under documented SRC variable.

Currently SRC set to 0.0.0.0:8048 works as the best workaround until either an additional variable is added (something like DEFAULT_LOCAL_PORT) or SRC is parsed to allow just the port (:8048).  Currently an undefined SRC variable will set it to the local IP but only if no port is defined.  You can also set SRC explicitly to your current IP plus port but that has failed for me when the container IP changes.  0.0.0.0 seems to work instead.

Snuffy2, 2024-06-06:
You can now specify different ports in v0.3.0 so you should now be able to run multiple versions of the container.

closed 2024-06-06 by Snuffy2