JuiceRescue/juicepassproxy #46: Best practices to continue using old firmware?

aminorjourney, 2023-12-13:
Hi all. Great project, glad to see it in use.  I doubt I'm going to be useful for coding, but happy to test. Will try and spin up a couple of dockers this weekend to test. 

Currently still rocking the old EnelX JuiceNet app, and still have both load-sharing and the HA integration working. Until I have more time, I'm going to block both JuiceBox Pros from accessing the Internet until I spin up a couple of dockers. 

I'm curious if anyone has thought about reprogramming the original chipset in the JuiceBoxes with new firmware?

FalconFour, 2024-03-27:
JuiceBox uses Zentri (now SiLabs) DMS. If you block access to the DMS server (to which I'm not sure the name, but the docs will certainly help), of course OTA will fail. Not a bad idea -- though I'm increasingly starting to think that the best way to proxy the JB would be a dedicated, customized WiFi access point that proxies only selected traffic - so since the only thing that'd pass is UDP traffic, it'd fall back to the known protocols anyway.

I can definitely say that reprogramming the chipset would be a monumental task. The chip you'd be looking to hack would be the WiFi processor (SiLabs/Zentri), and it takes security seriously. You can probably JTAG into it and use the Zentri dev kit to reflash it without "owning" that chip's specific UUID (internal identifier to the Zentri systems), but it gets hairy just to program it. Not really a long-term solution for making it accessible to other people without JTAG and soldering skills. Typically, and even at manufacturing, the firmware is downloaded via WiFi by having it check-in to the Zentri server and download its assigned firmware.

I've got increasing interest in getting my JuiceBox(s) hooked into my HA system, though I don't charge at home, so it's more a curiosity and desire to help than anything ;)

Snuffy2, 2024-06-06:
The IGNORE_ENELX option will block UDP traffic from EnelX to the JuiceBox. JuicePass Proxy does not block or mess with any other connections between a JuiceBox and EnelX. You can likely block those within your router settings if needed.

closed 2024-06-06 by Snuffy2