JuiceRescue/juicepassproxy #50: add extra option without changing DNS

ivanfmartinez, 2024-02-14:
I'm still using the official homeassistant integration that is working, but I want to test this proxy without making changes to my juicebox device settings, because I still need the functionality of integration that allow to set maximum current (I use this to control charging current based on solar panel generation).

The documentation show 2 options https://github.com/snicker/juicepassproxy/blob/master/README.MD?plain=1#L28 , but there are a 3rd one without creating DNS and without changing data on the device

No need to configure an specific DNS entry or change juicebox to use different address, this option requires that router supports redirection and masquerade, and can easily disabled to return to direct use of enel x services.

What should be done :
- check the port that juicebox use using the telnet command
- create a redirect rule on router redirecting all udp traffic that comes from juicebox to that udp port to the juicepassproxy IP:PORT
- create a masquerade rule on router to masquerade the traffic that goes to juicepassproxy IP:PORT
- depending of your forward rules, maybe you need to create an extra rule to allow that forwarding traffic

```
# This are the mikrotik rules for my case
# 10.1.12.230 - juicebox
# 10.1.0.22 - juicepassproxy
add action=dst-nat chain=dstnat comment=juicebox dst-port=8042 protocol=udp src-address=10.1.12.230 to-addresses=10.1.0.22 to-ports=8042
#
add action=masquerade chain=srcnat comment=juicebox dst-address=10.1.0.22 dst-port=8042 protocol=udp src-address=10.1.12.230

```

Scott8586, 2024-03-16:
I also have this working via a pair of NAT rules: dst-nat+masquerade on an edgerouter x (without using UPDATE_UDPC).  I can provide examples and screenshots of the rules GUI if desired.

Snuffy2, 2024-06-06:
Feel free to submit a PR explaining the alternate setup option in the readme.

closed 2024-06-07 by Snuffy2