JuiceRescue/juicepassproxy #52: Identify unknown fields on messages

ivanfmartinez, 2024-03-14:
I'm checking some messages that are sent by my juicebox to try to understand more about some missing fields that are available on home assistant juicenet  integration and not in this proxy.

This line of code have some unidentified fields
https://github.com/snicker/juicepassproxy/blob/master/juicepassproxy.py#L231

looking the messages that my device is sending there are some fields that I have some idea about what they represent, and maybe more people can confirm or help to get more information

- v 
    - always 09u on my messages
 
- s 
    -  always 689 on my messages

- F
    - allways 11 on my messages

- u
     -  incrementing value, does not appear to be a timestamp (grows less than the elapsed time)

- M an C 
    -  appear to be related to configured max current to be used, both mine came with 0024 which was the last value that I have configured when HA integration was working

- m 
    - came with 0032 which appear to be the maximum current on the instalation.

- t
    - allways 09 on my messages

- e
    - variable values (including, positive, negative, zero)

- r
    - allways 60 on my messages

- i  
    - 2 digit sequence/counter ?

- b B and P 
    -  allways zero on my captures

- r
    - allways 995 on my messages

FalconFour, 2024-03-27:
Would help to post a (near) full report string to digest - the protocol always puts the digits in the same order but, for example, your list contains two "r"'s at the end and not quite sure if it matches up. It seems the protocol has been extended a bit and some stuff might not be knowable, but I know a few of them...

v - Protocol version (09u on yours) - I think I last knew of protocol v7. Tells the server what fields to expect (using a "scanf" style syntax).
i - Interval number. It contains a 96-slot interval memory (15-minute x 24-hour cycle) and this tells you how much energy was consumed in the rolling window as it reports one past (or current, if it's reporting the "right-now" interval) interval per message.
The letter after "i" = the energy in that interval (usually 0 if you're not charging basically 24/7)
t - If I recall correctly, this is the report time in seconds - "every 9 seconds" (or may end up being 10). It can change its reporting interval if the bit mask in the reply command indicates that it should send reports faster (yet to be determined).
m - Unit rating - 32 amps in your case ("I'm a JuiceBox 32")
M - echoed ampcmd - reports the current that it's set to (you guessed it!)

ivanfmartinez, 2024-03-31:
@FalconFour thanks for the information

I just edited my description The second "r" entry was "E" that I just corrected.

I'm started a branch to create some sensors for this extra information, andf make a PR after testing to get this new values

Snuffy2, 2024-06-08:
Referenced from #77
Please post further comments there.

closed 2024-06-08 by Snuffy2