JuiceRescue/juicepassproxy #72: How to add to HA OS non docker.

troydangerfield, 2024-06-04:
This may be a stupid qustion, but I have spent some time looking.  My Juicebox 40 no longer works via the app, no support so I want to just capture data from the unit.  I have HA installed on an Intel PC via an image. so not in a docker.  How do I use this great project?  I found a wrapper but this will not work with my setup.  Looking foward to finally getting some data from this unit.  Thanks again for your great work so far.

troydangerfield, 2024-06-05:
You can close off this item, Wozz  (https://github.com/wozz/ha-addons/tree/main/juicepassproxy) has put a wrapper around your code.  Thanks Wozz :)

closed 2024-06-06 by Snuffy2

starsoccer, 2024-06-20:
Any plans to convert this repo to an home assistant addon natively so another repo does not need to be used? Support could still be maintained for using docker standalone

starsoccer, 2024-07-30:
Bump on this