JuiceRescue/juicepassproxy #80: Voltage not maching string

troydangerfield, 2024-06-19:
When I pull the sensor data into HA I see my voltage as 24.7 but in the string it shows as V247 any suggestions on how to fix this?

Thanks again for all the help with this great project!!!!

Let me know if you need any other information.
Troy.

LOG:

2024-06-19 12:50:52  INFO      [ha_mqtt_discoverable.sensors] Setting Voltage to **24.7** using hmd/sensor/JuiceBox/Voltage/state
..
2024-06-19 12:50:52  INFO      [ha_mqtt_discoverable.sensors] Setting Temperature to 93.2 using hmd/sensor/JuiceBox/Temperature/state
2024-06-19 12:50:52  INFO      [ha_mqtt_discoverable.sensors] Setting Power to 0.0 using hmd/sensor/JuiceBox/Power/state
2024-06-19 12:51:22  DEBUG     [juicebox_mqtthandler] From JuiceBox: b'081107211258014850420158808:**V247**,L11097,S0,T34,E14,i84,e1,t30:\n\r\n'

ivanfmartinez, 2024-06-19:
What is the version and model of your juicebox ? 

This message does not have the starting version and neither the checksum, and the Voltage field is missing one digit compared with others.

Thats why value is wrong.

ivanfmartinez, 2024-06-19:
@troydangerfield can you send more messages from you device, when not connected, connected and charging also ?

and please indicate the what the condition for each message

troydangerfield, 2024-06-20:
> What is the version and model of your juicebox ?
> 
> This message does not have the starting version and neither the checksum, and the Voltage field is missing one digit compared with others.
> 
> Thats why value is wrong.

This is an old unit **JuiceBox 40 Pro** No LED's or display just a box and cable.  I lost app support about 6 months ago, so this project has allowed me to read the data.  

Telnet info is 
**system.version: WiConnect-beta-2.2.0.6, Built:2015-03-23 16:37:59 for AMW006.4, Board:AMW006-A01.1**

Looking over the code I had to set the following to fix the id error 

**email.name_address:** 

I also changed the remote server address to point to my HA box (note I am using Wozz's wrapper.)

I will send more Debug logs once I put the car on charge tonight.

One option I guess is I could just create a new sensor and X's the value by 10 :)

Thanks

ivanfmartinez, 2024-06-20:
@troydangerfield I added to my PR https://github.com/snicker/juicepassproxy/pull/69 decoding for the message that you sent, and if no version found in message it will not multiply by 0.1

If you can test this changes we will know also if the commands work in your device.

Also I dont think that will be a problem to don't multiply by 0.1 if voltage that came from juicebox is lower than 300.

Probably more people with other versions will came searching for solution with other versions and we will have more information about best solution this this differecences.

troydangerfield, 2024-06-22:
As I have used the wrapper, I can not seem to 1 find the code on the HA system ( I have looked, and would love some direction) or 2 how to load a differnt version.  Any help would be great.

On the log side, this is plugged in

hmd/sensor/JuiceBox/Power/state
2024-06-22 03:55:23  DEBUG     [juicebox_mqtthandler] From JuiceBox: b'081107211258014850420158808:V247,L11097,E67,A0,T20,t10,E14,i49,e1:\n\r\n'
2024-06-22 03:55:23  DEBUG     [juicebox_mqtthandler] Publish Basic Message: {'type': 'basic', 'current': 0.0, 'energy_session': 14.0, 'voltage': 24.7, 'energy_lifetime': 11097.0, 'temperature': 68.0, 'report_time': '10', 'interval': '49', 'power': 0.0, 'data_from_juicebox': '081107211258014850420158808:V247,L11097,E67,A0,T20,t10,E14,i49,e1:\n\r\n'}
2024-06-22 03:55:23  INFO      [ha_mqtt_discoverable.sensors] Setting Current to 0.0 using hmd/sensor/JuiceBox/Current/state
2024-06-22 03:55:23  INFO      [ha_mqtt_discoverable.sensors] Setting Energy (Session) to 14.0 using hmd/sensor/JuiceBox/Energy--Session-/state
2024-06-22 03:55:23  INFO      [ha_mqtt_discoverable.sensors] Setting Voltage to 24.7 using hmd/sensor/JuiceBox/Voltage/state
2024-06-22 03:55:23  INFO      [ha_mqtt_discoverable.sensors] Setting Energy (Lifetime) to 11097.0 using hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-06-22 03:55:23  INFO      [ha_mqtt_discoverable.sensors] Setting Temperature to 68.0 using hmd/sensor/JuiceBox/Temperature/state
2024-06-22 03:55:23  INFO      [ha_mqtt_discoverable.sensors] Setting Power to 0.0 using hmd/sensor/JuiceBox/Power/state

This is charging

[ha_mqtt_discoverable.sensors] Setting Power to 338.39 using hmd/sensor/JuiceBox/Power/state
2024-06-22 04:02:34  DEBUG     [juicebox_mqtthandler] From JuiceBox: b'081107211258014850420158808:V247,L11097,E60,A137,T20,t10,E14,i94,e2:\n\r\n'
2024-06-22 04:02:34  DEBUG     [juicebox_mqtthandler] Publish Basic Message: {'type': 'basic', 'current': 13.7, 'energy_session': 14.0, 'voltage': 24.7, 'energy_lifetime': 11097.0, 'temperature': 68.0, 'report_time': '10', 'interval': '94', 'power': 338.39, 'data_from_juicebox': '081107211258014850420158808:V247,L11097,E60,A137,T20,t10,E14,i94,e2:\n\r\n'}
2024-06-22 04:02:34  INFO      [ha_mqtt_discoverable.sensors] Setting Current to 13.7 using hmd/sensor/JuiceBox/Current/state
2024-06-22 04:02:34  INFO      [ha_mqtt_discoverable.sensors] Setting Energy (Session) to 14.0 using hmd/sensor/JuiceBox/Energy--Session-/state
2024-06-22 04:02:34  INFO      [ha_mqtt_discoverable.sensors] Setting Voltage to 24.7 using hmd/sensor/JuiceBox/Voltage/state
2024-06-22 04:02:34  INFO      [ha_mqtt_discoverable.sensors] Setting Energy (Lifetime) to 11097.0 using hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-06-22 04:02:34  INFO      [ha_mqtt_discoverable.sensors] Setting Temperature to 68.0 using hmd/sensor/JuiceBox/Temperature/state
2024-06-22 04:02:34  INFO      [ha_mqtt_discoverable.sensors] Setting Power to 338.39 using hmd/sensor/JuiceBox/Power/state

I have also noted that the voltage always shows 24.7 and I do know our voltage in this are goes up and down, and Setting Energy (Session) is always showing 14.0.

The Temp / current / energy all change when charging.

Thanks again for your keen help.

Troy

ivanfmartinez, 2024-06-23:
@troydangerfield this two messages does not have the S field, the first that you send has the S0, what was the state when that message was sent ?  The car was really unplugged that time ?

The Energy Lifetime is going up at least ?

Maybe E is a different thing in this version of protocol.

About Voltage does not changing we cannot do much if the device does not send correct values.
Here I have found a bug with voltage on the charger, when I plug Volvo XC40 for charge the voltage drops from around 220 to around 139-140 V (with other brands that I have tested there is no problem)

I dont know if I understand correctly about "As I have used the wrapper, I can not seem to 1 find the code on the HA system ( I have looked, and would love some direction) or 2 how to load a differnt version."

I use a docker container and to use different version I have override the files using VOLUMES:
```
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./tests/juicepassproxy.py:/juicepassproxy/juicepassproxy.py:ro

```

I'm placing this two new messages on my version at #69 and this can help integrate later on main code.

troydangerfield, 2024-06-28:
Update:  
I have been unable to locate the code on the HA sub system (have installed SSH etc) so I cannot test your hard work above.

So I created some extra sensors, that * voltage by 10 and then by the current to work out the power.  This is a work around until I can find a way to access the code and make changes.  It has been suggested we turn this into a Addon for HAOS, which would mean we can update the code via system.

Also some extra logs,

'081107211258014850420158808:**V221**,L11162,S0,T27,**E14**,i73,e1,t30:\n\r\n'
I note that the Voltage changes each time the unit is turned on, so my guess is it samples the voltage and stores this.
The E14 I believe is the AMPS set via the app, this was the last value I set before I lost access to it.

'081107211258014850420158808:V247,L11156,**E13322**,A138,T28,t10,E14,i41,e1:\n\r\n'
Here the second E is the "session power", so it seems that the values are only sent if they change ie Status is only sent if it changes.

Again thanks, I now have some values on my dashboard on what is happing with my EV Charger. :)

ivanfmartinez, 2024-06-29:
@troydangerfield I have added code to keep this duplicate values in my PR, but I don´t know exactly what is best behaviour to use this values now.

I have made an estimate for status based on current when it it not sent. But for now  without more constant logs I dont think that I can support all this differences from newer protocol.

snicker, 2025-01-20:
@ivanfmartinez  is this issue still worth investigating?

troydangerfield, 2025-01-20:
At this stage no. I am only capturing data for historical purposes and my work around is holding up. Thanks again for checking. Have a great day.

closed 2025-01-20 by snicker