JuiceRescue/juicepassproxy #83: Keep getting a Proxy task failed:ConnectionRefusedError and then the program crashes

rszimm, 2024-08-21:
I keep getting this error in the log:
`ERROR     [__main__] A JuicePass Proxy task failed: ConnectionRefusedError: [Errno 111] Connection refused`

Here's what's in my juicepassproxy.yaml
```
ENELX_IP: 158.47.1.128
ENELX_PORT: '8051'
ENELX_SERVER: juicenet-udp-prod4-usa.enelx.com
JUICEBOX_ID: <redacted>
LOCAL_IP: 172.16.100.2
```
Here's what I get when I telnet into my juicebox before I run juiceboxproxy:

```
Trying 192.168.10.18...
Connected to juicebox1.lan.
Escape character is .
EMWERK-JB201-1.0.46, Gecko_OS-STANDARD-4.2.7-11064, WGM160P
> list
list
! # Type  Info
# 1 FILE  godaddy_ca.pem-1.0.0 (1390, 0)
# 2 UDPC  juicenet-udp-prod4-usa.enelx.com:8051 (6070)
> exit
```

Eventually the whole thing craps out (you can see the unhandled exception in the log)

[juicepassproxy.log](https://github.com/user-attachments/files/16697429/juicepassproxy.log)

rszimm, 2024-08-21:
Oh, one other thing.  Here's my docker-compose file:
```
networks:
  default:
    driver: bridge
    ipam:
      config:
        - subnet: 172.16.100.0/24
services:
  juicepassproxy:
    image: ghcr.io/snicker/juicepassproxy:latest
    hostname: juicepassproxy
    container_name: juicepassproxy
    restart: unless-stopped
    logging:
      driver: json-file
    ports:
      - 8047:8047/udp
      - 8051:8051/udp
    environment:
      - JUICEBOX_HOST=192.168.10.18
      - UPDATE_UDPC=true
      - JPP_HOST=192.168.10.10
      - ENELX_PORT=8051
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./config:/config
      - ./log:/log #Optional
```

ivanfmartinez, 2024-08-21:
have you tried the telnet inside the container ? 

If it does not work the problem is related to your network configuration, my container works without explicit network configuration.

try this :
```
docker-compose run --rm --entrypoint=/bin/bash juicepassproxy

# and install telnet inside the container to use it later

apt-get install telnet
```

rszimm, 2024-08-21:
I can telnet from within the container.  
```
root@hydra:~/juicepassproxy# docker-compose run --rm --entrypoint=/bin/bash juicepassproxy
Creating juicepassproxy_juicepassproxy_run ... done
root@juicepassproxy:/# telnet juicebox1 2000
Trying 192.168.10.18...
Connected to juicebox1.
Escape character is '^]'.
EMWERK-JB201-1.0.46, Gecko_OS-STANDARD-4.2.7-11064, WGM160P
> list
list
! # Type  Info
# 1 FILE  godaddy_ca.pem-1.0.0 (1390, 0)
# 2 UDPC  juicenet-udp-prod4-usa.enelx.com:8051 (20265)
> exit
exit
Connection closed by foreign host.
root@juicepassproxy:/# exit
```
I didn't think that was the issue as the logs seem to indicated that it's connecting to the telnet at some point, and after I launch the service I can telnet into the juicebox and see that the settings have changed:
```
12:59:55root@hydra:~/juicepassproxy# telnet juicebox1 2000
Trying 192.168.10.18...
Connected to juicebox1.lan.
Escape character is '^]'.
EMWERK-JB201-1.0.46, Gecko_OS-STANDARD-4.2.7-11064, WGM160P
> list
list
! # Type  Info
# 1 FILE  godaddy_ca.pem-1.0.0 (1390, 0)
# 2 UDPC  192.168.10.10:8051 (60102)
> exit
```

ivanfmartinez, 2024-08-21:
maybe the connection refused is in the connection to ENEL_X server.

Have you tried the ignore_enelx option ?

rszimm, 2024-08-21:
Tried that.  I've attached the updated logs with the ignore_enelx on and debug logging
[juicepassproxy.log](https://github.com/user-attachments/files/16698919/juicepassproxy.log)

ivanfmartinez, 2024-08-21:
```
<Task finished name='mqtt_handler' coro=<JuiceboxMQTTHandler.start() done, defined at /juicepassproxy/juicebox_mqtthandler.py:320> exception=ConnectionRefusedError(111, 'Connection refused')>
```

You did not have defined the mqtt server, if not defined the default is 127.0.0.1 which will not work on the default docker image.

rszimm, 2024-08-21:
that was it.  I gave it the external host ip in the config and it still got connection failure.  A bit later I discovered that my mqtt server was on the same host, but it was only listening for connections from localhost.  Because the docker image sets up a local network between the host and the image it couldn't connect to it. 

Reconfiguring the mqtt server to listen on all addresses fixed it.

What a pain.

closed 2024-09-12 by Snuffy2