JuiceRescue/juicepassproxy #84: Lingering power reading (not real)

barrymichels, 2024-09-09:
When my charging session is complete, Home Assistant shows the charger is still delivering around 325W for hours afterward. The EnelXWay app does not show this current in the charging curve. My home power monitor doesn't show any extra power either. I can't tell if it's the proxy or HA that is holding onto this value.
Any suggestions on where to start figuring out where this value is getting stuck?
[image]
[image]

ivanfmartinez, 2024-09-09:
There are some strange bugs on the JuiceBox.

Here (and others confirmed same behaviour) when I plug my Volvo XC40 the Voltage indication changes from 220V to around 139V. This makes wrong power values but apparently the energy stay correct.

On the same Juicebox when I plug Bolt or BYD Yuan Plus the Voltage and power stay correct.

Juicepassproxy only send the values sent from juicebox to HA. Maybe the enelx servers are "correcting"  some bug on the juicebox before showing the value to the user

Legot, 2024-10-09:
> Here (and others confirmed same behaviour) when I plug my Volvo XC40 the Voltage indication changes from 220V to around 139V. This makes wrong power values but apparently the energy stay correct.

~139V is the average DC voltage out of a full bridge rectifier being run off of 220V single phase. The low voltage side of the juicebox is probably just reading that DC voltage from it's power supply and multiplying it by 1.57 (when running on single phase) to estimate the mains voltage.

For a 3 phase input it is probably doing something similar and rectifying them then multiplying the output DC voltage by 1.654

ivanfmartinez, 2024-10-09:
> ~139V is the average DC voltage out of a full bridge rectifier being run off of 220V single phase. The low voltage side of the juicebox is probably just reading that DC voltage from it's power [...]

In my case I have two phases to get the 220V, and the value is correct when Bolt is connected, but not when the Volvo is connected on same charger.

ivanfmartinez, 2024-10-19:
@barrymichels you still have this problem ? can you send some logs with the messages that came from Juicebox to JPP ?

barrymichels, 2024-10-19:
Yes, it was an issue 2 charging sessions ago. But my session last night
wasn't logged. Juicepassproxy failed yesterday and I missed the logging for
the charging session last night.

Looks like my charger has gone offline. I'll try again tonight to capture
the log. It happens every time. When the charging session is over, it shows
a 300W load for about 3 hours afterward. The home power monitor does not
show that load.

barrymichels, 2024-10-20:
2024-10-20 08:28:37  INFO      [ha_mqtt_discoverable.sensors] Setting
Current to 1.3 using hmd/sensor/JuiceBox/Current/state
2024-10-20 08:28:37  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Session) to 15415.0 using hmd/sensor/JuiceBox/Energy--Session-/state
2024-10-20 08:28:37  INFO      [ha_mqtt_discoverable.sensors] Setting
Voltage to 247.8 using hmd/sensor/JuiceBox/Voltage/state
2024-10-20 08:28:37  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Lifetime) to 6302442.0 using
hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-10-20 08:28:37  INFO      [ha_mqtt_discoverable.sensors] Setting
Status to Charging using hmd/sensor/JuiceBox/Status/state
2024-10-20 08:28:37  INFO      [ha_mqtt_discoverable.sensors] Setting
Temperature to 68.0 using hmd/sensor/JuiceBox/Temperature/state
2024-10-20 08:28:37  INFO      [ha_mqtt_discoverable.sensors] Setting
Current Rating to 40.0 using hmd/sensor/JuiceBox/Current-Rating/state
2024-10-20 08:28:37  INFO      [ha_mqtt_discoverable.sensors] Setting
Frequency to 60.03 using hmd/sensor/JuiceBox/Frequency/state
2024-10-20 08:28:37  INFO      [ha_mqtt_discoverable.sensors] Setting Power
to 322.14 using hmd/sensor/JuiceBox/Power/state
2024-10-20 08:28:46  INFO      [ha_mqtt_discoverable.sensors] Setting
Current to 1.3 using hmd/sensor/JuiceBox/Current/state
2024-10-20 08:28:46  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Session) to 15416.0 using hmd/sensor/JuiceBox/Energy--Session-/state
2024-10-20 08:28:46  INFO      [ha_mqtt_discoverable.sensors] Setting
Voltage to 247.7 using hmd/sensor/JuiceBox/Voltage/state
2024-10-20 08:28:46  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Lifetime) to 6302443.0 using
hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-10-20 08:28:46  INFO      [ha_mqtt_discoverable.sensors] Setting
Status to Charging using hmd/sensor/JuiceBox/Status/state
2024-10-20 08:28:46  INFO      [ha_mqtt_discoverable.sensors] Setting
Temperature to 68.0 using hmd/sensor/JuiceBox/Temperature/state
2024-10-20 08:28:46  INFO      [ha_mqtt_discoverable.sensors] Setting
Current Rating to 40.0 using hmd/sensor/JuiceBox/Current-Rating/state
2024-10-20 08:28:46  INFO      [ha_mqtt_discoverable.sensors] Setting
Frequency to 60.03 using hmd/sensor/JuiceBox/Frequency/state
2024-10-20 08:28:46  INFO      [ha_mqtt_discoverable.sensors] Setting Power
to 322.01 using hmd/sensor/JuiceBox/Power/state
2024-10-20 08:28:56  INFO      [ha_mqtt_discoverable.sensors] Setting
Current to 1.3 using hmd/sensor/JuiceBox/Current/state
2024-10-20 08:28:56  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Session) to 15417.0 using hmd/sensor/JuiceBox/Energy--Session-/state
2024-10-20 08:28:56  INFO      [ha_mqtt_discoverable.sensors] Setting
Voltage to 247.7 using hmd/sensor/JuiceBox/Voltage/state
2024-10-20 08:28:56  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Lifetime) to 6302444.0 using
hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-10-20 08:28:56  INFO      [ha_mqtt_discoverable.sensors] Setting
Status to Charging using hmd/sensor/JuiceBox/Status/state
2024-10-20 08:28:56  INFO      [ha_mqtt_discoverable.sensors] Setting
Temperature to 68.0 using hmd/sensor/JuiceBox/Temperature/state
2024-10-20 08:28:56  INFO      [ha_mqtt_discoverable.sensors] Setting
Current Rating to 40.0 using hmd/sensor/JuiceBox/Current-Rating/state
2024-10-20 08:28:56  INFO      [ha_mqtt_discoverable.sensors] Setting
Frequency to 60.03 using hmd/sensor/JuiceBox/Frequency/state
2024-10-20 08:28:56  INFO      [ha_mqtt_discoverable.sensors] Setting Power
to 322.01 using hmd/sensor/JuiceBox/Power/state
2024-10-20 08:29:05  INFO      [ha_mqtt_discoverable.sensors] Setting
Current to 1.3 using hmd/sensor/JuiceBox/Current/state
2024-10-20 08:29:05  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Session) to 15418.0 using hmd/sensor/JuiceBox/Energy--Session-/state
2024-10-20 08:29:05  INFO      [ha_mqtt_discoverable.sensors] Setting
Voltage to 247.7 using hmd/sensor/JuiceBox/Voltage/state
2024-10-20 08:29:05  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Lifetime) to 6302445.0 using
hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-10-20 08:29:05  INFO      [ha_mqtt_discoverable.sensors] Setting
Status to Charging using hmd/sensor/JuiceBox/Status/state
2024-10-20 08:29:05  INFO      [ha_mqtt_discoverable.sensors] Setting
Temperature to 68.0 using hmd/sensor/JuiceBox/Temperature/state
2024-10-20 08:29:05  INFO      [ha_mqtt_discoverable.sensors] Setting
Current Rating to 40.0 using hmd/sensor/JuiceBox/Current-Rating/state
2024-10-20 08:29:05  INFO      [ha_mqtt_discoverable.sensors] Setting
Frequency to 60.03 using hmd/sensor/JuiceBox/Frequency/state
2024-10-20 08:29:05  INFO      [ha_mqtt_discoverable.sensors] Setting Power
to 322.01 using hmd/sensor/JuiceBox/Power/state
2024-10-20 08:29:05  INFO      [juicebox_udpcupdater] JuiceboxUDPCUpdater
Check Starting
2024-10-20 08:29:06  INFO      [juicebox_udpcupdater] UDPC IP correct

Not sure how to get more logs out of the system.
The 322.01W power reading is not real. My entire house is currently drawing
145W total. The Enel X Way app is currently showing 0W.

[image: image.png]

barrymichels, 2024-10-20:
Turned on DEBUG=true

2024-10-20 08:36:50  DEBUG     [juicebox_mqtthandler] From JuiceBox:
b'0910042001280653621122628723:v09u,s997,F11,u00083333,V2475,L00006302485,S02,T20,M0040,C0040,m0040,t09,i13,e00000,f6002,r38,b000,B0000000,P0,E0015458,A00013,p0009!B7C:'
2024-10-20 08:36:50  DEBUG     [juicebox_mqtthandler] Publish Basic
Message: {'type': 'basic', 'current': 1.3, 'energy_session': 15458.0,
'protocol_version': '09u', 'unknown_s': '997', 'unknown_F': '11',
'loop_counter': '00083333', 'voltage': 247.5, 'energy_lifetime': 6302485.0,
'status': 'Charging', 'temperature': 68.0, 'current_setting': 40.0,
'unknown_C': '0040', 'current_rating': 40.0, 'report_time': '09',
'interval': '13', 'unknown_e': '00000', 'frequency': 60.02, 'unknown_r':
'38', 'unknown_b': '000', 'unknown_B': '0000000', 'unknown_P': '0',
'unknown_p': '0009', 'power': 321.75, 'data_from_juicebox':
'0910042001280653621122628723:v09u,s997,F11,u00083333,V2475,L00006302485,S02,T20,M0040,C0040,m0040,t09,i13,e00000,f6002,r38,b000,B0000000,P0,E0015458,A00013,p0009!B7C:'}
2024-10-20 08:36:50  INFO      [ha_mqtt_discoverable.sensors] Setting
Current to 1.3 using hmd/sensor/JuiceBox/Current/state
2024-10-20 08:36:50  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Session) to 15458.0 using hmd/sensor/JuiceBox/Energy--Session-/state
2024-10-20 08:36:50  INFO      [ha_mqtt_discoverable.sensors] Setting
Voltage to 247.5 using hmd/sensor/JuiceBox/Voltage/state
2024-10-20 08:36:50  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Lifetime) to 6302485.0 using
hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-10-20 08:36:50  INFO      [ha_mqtt_discoverable.sensors] Setting
Status to Charging using hmd/sensor/JuiceBox/Status/state
2024-10-20 08:36:50  INFO      [ha_mqtt_discoverable.sensors] Setting
Temperature to 68.0 using hmd/sensor/JuiceBox/Temperature/state
2024-10-20 08:36:50  INFO      [ha_mqtt_discoverable.sensors] Setting
Current Rating to 40.0 using hmd/sensor/JuiceBox/Current-Rating/state
2024-10-20 08:36:50  INFO      [ha_mqtt_discoverable.sensors] Setting
Frequency to 60.02 using hmd/sensor/JuiceBox/Frequency/state
2024-10-20 08:36:50  INFO      [ha_mqtt_discoverable.sensors] Setting Power
to 321.75 using hmd/sensor/JuiceBox/Power/state
2024-10-20 08:36:52  DEBUG     [juicebox_mqtthandler] From EnelX:
b'CMD00836A0040M040C006S998!32W$'

ivanfmartinez, 2024-10-20:
@barrymichels this information indicates that you are not running the latest version of my branch https://github.com/JuiceRescue/juicepassproxy/pull/69

the messages that come from juicebox indicates a current of "1.3" A, and voltage 247.5, and that indicates the power of 321 W.

If the your home meter don't indicate this consumption juicebox is sending wrong values. And maybe is a BUG that ENEL X known and solved on server side.

I recommend you to use latest version from my branch, or wait some days for the merge. But at this moment I can't see a way to know that indicates for JPP this "bug".

barrymichels, 2024-10-24:
I don't charge that often, so it took some days to get another log.
Switched to branch #69 and charged last night. It didn't have an effect on
the phantom power issue.

2024-10-24 08:54:35  INFO      [juicebox_udpcupdater] JuiceboxUDPCUpdater
Check Starting
2024-10-24 08:54:35  DEBUG     [juicebox_udpcupdater] Active UDPC Stream: 4
2024-10-24 08:54:35  INFO      [juicebox_udpcupdater] UDPC IP correct
2024-10-24 08:54:36  INFO      [juicebox_message] from_string
0910042001280653621122628723:v09u,s343,F11,u00427479,V2476,L00006321756,S02,T22,M0040,C0040,m0040,t09,i12,e00000,f6003,r61,b000,B0000000,P0,E0018917,A00013,p0008!2TS:
2024-10-24 08:54:36  DEBUG     [juicebox_mqtthandler] From JuiceBox:
b'0910042001280653621122628723:v09u,s343,F11,u00427479,V2476,L00006321756,S02,T22,M0040,C0040,m0040,t09,i12,e00000,f6003,r61,b000,B0000000,P0,E0018917,A00013,p0008!2TS:'
decoded=0910042001280653621122628723:v09u,s343,F11,u00427479,V2476,L00006321756,S02,T22,M0040,C0040,m0040,t09,i12,e00000,f6003,r61,b000,B0000000,P0,E0018917,A00013,p0008!2TS:
2024-10-24 08:54:36  DEBUG     [juicebox_mqtthandler] decode/parsed message
= {'type': 'basic', 'current': 1.3, 'energy_session': 18917, 'serial':
'0910042001280653621122628723', 'protocol_version': '09u', 'counter':
'343', 'F': '11', 'loop_counter': '00427479', 'voltage': 247.6,
'energy_lifetime': 6321756, 'status': 'Charging', 'temperature': 71.6,
'current_max_online': 40, 'current_max_offline': 40, 'current_rating': 40,
'report_time': '09', 'interval': 12, 'e': '00000', 'frequency': 60.03, 'r':
'61', 'b': '000', 'B': '0000000', 'P': '0', 'p': '0008', 'power': 322}
2024-10-24 08:54:36  DEBUG     [juicebox_mqtthandler] Publish Basic
Message: {'type': 'basic', 'current': 1.3, 'energy_session': 18917,
'serial': '0910042001280653621122628723', 'protocol_version': '09u',
'counter': '343', 'F': '11', 'loop_counter': '00427479', 'voltage': 247.6,
'energy_lifetime': 6321756, 'status': 'Charging', 'temperature': 71.6,
'current_max_online': 40, 'current_max_offline': 40, 'current_rating': 40,
'report_time': '09', 'interval': 12, 'e': '00000', 'frequency': 60.03, 'r':
'61', 'b': '000', 'B': '0000000', 'P': '0', 'p': '0008', 'power': 322}
2024-10-24 08:54:36  INFO      [ha_mqtt_discoverable.sensors] Setting
Current to 1.3 using hmd/sensor/JuiceBox/Current/state
2024-10-24 08:54:36  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Session) to 18917 using hmd/sensor/JuiceBox/Energy--Session-/state
2024-10-24 08:54:36  INFO      [ha_mqtt_discoverable.sensors] Setting
Voltage to 247.6 using hmd/sensor/JuiceBox/Voltage/state
2024-10-24 08:54:36  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Lifetime) to 6321756 using
hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-10-24 08:54:36  INFO      [ha_mqtt_discoverable.sensors] Setting
Status to Charging using hmd/sensor/JuiceBox/Status/state
2024-10-24 08:54:36  INFO      [ha_mqtt_discoverable.sensors] Setting
Temperature to 71.6 using hmd/sensor/JuiceBox/Temperature/state
2024-10-24 08:54:36  INFO      [ha_mqtt_discoverable.sensors] Setting Max
Current(Online/Device) to 40 using
hmd/sensor/JuiceBox/Max-Current-Online-Device-/state
2024-10-24 08:54:36  INFO      [ha_mqtt_discoverable.sensors] Setting Max
Current(Offline/Device) to 40 using
hmd/sensor/JuiceBox/Max-Current-Offline-Device-/state
2024-10-24 08:54:36  INFO      [ha_mqtt_discoverable.sensors] Setting
Current Rating to 40 using hmd/sensor/JuiceBox/Current-Rating/state
2024-10-24 08:54:36  INFO      [ha_mqtt_discoverable.sensors] Setting
Frequency to 60.03 using hmd/sensor/JuiceBox/Frequency/state
2024-10-24 08:54:36  INFO      [ha_mqtt_discoverable.sensors] Setting Power
to 322 using hmd/sensor/JuiceBox/Power/state
2024-10-24 08:54:40  DEBUG     [juicebox_mqtthandler] From EnelX:
b'CMD40854A0040M040C242S371!AH6$'

barrymichels, 2024-10-24:
> @barrymichels <https://github.com/barrymichels> can you compare the
> unknown parameters during the time that value are correct and when not
> correct ?
>
>
> https://github.com/JuiceRescue/juicepassproxy/blob/95c22a82d44bed4af984f575c1a66b4f36009928/juicebox_message.py#L96
>
> My *p* value is most of the time over 0980, and yours is 0008 and 0009 in
> the logs that you sent. Maybe this if some kind of correction factor that
> we are not using now and ENEL X is using .
>
> my values from log
>
> docker-compose logs | grep -o -E "p[0-9]{3,}"   | sort | uniq -c
>       1 p0013
>       1 p0303
>       1 p0979
>       1 p0982
>     428 p0983
>     418 p0984
>      54 p0985
>      31 p0986
>      23 p0987
>      43 p0990
>     266 p0991
>      17 p0992
>      21 p0993
>     536 p0994
>       2 p0995
>      28 p0997
>
>
> If you get this value (and other unknown) on more situations maybe we can
> find something that we are not using and can use to solve this.

3 p0006
    345 p0007
   1311 p0008
    948 p0009
     81 p0010
     12 p0011
      3 p0102
      3 p0154
      3 p0978
      3 p0979
      9 p0989
   1776 p0990

ivanfmartinez, 2024-10-24:
@barrymichels can you confirm that when power is correct the values is around "099*" and when incorrect like "0000*" ?

if you confirm this, probably this should be used to solve the problem, but we need to figure how this value is used.

barrymichels, 2024-10-24:
Plugged my car in last night, but until the charging started around 4am,
the 'p' value was completely missing. Here's some samples during the charge
session:

power: 0 (p missing)
power: 1895, p: 0102
power: 4927, p: 0978
power: 9612, p: 0989
power: 9878, p: 0990
power: 9906, p: 0990
power: 9882, p: 0990

for the rest of the charging session, p stayed at 0990 while the power
fluctuated slightly around 9.8kW. At the end of the charging session:

power: 9833, p: 0990
power: 322, p: 0010
power 322, p: 0008
power 321, p: 0010
power: 322, p: 0009
power: 321, p: 0011
power: 324, p: 0009

It kept up with this phantom 320W for about 3 hours after the charging
session ended. p fluctuated between 0007 and 0011. power fluctuated between
320 and 324

Will the full log help?

ivanfmartinez, 2024-10-24:
@barrymichels  full log will not help without the details about what is the correct power usage. 

What I can think is that "p" can be a factor with 1000 being 100% and 0011 being 1.1%, in this case if we consider 1% of 320 W this will be 3 W....

This factor should be applied to the current value and this will correct the power.

We probably need someone to get the power reported from juicebox/juicepassproxy compared to an external meter to confirm that case.

Maybe FalconFour can have more information. I asked him on discord.

ivanfmartinez, 2024-10-24:
Just completed charge on Bolt the p value goes down when car stops charging

```
v09u,s411,F11,u00152597,V2122,L00005393884,S02,T31,M0017,C0026,m0032,t09,i73,e00000,f5999,r70,b000,B0000000,P0,E0004992,A00170,p0993!6HF:

v09u,s412,F11,u00152622,V2167,L00005393900,S02,T31,M0017,C0026,m0032,t09,i74,e00000,f6001,r72,b000,B0000000,P0,E0005008,A00004,p0004!IUV:

v09u,s413,F11,u00152648,V2178,L00005393901,S02,T31,M0017,C0026,m0032,t09,i75,e00000,f6000,r72,b000,B0000000,P0,E0005009,A00004,p0004!LTI:

```

FalconFour, 2024-10-24:
Something strange going on. S02 = Charging state, which comes from J1772 pilot. That means relay is closed, charging is active. That's my first clue. "p" is power factor (https://theengineeringmindset.com/power-factor-explained/) and it's basically a power-nerd stat for our purposes; it's merely reporting what the meter IC is saying, but the meter IC already accounts for it in its calculations. Nothing to act on there, but having such an incredibly low power factor is definitely a bit of a clue. Suggests there's some hardware issue in the box...

The strangest thing I see is reporting Charging state (J1772 state C) when the vehicle is unplugged, as well as reporting current when none can be present. All this is coming from the box itself (not JPP's fault, which is just reporting the info correctly).

ivanfmartinez, 2024-10-24:
@FalconFour in my case https://github.com/JuiceRescue/juicepassproxy/issues/84#issuecomment-2435723692  the device has type 2 connector, I dont know if there is some difference from the ones with J1772 connector.

FalconFour, 2024-10-25:
State is state. If the box says it's in state C (S02) when it's unplugged, something is definitely wrong with the box hardware. 🤷‍♂️

ivanfmartinez, 2024-10-25:
In my case it is not unplugged, it is not charging.

And I understood that was the same case for @barrymichels , the charging has stopped but car still plugged.

ivanfmartinez, 2024-10-25:
F value does not change when car is unplugged, only S changes :

```
 
2024-10-25 13:29:03 :v09u,s052,F10,u00240514,V2128,L00005426922,S01,T25,M0020,C0022,m0032,t29,i43,e00496,f6001,r68,b000,B0000000!IVL:

2024-10-25 13:29:20  :v09u,s053,F10,u00240531,V2118,L00005426922,S00,T25,M0020,C0022,m0032,t29,i44,e00254,f6001,r68,b000,B0000000!CBZ:

```

snicker, 2025-01-20:
Closing for inactivity - please try v0.5.0 and report back if issue is not resolved.

closed 2025-01-20 by snicker

barrymichels, 2025-01-22:
Upgraded to v0.5.0, reset my unit to stop the encrypted packets, charged. Still showing lingering power reading at the end of the charging session.

So, not resolved. Probably can't be resolved by this proxy since it's an issue in what the controller is reporting.