JuiceRescue/juicepassproxy #86: ERROR     [__main__] A JuicePass Proxy task failed: ValueError: reuse_port not supported by socket module (run on Windows)

BstNst99, 2024-09-10:
I run juicepassproxy on Windows. it will error by message,
ERROR     [__main__] A JuicePass Proxy task failed: ValueError: reuse_port not supported by socket module

cause of  reuse_port not compatible with Windows.

I just fix problem by edit line 83 of juicebox_mitm.py to from "reuse_port=True" to "reuse_port=False". It work correctly.

FalconFour, 2024-10-05:
Awesome find. My setup just did the same thing when I tried setting it up:
```
2024-10-04 22:22:23  ERROR     [__main__] A JuicePass Proxy task failed: ValueError: reuse_port not supported by socket module
Traceback (most recent call last):
  File "C:\Users\micro\Desktop\juicepassproxy-juicebox_commands\juicepassproxy.py", line 565, in main
    await asyncio.gather(
  File "C:\Users\micro\Desktop\juicepassproxy-juicebox_commands\juicebox_mitm.py", line 59, in start
    await self._connect()
  File "C:\Users\micro\Desktop\juicepassproxy-juicebox_commands\juicebox_mitm.py", line 87, in _connect
    self._dgram = await asyncio_dgram.bind(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\micro\AppData\Local\Programs\Python\Python312\Lib\site-packages\asyncio_dgram\aio.py", line 244, in bind
    transport, protocol = await loop.create_datagram_endpoint(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\micro\AppData\Local\Programs\Python\Python312\Lib\asyncio\base_events.py", line 1399, in create_datagram_endpoint
    _set_reuseport(sock)
  File "C:\Users\micro\AppData\Local\Programs\Python\Python312\Lib\asyncio\base_events.py", line 92, in _set_reuseport
    raise ValueError('reuse_port not supported by socket module')
ValueError: reuse_port not supported by socket module
```

Possibly an issue of little consequence since the target system is Linux (within Home Assistant), but for debug testing on a separate machine, this could be fixed to make life easier.

BstNst99, 2024-10-06:
Now. I change more to support Windows and can run as long service.
1. edit line 78 and 83 of juicebox_mitm.py to from "reuse_port=True" to "reuse_port=False".
2. change line 26 of const.py from  MAX_JPP_LOOP = 10 to MAX_JPP_LOOP = 999999 to protect proxy about when XWaybox temporary offline

FalconFour, 2024-10-11:
FWIW, whenever I pull the latest commit, the first/only thing I have to change is "reuse_port=False" in two lines. There's a tradeoff in making that second (unrelated) change - it's likely meant as a safeguard against the software becoming unstable, but it does indeed have the effect of "crashing" if the box is ever offline (which mine is, unplugged 99% of the time because I'm a giant nerd always powering it off weird batteries and cords - I usually charge at work, ha).

ivanfmartinez, 2024-10-11:
@FalconFour please check the last commit which has a option to select on command line or configuration file

ivanfmartinez, 2024-10-11:
best will be if someone can detect the environment to select correct default on windows