JuiceRescue/juicepassproxy #91: Not seeming to work - or maybe it is?

jonlove234, 2024-10-03:
Hi 👋🏻. After the Enel X shutdown, thought I'd give it a go. I'm not sure it's working, here's the log messages:

`2024-10-03 12:45:40  INFO      [entrypoint.sh] Starting JuicePass Proxy

2024-10-03 12:45:40  INFO      [entrypoint.sh] Docker Environment Variables:
2024-10-03 12:45:40  INFO      [entrypoint.sh] JUICEBOX_HOST: 192.168.13.110
2024-10-03 12:45:40  INFO      [entrypoint.sh] MQTT_HOST: 192.168.13.59
2024-10-03 12:45:40  INFO      [entrypoint.sh] MQTT_PORT: 1883
2024-10-03 12:45:40  INFO      [entrypoint.sh] MQTT_USER: [redacted in case sensitive - but same as used by zwavejs ui]
2024-10-03 12:45:40  INFO      [entrypoint.sh] MQTT_PASS: ****************************************************************
2024-10-03 12:45:40  INFO      [entrypoint.sh] JPP_HOST: [redacted in case sensitive - ip of machine/unraid running docker]
2024-10-03 12:45:40  INFO      [entrypoint.sh] DEBUG: true
2024-10-03 12:45:40  INFO      [entrypoint.sh] UPDATE_UDPC: false
2024-10-03 12:45:40  INFO      [entrypoint.sh] IGNORE_ENELX: true
2024-10-03 12:45:40  INFO      [entrypoint.sh] EXPERIMENTAL: true
2024-10-03 12:45:40  DEBUG     [entrypoint.sh] COMMAND: python3 /juicepassproxy/juicepassproxy.py --juicebox_host 192.168.13.110 --mqtt_host 192.168.13.59 --mqtt_port 1883 --mqtt_user [redacted in case sensitive] --mqtt_password ***** --juicepass_proxy_host [redacted in case sensitive] --config_loc /config --log_loc /log --debug --ignore_enelx --experimental
2024-10-03 12:45:40  WARNING   [__main__] Starting JuicePass Proxy v0.3.1 (Log Level: DEBUG)
2024-10-03 12:45:40  INFO      [__main__] log_loc: /log/juicepassproxy.log
2024-10-03 12:45:40  INFO      [__main__] config_loc: /config/juicepassproxy.yaml
2024-10-03 12:45:40  INFO      [__main__] telnet timeout: 30
2024-10-03 12:45:40  INFO      [__main__] ignore_enelx: True
2024-10-03 12:45:40  INFO      [__main__] enelx_server: juicenet-udp-prod3-usa.enelx.com
2024-10-03 12:45:40  INFO      [__main__] enelx_port: 8047
2024-10-03 12:45:40  INFO      [__main__] local_addr: 172.17.0.8:8047
2024-10-03 12:45:40  INFO      [__main__] enelx_addr: 158.47.1.128:8047
2024-10-03 12:45:40  INFO      [telnetlib3.client] Connected to <Peer 192.168.13.110 2000>
2024-10-03 12:45:42  INFO      [__main__] juicebox_id: [redacted in case sensitive]
2024-10-03 12:45:42  INFO      [__main__] experimental: True
2024-10-03 12:45:42  INFO      [telnetlib3.client] Connection closed to <Peer 192.168.13.110 2000>
2024-10-03 12:45:42  INFO      [juicebox_mqtthandler] Starting JuiceboxMQTTHandler
2024-10-03 12:45:42  INFO      [juicebox_mitm] Starting JuiceboxMITM
2024-10-03 12:45:42  DEBUG     [juicebox_mitm] JPP: 172.17.0.8:8047
2024-10-03 12:45:42  DEBUG     [juicebox_mitm] EnelX: 158.47.1.128:8047
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Status/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Status/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Current/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Current/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Current-Rating/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Current-Rating/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Frequency/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Frequency/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Energy--Lifetime-/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Energy--Session-/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Energy--Session-/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Temperature/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Temperature/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Voltage/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Voltage/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Power/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Power/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Last-Debug-Message/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Last-Debug-Message/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable.sensors] Setting Last Debug Message to INFO: Starting JuicePass Proxy v0.3.1 using hmd/sensor/JuiceBox/Last-Debug-Message/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Data-from-JuiceBox/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Data-from-JuiceBox/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Data-from-EnelX/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Data-from-EnelX/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/text/JuiceBox/Send-Command-to-JuiceBox/config
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable] state_topic: hmd/text/JuiceBox/Send-Command-to-JuiceBox/state
2024-10-03 12:45:42  INFO      [ha_mqtt_discoverable.sensors] Setting Send Command to JuiceBox to Send Command to JuiceBox using hmd/text/JuiceBox/Send-Command-to-JuiceBox/state
2024-10-03 12:45:47  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop
2024-10-03 12:47:47  WARNING   [juicebox_mitm] No Message Received after 120 sec. (TimeoutError: )
2024-10-03 12:47:47  DEBUG     [juicebox_mitm] Errors in last 60 min: 1
2024-10-03 12:47:47  WARNING   [juicebox_mitm] JuiceboxMITM Reconnecting.
2024-10-03 12:47:47  DEBUG     [juicebox_mitm] Errors in last 60 min: 2
2024-10-03 12:47:53  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop
2024-10-03 12:49:53  WARNING   [juicebox_mitm] No Message Received after 120 sec. (TimeoutError: )
2024-10-03 12:49:53  DEBUG     [juicebox_mitm] Errors in last 60 min: 3
2024-10-03 12:49:53  WARNING   [juicebox_mitm] JuiceboxMITM Reconnecting.
2024-10-03 12:49:53  DEBUG     [juicebox_mitm] Errors in last 60 min: 4
2024-10-03 12:49:58  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop
2024-10-03 12:51:58  WARNING   [juicebox_mitm] No Message Received after 120 sec. (TimeoutError: )
2024-10-03 12:51:58  DEBUG     [juicebox_mitm] Errors in last 60 min: 5
2024-10-03 12:51:58  WARNING   [juicebox_mitm] JuiceboxMITM Reconnecting.
2024-10-03 12:51:58  DEBUG     [juicebox_mitm] Errors in last 60 min: 6
2024-10-03 12:52:03  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop
`
The last one just loops repeatedly.

I've used AdGuard home to fix the DNS of the Juicebox server to my local machine (running unRaid). I added experimental and debugging just in case; some outcome without those. I can telnet into the Juicebox, as it looks like the docker can too.

Maybe the Juicebox just doesn't communicate very frequently?

FalconFour, 2024-10-03:
You say you fixed the DNS to your local machine. You should have two entries - "jbv1.emotorwerks.com" (most likely - will need to check JB2.x firmware) and "directory-api.emotorwerks.com" (ideally direct to localhost so it can't connect to directory API and change the target it's pointing to). It'll always connect to jbv1.emotorwerks.com if the directory server fails.

What model box do you have - metal with or without LEDs (JB1.x) or grey plastic with Enel branding (JB2.x)?

jonlove234, 2024-10-03:
Got it - didn't see those other 2 in the docs (but I might very well have missed/assumed).

So both:
- jbv1.emotorwerks.com
- directory-api.emotorwerks.com

Should be forced to resolve to localhost, or to the machine hosting the proxy?

I've got a grey plastic JB2.X I believe:
`EMWERK-JB201-1.0.46, Gecko_OS-STANDARD-4.2.7-11064, WGM160P`
According to telnet

jonlove234, 2024-10-03:
I tried setting forced DNS per screenshot
[image]

Still no luck. Here's the log:

```2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Status/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Status/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Current/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Current/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Current-Rating/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Current-Rating/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Frequency/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Frequency/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Energy--Lifetime-/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Energy--Session-/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Energy--Session-/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Temperature/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Temperature/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Voltage/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Voltage/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Power/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Power/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Last-Debug-Message/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Last-Debug-Message/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable.sensors] Setting Last Debug Message to INFO: Starting JuicePass Proxy v0.3.1 using hmd/sensor/JuiceBox/Last-Debug-Message/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Data-from-JuiceBox/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Data-from-JuiceBox/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/JuiceBox/Data-from-EnelX/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/JuiceBox/Data-from-EnelX/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/text/JuiceBox/Send-Command-to-JuiceBox/config
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable] state_topic: hmd/text/JuiceBox/Send-Command-to-JuiceBox/state
2024-10-03 14:51:32  INFO      [ha_mqtt_discoverable.sensors] Setting Send Command to JuiceBox to Send Command to JuiceBox using hmd/text/JuiceBox/Send-Command-to-JuiceBox/state
2024-10-03 14:51:37  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop
2024-10-03 14:53:37  WARNING   [juicebox_mitm] No Message Received after 120 sec. (TimeoutError: )
2024-10-03 14:53:37  DEBUG     [juicebox_mitm] Errors in last 60 min: 1
2024-10-03 14:53:37  WARNING   [juicebox_mitm] JuiceboxMITM Reconnecting.
2024-10-03 14:53:37  DEBUG     [juicebox_mitm] Errors in last 60 min: 2
2024-10-03 14:53:42  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop
2024-10-03 14:55:42  WARNING   [juicebox_mitm] No Message Received after 120 sec. (TimeoutError: )
2024-10-03 14:55:42  DEBUG     [juicebox_mitm] Errors in last 60 min: 3
2024-10-03 14:55:42  WARNING   [juicebox_mitm] JuiceboxMITM Reconnecting.
2024-10-03 14:55:42  DEBUG     [juicebox_mitm] Errors in last 60 min: 4
2024-10-03 14:55:47  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop
2024-10-03 14:57:47  WARNING   [juicebox_mitm] No Message Received after 120 sec. (TimeoutError: )
2024-10-03 14:57:47  DEBUG     [juicebox_mitm] Errors in last 60 min: 5
2024-10-03 14:57:47  WARNING   [juicebox_mitm] JuiceboxMITM Reconnecting.
2024-10-03 14:57:47  DEBUG     [juicebox_mitm] Errors in last 60 min: 6
2024-10-03 14:57:52  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop
2024-10-03 14:59:52  WARNING   [juicebox_mitm] No Message Received after 120 sec. (TimeoutError: )
2024-10-03 14:59:52  DEBUG     [juicebox_mitm] Errors in last 60 min: 7
2024-10-03 14:59:52  WARNING   [juicebox_mitm] JuiceboxMITM Reconnecting.
2024-10-03 14:59:52  DEBUG     [juicebox_mitm] Errors in last 60 min: 8
2024-10-03 14:59:57  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop
2024-10-03 15:01:57  WARNING   [juicebox_mitm] No Message Received after 120 sec. (TimeoutError: )
2024-10-03 15:01:57  DEBUG     [juicebox_mitm] Errors in last 60 min: 9
2024-10-03 15:01:57  WARNING   [juicebox_mitm] JuiceboxMITM Reconnecting.
2024-10-03 15:01:57  DEBUG     [juicebox_mitm] Errors in last 60 min: 10
2024-10-03 15:01:57  ERROR     [__main__] A JuicePass Proxy task failed: ChildProcessError: JuiceboxMITM: Unable to start MITM UDP Server.
[... cut ...]
2024-10-03 15:02:07  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/text/JuiceBox/Send-Command-to-JuiceBox/config
2024-10-03 15:02:07  INFO      [ha_mqtt_discoverable] state_topic: hmd/text/JuiceBox/Send-Command-to-JuiceBox/state
2024-10-03 15:02:07  INFO      [ha_mqtt_discoverable.sensors] Setting Send Command to JuiceBox to Send Command to JuiceBox using hmd/text/JuiceBox/Send-Command-to-JuiceBox/state
2024-10-03 15:02:12  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop
```

FalconFour, 2024-10-03:
You got it.
directory-server -> force into the trash can (localhost)
jbv1 -> send to your server (JuicePassProxy)

JPP should listen on port 8042 (was 8047 with the directory-server response, it changes the host and the port). That may be mistaken in your JPP config. Default is 8042.

If you change that port (and possibly power-cycle the JB so it returns to default after setup mode), does it work out?

Protip: if you want to kick the JB out of Setup mode quicker, connect to its AP and issue http://10.10.10.1/command/setup%20stop :)

jonlove234, 2024-10-03:
So I've set the JPP port to be 8042, but still doesn't seem to be working and for some reason the Docker, when run, still shows 8047?

[image]

```2024-10-03 19:38:31  INFO      [entrypoint.sh] Starting JuicePass Proxy

2024-10-03 19:38:31  INFO      [entrypoint.sh] Docker Environment Variables:
2024-10-03 19:38:31  INFO      [entrypoint.sh] JUICEBOX_HOST: [redacted in case sensitive]
2024-10-03 19:38:31  INFO      [entrypoint.sh] MQTT_HOST: 192.168.13.59
2024-10-03 19:38:31  INFO      [entrypoint.sh] MQTT_PORT: 1883
2024-10-03 19:38:31  INFO      [entrypoint.sh] MQTT_USER: [redacted in case sensitive]
2024-10-03 19:38:31  INFO      [entrypoint.sh] MQTT_PASS: ****************************************************************
2024-10-03 19:38:31  INFO      [entrypoint.sh] JPP_HOST: [redacted in case sensitive]
2024-10-03 19:38:31  INFO      [entrypoint.sh] DEBUG: true
2024-10-03 19:38:31  INFO      [entrypoint.sh] UPDATE_UDPC: false
2024-10-03 19:38:31  INFO      [entrypoint.sh] IGNORE_ENELX: true
2024-10-03 19:38:31  INFO      [entrypoint.sh] EXPERIMENTAL: true
2024-10-03 19:38:31  DEBUG     [entrypoint.sh] COMMAND: python3 /juicepassproxy/juicepassproxy.py --juicebox_host 192.168.13.110 --mqtt_host 192.168.13.59 --mqtt_port 1883 --mqtt_user addons --mqtt_password ***** --juicepass_proxy_host 192.168.1.106 --config_loc /config --log_loc /log --debug --ignore_enelx --experimental
2024-10-03 19:38:31  WARNING   [__main__] Starting JuicePass Proxy v0.3.1 (Log Level: DEBUG)
2024-10-03 19:38:31  INFO      [__main__] log_loc: /log/juicepassproxy.log
2024-10-03 19:38:31  INFO      [__main__] config_loc: /config/juicepassproxy.yaml
2024-10-03 19:38:31  INFO      [__main__] telnet timeout: 30
2024-10-03 19:38:31  INFO      [__main__] ignore_enelx: True
2024-10-03 19:38:31  INFO      [__main__] enelx_server: juicenet-udp-prod3-usa.enelx.com
2024-10-03 19:38:31  INFO      [__main__] enelx_port: 8047
2024-10-03 19:38:31  INFO      [__main__] local_addr: 172.17.0.8:8047
2024-10-03 19:38:31  INFO      [__main__] enelx_addr: 158.47.1.128:8047
2024-10-03 19:38:32  ERROR     [aiorun] Unhandled exception; stopping loop: 'Task exception was never retrieved'
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiorun.py", line 219, in new_coro
    await coro
  File "/juicepassproxy/juicepassproxy.py", line 483, in main
    elif juicebox_id := await get_juicebox_id(
                        ^^^^^^^^^^^^^^^^^^^^^^
  File "/juicepassproxy/juicepassproxy.py", line 156, in get_juicebox_id
    async with JuiceboxTelnet(
  File "/juicepassproxy/juicebox_telnet.py", line 20, in __aenter__
    if await self.open():
       ^^^^^^^^^^^^^^^^^
  File "/juicepassproxy/juicebox_telnet.py", line 60, in open
    self.reader, self.writer = await telnetlib3.open_connection(
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/telnetlib3/client.py", line 352, in open_connection
    transport, protocol = await asyncio.get_event_loop().create_connection(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1122, in create_connection
    raise exceptions[0]
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1104, in create_connection
    sock = await self._connect_sock(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1007, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 651, in sock_connect
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 691, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
OSError: [Errno 113] Connect call failed ('192.168.13.110', 2000)
2024-10-03 19:38:32  INFO      [aiorun] Entering shutdown phase.
2024-10-03 19:38:32  INFO      [aiorun] Waiting for executor shutdown.
2024-10-03 19:38:32  INFO      [aiorun] Shutting down async generators
2024-10-03 19:38:32  INFO      [aiorun] Closing the loop.
2024-10-03 19:38:32  INFO      [aiorun] Leaving. Bye!
2024-10-03 19:38:32  INFO      [aiorun] Reraising unhandled exception
Traceback (most recent call last):
  File "/juicepassproxy/juicepassproxy.py", line 586, in <module>
    run(main(), stop_on_unhandled_errors=True)
  File "/usr/local/lib/python3.12/site-packages/aiorun.py", line 370, in run
    raise pending_exception_to_raise
  File "/usr/local/lib/python3.12/site-packages/aiorun.py", line 219, in new_coro
[... 4 lines cut ...]
  File "/juicepassproxy/juicepassproxy.py", line 156, in get_juicebox_id
    async with JuiceboxTelnet(
  File "/juicepassproxy/juicebox_telnet.py", line 20, in __aenter__
    if await self.open():
       ^^^^^^^^^^^^^^^^^
  File "/juicepassproxy/juicebox_telnet.py", line 60, in open
    self.reader, self.writer = await telnetlib3.open_connection(
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/telnetlib3/client.py", line 352, in open_connection
    transport, protocol = await asyncio.get_event_loop().create_connection(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1122, in create_connection
    raise exceptions[0]
[... cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1007, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 651, in sock_connect
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 691, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
OSError: [Errno 113] Connect call failed ('192.168.13.110', 2000)
```

jonlove234, 2024-10-03:
Oh never mind on the error - seems like something is happening to the connection that it doesn't like as I can't telnet to it either. But is that port setup correct?

FalconFour, 2024-10-04:
Odd, yeah it still seems to be listening on 8047. Telnet should be 2000, yep. Maybe your JB's IP address drifted between boots? That's another thing I'm worried about, having to wrestle with keeping hands on that slippery address...

jonlove234, 2024-10-04:
> Odd, yeah it still seems to be listening on 8047 instead of 8042. Telnet

It's still on that IP (according to my router), so not sure why it became
non-responsive but will reboot/play around and see.

wozz, 2024-10-05:
I seem to be having very similar results:

I get the same errors just looping if I enable the ignore_enelx setting. If I disable the ignore_enelx setting, things appear to work as before.

wozz, 2024-10-05:
It looks like the port was wrong for me. I was using the telnet based inspection to determine the port, which was using 8047. However it appears to actually still be using 8042.

jonlove234, 2024-10-05:
Did you work out how to adjust the port? Mine seems stuck no matter what
the docker port config is

wozz, 2024-10-05:
I'm using HA add-on config. The setup for container port is here: https://github.com/wozz/ha-addons/blob/main/juicepassproxy/config.yaml#L7

I override the port inside the container to 8042: https://github.com/wozz/ha-addons/blob/main/juicepassproxy/run.sh#L11

And then I can just adjust the port exposed by the container in the HA config (although if you're using docker directly, you can change it in the docker run flags).

jonlove234, 2024-10-07:
Tried a new setup, and still running into problems. The DNS is "definitely" working as:

```> list
list
! # Type  Info
# 1 FILE  godaddy_ca.pem-1.0.0 (1390, 0)
# 2 UDPC  juicenet-udp-prod3-usa.enelx.com:8047 (6536)
> nlo juicenet-udp-prod3-usa.enelx.com
nlo juicenet-udp-prod3-usa.enelx.com
192.168.1.115
```

[image]

No matter whether the docker (now using the HA add-on from wozz) is using port 8047 or 8042, still not seeing anything:

```2024-10-07 10:06:49  INFO      [entrypoint.sh] Starting JuicePass Proxy

2024-10-07 10:06:49  INFO      [entrypoint.sh] Docker Environment Variables:
2024-10-07 10:06:49  INFO      [entrypoint.sh] DEVICE_NAME: gecko_os-926
2024-10-07 10:06:49  INFO      [entrypoint.sh] JUICEBOX_HOST: 192.168.1.222
2024-10-07 10:06:49  INFO      [entrypoint.sh] LOCAL_PORT: 8042
2024-10-07 10:06:49  INFO      [entrypoint.sh] MQTT_HOST: core-mosquitto
2024-10-07 10:06:49  INFO      [entrypoint.sh] MQTT_USER: addons
2024-10-07 10:06:49  INFO      [entrypoint.sh] MQTT_PASS: ****************************************************************
2024-10-07 10:06:49  INFO      [entrypoint.sh] DEBUG: true
2024-10-07 10:06:49  INFO      [entrypoint.sh] UPDATE_UDPC: false
2024-10-07 10:06:49  INFO      [entrypoint.sh] IGNORE_ENELX: false
2024-10-07 10:06:49  INFO      [entrypoint.sh] EXPERIMENTAL: false
2024-10-07 10:06:49  DEBUG     [entrypoint.sh] COMMAND: python3 /juicepassproxy/juicepassproxy.py --name gecko_os-926 --juicebox_host 192.168.1.222 --local_port 8042 --mqtt_host core-mosquitto --mqtt_user addons --mqtt_password ***** --config_loc /config --log_loc /log --debug
2024-10-07 10:06:49  WARNING   [__main__] Starting JuicePass Proxy v0.3.1 (Log Level: DEBUG)
2024-10-07 10:06:49  INFO      [__main__] log_loc: /log/juicepassproxy.log
2024-10-07 10:06:49  INFO      [__main__] config_loc: /config/juicepassproxy.yaml
2024-10-07 10:06:49  INFO      [__main__] telnet timeout: 30
2024-10-07 10:06:49  INFO      [__main__] ignore_enelx: False
2024-10-07 10:06:49  INFO      [telnetlib3.client] Connected to <Peer 192.168.1.222 2000>
2024-10-07 10:06:53  DEBUG     [__main__] enelx_server_port: juicenet-udp-prod3-usa.enelx.com:8047
2024-10-07 10:06:53  INFO      [__main__] enelx_server: juicenet-udp-prod3-usa.enelx.com
2024-10-07 10:06:53  INFO      [__main__] enelx_port: 8047
2024-10-07 10:06:53  INFO      [telnetlib3.client] Connection closed to <Peer 192.168.1.222 2000>
2024-10-07 10:06:53  INFO      [__main__] local_addr: 172.30.33.2:8042
2024-10-07 10:06:53  INFO      [__main__] enelx_addr: 158.47.1.128:8047
2024-10-07 10:06:53  INFO      [telnetlib3.client] Connected to <Peer 192.168.1.222 2000>
2024-10-07 10:06:55  INFO      [__main__] juicebox_id: 0910042001260546348022626331
2024-10-07 10:06:55  INFO      [__main__] experimental: False
2024-10-07 10:06:55  INFO      [__main__] max_current: 48
2024-10-07 10:06:55  INFO      [telnetlib3.client] Connection closed to <Peer 192.168.1.222 2000>
2024-10-07 10:06:55  INFO      [juicebox_mqtthandler] Starting JuiceboxMQTTHandler
2024-10-07 10:06:55  INFO      [juicebox_mitm] Starting JuiceboxMITM
2024-10-07 10:06:55  DEBUG     [juicebox_mitm] JPP: 172.30.33.2:8042
2024-10-07 10:06:55  DEBUG     [juicebox_mitm] EnelX: 158.47.1.128:8047
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/gecko_os-926/Status/config
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/gecko_os-926/Status/state
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/gecko_os-926/Current/config
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/gecko_os-926/Current/state
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/gecko_os-926/Current-Rating/config
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/gecko_os-926/Current-Rating/state
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/number/gecko_os-926/Max-Current/config
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] state_topic: hmd/number/gecko_os-926/Max-Current/state
2024-10-07 10:06:55  WARNING   [juicebox_mqtthandler] Max Current has no initial_state
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/number/gecko_os-926/Max-Charging-Current/config
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] state_topic: hmd/number/gecko_os-926/Max-Charging-Current/state
2024-10-07 10:06:55  WARNING   [juicebox_mqtthandler] Max Charging Current has no initial_state
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/gecko_os-926/Frequency/config
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/gecko_os-926/Frequency/state
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/gecko_os-926/Energy--Lifetime-/config
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/gecko_os-926/Energy--Lifetime-/state
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/gecko_os-926/Energy--Session-/config
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/gecko_os-926/Energy--Session-/state
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/gecko_os-926/Temperature/config
[... cut ...]
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/gecko_os-926/Last-Debug-Message/config
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/gecko_os-926/Last-Debug-Message/state
2024-10-07 10:06:55  INFO      [ha_mqtt_discoverable.sensors] Setting Last Debug Message to INFO: Starting JuicePass Proxy v0.3.1 using hmd/sensor/gecko_os-926/Last-Debug-Message/state
2024-10-07 10:07:00  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop```

FalconFour, 2024-10-07:
Chasing ports back and forth. Now that you're using the DNS approach, it performs a directory server lookup and the result tells the box to use port 8047 instead of 8042.
```
# 2 UDPC  juicenet-udp-prod3-usa.enelx.com:8047 (6536)
```

And the server is now listening on 8042:
```
2024-10-07 10:06:49  INFO      [entrypoint.sh] LOCAL_PORT: 8042
```

Definitely think the ideal path for the project is a man-in-the-middle WiFi device, at least given the info we have found at the moment. DNS spoofing is a pain in the rear, directory server spoofing seems to only work for half the boxes (option to ignore cert errors is only in JB2.0), redirecting UDP traffic is a somewhat niche router option, and swapping the handle (UPDATE_UDPC in Telnet) seems to break incoming connections.

But I think you're nearly at the finish line - having achieved the DNS spoof, just one more hurdle to clear and you're home free!

FalconFour, 2024-10-07:
Whichever the box is listening on. Now that you have redirected `juicenet-udp-prod3-usa.enelx.com` to your server for sure, you can see that the box is connecting on `# 2 UDPC  juicenet-udp-prod3-usa.enelx.com:8047 (6536)`.

The box connect to `jbv1.emotorwerks.com` on port 8042 by default, so it's ... less than easy to say a sure thing at the moment. Thus now that we have the solid confirmation that (a) it's using the directory server response, and (b) that response is pointing to that server on port 8047, then you ought to set the server to use port 8047 :)

Must admit it's anything but straightforward at the moment.

jonlove234, 2024-10-07:
> (not the defaults), and (b) that response is pointing to that server on

OK so still in a loop, even using port 8047:

[image: image.png]

2024-10-07 13:38:47  INFO      [entrypoint.sh] Starting JuicePass Proxy

2024-10-07 13:38:47  INFO      [entrypoint.sh] Docker Environment Variables:
2024-10-07 13:38:47  INFO      [entrypoint.sh] DEVICE_NAME: gecko_os-926
2024-10-07 13:38:47  INFO      [entrypoint.sh] JUICEBOX_HOST: 192.168.1.222
2024-10-07 13:38:47  INFO      [entrypoint.sh] LOCAL_PORT: 8042
2024-10-07 13:38:47  INFO      [entrypoint.sh] MQTT_HOST: core-mosquitto
2024-10-07 13:38:47  INFO      [entrypoint.sh] MQTT_USER: addons
2024-10-07 13:38:47  INFO      [entrypoint.sh] MQTT_PASS:
****************************************************************
2024-10-07 13:38:47  INFO      [entrypoint.sh] DEBUG: true
2024-10-07 13:38:47  INFO      [entrypoint.sh] UPDATE_UDPC: false
2024-10-07 13:38:47  INFO      [entrypoint.sh] IGNORE_ENELX: false
2024-10-07 13:38:47  INFO      [entrypoint.sh] EXPERIMENTAL: false
2024-10-07 13:38:47  DEBUG     [entrypoint.sh] COMMAND: python3
/juicepassproxy/juicepassproxy.py --name gecko_os-926 --juicebox_host
192.168.1.222 --local_port 8042 --mqtt_host core-mosquitto --mqtt_user
addons --mqtt_password ***** --config_loc /config --log_loc /log
--debug
2024-10-07 13:38:48  WARNING   [__main__] Starting JuicePass Proxy
v0.3.1 (Log Level: DEBUG)
2024-10-07 13:38:48  INFO      [__main__] log_loc: /log/juicepassproxy.log
2024-10-07 13:38:48  INFO      [__main__] config_loc:
/config/juicepassproxy.yaml
2024-10-07 13:38:48  INFO      [__main__] telnet timeout: 30
2024-10-07 13:38:48  INFO      [__main__] ignore_enelx: False
2024-10-07 13:38:48  INFO      [telnetlib3.client] Connected to <Peer
192.168.1.222 2000>
2024-10-07 13:38:50  DEBUG     [__main__] enelx_server_port:
juicenet-udp-prod3-usa.enelx.com:8047
2024-10-07 13:38:50  INFO      [__main__] enelx_server:
juicenet-udp-prod3-usa.enelx.com
2024-10-07 13:38:50  INFO      [__main__] enelx_port: 8047
2024-10-07 13:38:50  INFO      [telnetlib3.client] Connection closed
to <Peer 192.168.1.222 2000>
2024-10-07 13:38:50  INFO      [__main__] local_addr: 172.30.33.2:8042
2024-10-07 13:38:50  INFO      [__main__] enelx_addr: 158.47.1.128:8047
2024-10-07 13:38:50  INFO      [telnetlib3.client] Connected to <Peer
192.168.1.222 2000>
2024-10-07 13:38:52  INFO      [__main__] juicebox_id:
0910042001260546348022626331
2024-10-07 13:38:52  INFO      [__main__] experimental: False
2024-10-07 13:38:52  INFO      [__main__] max_current: 48
2024-10-07 13:38:52  INFO      [telnetlib3.client] Connection closed
to <Peer 192.168.1.222 2000>
2024-10-07 13:38:52  INFO      [juicebox_mqtthandler] Starting
JuiceboxMQTTHandler
2024-10-07 13:38:52  INFO      [juicebox_mitm] Starting JuiceboxMITM
2024-10-07 13:38:52  DEBUG     [juicebox_mitm] JPP: 172.30.33.2:8042
2024-10-07 13:38:52  DEBUG     [juicebox_mitm] EnelX: 158.47.1.128:8047
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] config_topic:
homeassistant/sensor/gecko_os-926/Status/config
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] state_topic:
hmd/sensor/gecko_os-926/Status/state
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] config_topic:
homeassistant/sensor/gecko_os-926/Current/config
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] state_topic:
hmd/sensor/gecko_os-926/Current/state
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] config_topic:
homeassistant/sensor/gecko_os-926/Current-Rating/config
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] state_topic:
hmd/sensor/gecko_os-926/Current-Rating/state
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] config_topic:
homeassistant/number/gecko_os-926/Max-Current/config
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] state_topic:
hmd/number/gecko_os-926/Max-Current/state
[... 29 lines cut ...]
homeassistant/sensor/gecko_os-926/Power/config
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] state_topic:
hmd/sensor/gecko_os-926/Power/state
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] config_topic:
homeassistant/switch/gecko_os-926/Act-as-Server/config
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] state_topic:
hmd/switch/gecko_os-926/Act-as-Server/state
2024-10-07 13:38:52  DEBUG     [juicebox_mqtthandler] Errors in last 60 min: 1
2024-10-07 13:38:52  WARNING   [juicebox_mqtthandler] Can't update
attributes for Act as Server as MQTT isn't connected/started.
(AttributeError: 'Switch' object has no attribute 'update_state')
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] config_topic:
homeassistant/sensor/gecko_os-926/Last-Debug-Message/config
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable] state_topic:
hmd/sensor/gecko_os-926/Last-Debug-Message/state
2024-10-07 13:38:52  INFO      [ha_mqtt_discoverable.sensors] Setting
Last Debug Message to INFO: Starting JuicePass Proxy v0.3.1 using
hmd/sensor/gecko_os-926/Last-Debug-Message/state
2024-10-07 13:38:57  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop
2024-10-07 13:40:57  WARNING   [juicebox_mitm] No Message Received
after 120 sec. (TimeoutError: )
2024-10-07 13:40:57  DEBUG     [juicebox_mitm] Errors in last 60 min: 1
2024-10-07 13:40:57  WARNING   [juicebox_mitm] JuiceboxMITM Reconnecting.
2024-10-07 13:40:57  DEBUG     [juicebox_mitm] Errors in last 60 min: 2
2024-10-07 13:41:02  DEBUG     [juicebox_mitm] Starting JuiceboxMITM Loop

ivanfmartinez, 2024-10-07:
> 2024-10-07 13:38:52 DEBUG [juicebox_mitm] JPP: 172.30.33.2:8042

If this is the IP of your docker container you have to specify the JPP_HOST parameter with the real IP of the machine where  you are running the container.

https://github.com/JuiceRescue/juicepassproxy?tab=readme-ov-file#docker-environment-variables

ivanfmartinez, 2024-10-07:
> OK so still in a loop, even using port 8047:

the logs indicate that you are still using the 8042 port on JPP.

jonlove234, 2024-10-07:
OK, I finally got it working! I have to run mine outside of HA and for some
reason it works fine in host mode but not in bridge mode. I'm not sure why,
but it's working, so I won't do too much more exploration:

2024-10-07 14:50:11  INFO      [ha_mqtt_discoverable.sensors] Setting
Current to 0 using hmd/sensor/JuiceBox/Current/state
2024-10-07 14:50:11  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Session) to 0 using hmd/sensor/JuiceBox/Energy--Session-/state
2024-10-07 14:50:11  INFO      [ha_mqtt_discoverable.sensors] Setting
Voltage to 245.9 using hmd/sensor/JuiceBox/Voltage/state
2024-10-07 14:50:11  INFO      [ha_mqtt_discoverable.sensors] Setting
Energy (Lifetime) to 9680591.0 using
hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-10-07 14:50:11  INFO      [ha_mqtt_discoverable.sensors] Setting
Status to Unplugged using hmd/sensor/JuiceBox/Status/state
2024-10-07 14:50:11  INFO      [ha_mqtt_discoverable.sensors] Setting
Temperature to 82.4 using hmd/sensor/JuiceBox/Temperature/state
2024-10-07 14:50:11  INFO      [ha_mqtt_discoverable.sensors] Setting
Current Rating to 40.0 using hmd/sensor/JuiceBox/Current-Rating/state
2024-10-07 14:50:11  INFO      [ha_mqtt_discoverable.sensors] Setting
Frequency to 60.01 using hmd/sensor/JuiceBox/Frequency/state
2024-10-07 14:50:11  INFO      [ha_mqtt_discoverable.sensors] Setting Power
to 0.0 using hmd/sensor/JuiceBox/Power/state

Thanks all for the help!

atc99, 2024-10-17:
It looks like there were some questions in this issue about whether DNS was working correctly.  You can validate that DNS is being overridden and resolved correctly on the JB by using the network_lookup (aka nlo) command from the telnet prompt and verfiying that the IP it displays is correct.

```
> network_lookup jvb1.emotorwerks.com
network_lookup jvb1.emotorwerks.com
192.168.0.18

> nlo directory-api.emotorwerks.com
nlo directory-api.emotorwerks.com
192.168.0.18

> nlo juicenet-udp-prod3-usa.enelx.com
nlo juicenet-udp-prod3-usa.enelx.com
192.168.0.18
```

closed 2024-10-19 by ivanfmartinez