JuiceRescue/juicepassproxy #92: Organizing Meeting Sat Oct 5 12:30pm ET

natematias, 2024-10-04:
Hi everyone,

We have a meeting time! It's 12:30pm on Saturday, October 5th. Thanks to @FalconFour for registering interest. If any of the other maintainers are able to join, I would value your input, if you're available! cc @Snuffy2, @snicker 

I have updates; I have spoken to Consumer Reports, folks at the New York State Energy Research and Development Authority, and some lawyers who can advise on the IP questions. I'm feeling hopeful about the potential to apply pressure to keep the lights on in the short term, as well as potentially even open up some of the IP in the longer term. I can share more details in the call. 

Details below- since I'm putting this on the public internet, I'm creating a waiting room. So if you plan to attend, please reply to this thread with who you are so I can admit you into the room.

J. Nathan Matias is inviting you to a scheduled Zoom meeting.

Topic: JuicepassProxy Meeting
Time: Oct 5, 2024 12:30 PM Eastern Time (US and Canada)

Join Zoom Meeting
https://cornell.zoom.us/j/98589865438?pwd=7ELCbifHnSCEyRi8PCspCwYyY6XDg9.1

Meeting ID: 985 8986 5438
Passcode: 075522

---

One tap mobile
+16465189805,,98589865438# US (New York)
+16468769923,,98589865438# US (New York)

---

Dial by your location
• +1 646 518 9805 US (New York)
• +1 646 876 9923 US (New York)
• +1 470 250 9358 US (Atlanta)
• +1 470 381 2552 US (Atlanta)
• +1 651 372 8299 US (Minnesota)
• +1 786 635 1003 US (Miami)
• +1 301 715 8592 US (Washington DC)
• +1 312 626 6799 US (Chicago)
• +1 971 247 1195 US (Portland)
• +1 253 215 8782 US (Tacoma)
• +1 346 248 7799 US (Houston)
• +1 602 753 0140 US (Phoenix)
• +1 669 219 2599 US (San Jose)
• +1 669 900 6833 US (San Jose)
• +1 720 928 9299 US (Denver)

Meeting ID: 985 8986 5438

Find your local number: https://cornell.zoom.us/u/abZ7tAOtbh

pulse-aj, 2024-10-04:
I will be attending this one. Thank you for organising it. We have have a community of home charging users with old versions of Juicebox that's running UDP and new ones thats run OCPP. So I am interested in know which year models are on UDP and which ones are on OCPP.

snicker, 2024-10-05:
I will try to attend! if not.. 

I am very interested in seeing the platform opened up for development, even if there is not a release for IP, at least a release for product owners to gain access to installing their own firmware, if developed.

I believe @falconfour mentioned that at the heart of several Juicebox products is just an ATMega, meaning there is plenty of opportunity to take control of the hardware if the appropriate agreements are reached and (i'd imagine) complete indemnity of Enel if a hardware owner chooses to take that path.

Full disclosure, and probably not well known within this space: while I am the owner of a Juicebox (among other EVSEs), I am employed by a fairly large entity in the EV and EVSE space, with direct responsibility in those market segments (I'll leave it at that, it wouldn't take much sleuthing to find the remaining details). I am most motivated here to ensure that the sudden loss of support and functionality of these products from the OEM is not a damaging setback to EV owners the electrification of our planet as a whole.

FalconFour, 2024-10-05:
Same position as @snicker myself. It's a bit of a balancing act but I don't think there's an argument for a conflict of interest when we're not selling anything. The same can't be said if I start helping out someone for profit, I think. My primary drive is to keep the unnecessary E-waste to a minimum - but of course similarly driven to keep EVs enjoyable despite issues of earlier hardware.

It is just an ATMega, but it's a massively complex firmware - stretching that chip to its absolute limits - having to use clever optimization to keep it within 32KB is like going back to NES development days. Using every single I/O pin and nearly every subsystem. The main focus of opening up the IP, though, should be on the Zentri(/Gecko) platform doing the communication - the core processor (Atmel) is pretty well-known and stable and doesn't really stand in the way of bringing about an open platform. There's also never been any OCPP capability in the box (it was all emulated by the Enel platform in the cloud), so... that's a factor to consider for connecting to other platforms.

Anyway, yep, I'm on board for the call! Let's make it happen.

eliotwinchell, 2024-10-05:
I will be in attendance! I'm a bit confused regarding IP, is there a legal risk if the company which owns the IP is shutting down entirely? I didn't think there was, unless they transferred it to another entity which doesn't appear to be the case... @natematias

ivanfmartinez, 2024-10-05:
I dont know if I can attend but here are some points to consider :

- ENELX USA and Canada is shutting down, but ENELX still an european company, and that still in other contries (including mine Brazil). 
    - The information published indicate that they will still work in other countries. Considering this, maybe even the servers will still working. 
    -  The server name that my device used and others still use have "USA" in the name, but the ASN is related do ENEL ITALY.

- I'm a programmer
    -  I just want to control my device. That is the need for most users. 
    -  even having possibility to change firmware  I dont want to do that because will give a lot of work. 
        - creating a new project to make new firmware will need a lot of personal time, that I dont think is really necessary at this moment
        - the current protocol is not the best but works
        - I my opinion the main focus should be now to can keep devices working inside our home networks using current protocol and firmware
    -  If they publish documentation that make us know the behaviour of all versions to respond correctly I think will be very nice for users around the world not only for USA/CA users.

- My v09u device are working very fine with my branch
    - but now others are trying to use and are different behaviour that must be tested to adjust the code. 
    - If more people can do the tests we can improve the code to work with all devices that send unencrypted messages with the information and devices that we have now.
    - The most difficult part was done by many people that discovered the parameters on the device and made juicepassproxy possible
    - This year I just connected some parts to make juicepassproxy respond to the device and this solved my need that was to control the charging current on device.

- The users in USA/CA that does not have the APP and reset their devices will need a way/app/scripts to connect their devices in factory state to connect on wifi.

wozz, 2024-10-05:
I will try to attend the meeting.

wtadler, 2024-10-05:
I would like to join but probably can't, so a readout of the call would be much appreciated!

natematias, 2024-10-05:
Thanks all! I'll make sure to post notes to this thread afterward.

Reminder that the conversation starts in just about 5 minutes. Here's a
rough agenda we can adjust as needed:

   -

   (5 mins) Intros & review agenda
   -

   10 minutes: summarizing the state of play with Enel X and Juicebox
   -

   10 mins: State of the project and future possibilities (ask folks who
   are maintainers)
   -

   25 mins: Discussing what needs to be done next:
   -

   10 mins: Agreeing on next steps

See you soon!

-- 
J. Nathan Matias <http://natematias.com/> : Cornell University : Citizens
and Technology Lab <https://citizensandtech.org>  : Coalition for
Independent Technology Research <http://independenttechresearch.org/> :
***@***.*** : blogs
<https://natematias.com/external-posts/> : daylight
time photos <https://social.coop/tags/outdoormoments>

natematias, 2024-10-05:
Hi all,

We had a *great* conversation, with multiple folks joining, including @wtadler, @FalconFour @pulse-aj, @eliotwinchell, and @wozz. 

Overall, we got a great sense of the range of needs and goals that people have, including individual users, utilities / government agencies that told people to buy juiceboxes, and EV charging network app producers, who want to support compatibility layers to theirs and other platforms. All of us learned something new about the needs, and opportunities— and it was a very encouraging conversation!

Next steps:
- We have created **a Discord called "Juicebox Rescue"** here: https://discord.gg/rBgbGZsA. Please join!
- Later today, I'm going to create a webpage with a sign-up and survey that points people to different Juicebox resources and invites people to follow and help. We're hoping this will help us connect with some of the energy companies and government energy units that might be especially keen to see solutions introduced
- We're envisioning effort on several fronts:
  - making it easier for non-technical people to set up juicepassproxy, so they can monitor/control, and connect to Home Assistant
  - making it easier for utilities and other organizations to bridge between juiceboxes and [OCPP](https://en.wikipedia.org/wiki/Open_Charge_Point_Protocol)
  - supporting ongoing maintenance of the hardware (we discussed this less, but @FalconFour brought it up, and it seems quite important)
  - organizing to understand and navigate the ongoing security/safety risks

I'm also going to reach out to the original developers of juicepassproxy to make sure we're including them in the conversation— which is pretty essential going forward.

[Full notes can be found here](https://docs.google.com/document/d/1Eot_P6rix8wCZwC9ZwnET8hQzCUm9FmPNIaLT6OC1Ho/edit)

closed 2025-01-20 by snicker