JuiceRescue/juicepassproxy #93: Telnet Connection Refused

dankvitek, 2024-10-05:
I was running juicepassproxy successfully 2/24-7/24, harvesting data for HA, but in July started getting connection refused from telnet.
```
Trying 10.26.20.33...
telnet: connect to address 10.26.20.33: Connection refused
telnet: Unable to connect to remote host
```
Ping still works fine, so I don't think it's a general routing issue to the Juicebox.

I tried a factory reset, but still no dice with telnet. The Juicebox local webserver console has this to say:
```
EnelX Web App Console - v3.1.5
> list
! # Type  Info
# 1 FILE  godaddy_ca.pem-1.0.0 (1390, 0)
> get ud
udp.client.auto_interface: default
udp.client.auto_retry: 0
udp.client.auto_start: 0
udp.client.local_port: 0
udp.client.remote_host: 
udp.client.remote_port: 3000
udp.server.auto_interface: default
udp.server.auto_start: 0
udp.server.data_gpio: -1
udp.server.lock_client: 0
udp.server.port: 3000
udp.server.remote_host: 
udp.server.remote_port: 0
```
This looks quite different than any other config I've seen. Note, my Juicebox has been blocked from accessing the internet since before the reset.

Any ideas on how to get the Juicebox back to a place where it can be used with juicepassproxy again?

FalconFour, 2024-10-05:
How odd, "EnelX web app console"? Hm.

Don't worry about the UDP client parameters, it doesn't use those in the ZAP system. Given that customized console string, I have zero doubt it's running ZAP firmware. Also curious you have a web console - that was removed/disabled long ago to save space. Maybe the WGM* module in JB2.x has those enabled?

Anyway, here are the relevant parameters:
```
running: JB1_0.zap
> ver
EMWERK-JB_1_1-1.4.0.28, 2021-04-27T20:39:50Z, ZentriOS-WZ-3.6.4.0
> get re
remote_terminal.enabled: 1
remote_terminal.interface: default
remote_terminal.password:
remote_terminal.port: 2000
remote_terminal.timeout: 60
>
```
Check those on yours.

dankvitek, 2024-10-07:
Thanks for the response Falcon, much appreciated. Looks like the remote_terminal option is not enabled on my box, which probably explains the inability to telnet into it.
```
EnelX Web App Console - v3.1.5
loading commands...
loading variables...
Ready.
> ver
EMWERK-JB201_V2-1.0.69, Gecko_OS-STANDARD-4.2.7-11064, WGM160P
> get re
remote_terminal.enabled: 0
remote_terminal.interface: default
remote_terminal.password: 
remote_terminal.port: 2000
remote_terminal.timeout: 60
> set remote_terminal.enabled 1
Set OK
> save
Success
> get re
remote_terminal.enabled: 1
remote_terminal.interface: default
remote_terminal.password: 
remote_terminal.port: 2000
remote_terminal.timeout: 60
```
This works, and I am able to telnet into the box. Cool!

However, after some time it starts refusing the telnet connection again, and when I log back into the console, I see that is has reverted back to `remote_terminal.enabled: 0` on it's own accord.

Is there scheduled code running inside the box that resets this value? Like I said above, the box is no longer connected to the internet.

FalconFour, 2024-10-07:
Agh... great. Well, the only way to know for sure would be to see if there's a recent firmware update posted to the 2.x boxes. ZAP twiddles a few settings on startup, and if so, it'd be really awful of them to have pushed out an update like that. It does those settings-twiddles on startup, and there are a handful of conditions (other than power cycle) that trigger a WiFi reboot, which would reload those forced settings. In my box, DHCP is one of them! So, no static IP will stick past a reboot...

Suppose that's on my goals list for next time I'm in the shop - fire up a JB2.0 board and see if I can get eyes on the latest firmware and its build date.

dankvitek, 2024-10-08:
Looking back at my HA data from jpp-- looks like the JB stopped playing nicely with jpp around June 5th. I had the JB connected to the internet until about a week ago. So it is possible a firmware update caused this.

If a firmware update really did brick the remote terminal on the JB, is the only way to recover from this to re-flash the firmware? I'm guessing OTA updates are not a thing.

Let me know if there is any other diagnostic information that would help troubleshooting this issue. Thanks.

FalconFour, 2024-10-29:
European JuiceBox Gen3? That's what the version string suggests. There's a whole huge number of unknowns there... you might have the entirely-new box that has very little if any relation to the JuiceBox as we know it, @tomayac :) Might be nice to pop it open and get photos to document it, but ... you may be blessed by native OCPP capability in the end, anyway!

tomayac, 2024-10-30:
> European JuiceBox Gen3? That's what the version string suggests. There's a whole huge number of unknowns there... you might have the entirely-new box that has very little if any relation to the [...]

Not sure how other JuiceBoxen look like, but this is mine:

[image]

FalconFour, 2024-10-31:
That looks so vastly different, it almost looks like it's made by a different company (and maybe it was) 😆

tomayac, 2024-10-31:
Yeah, not sure where this leaves folks with such devices. It looks like juicepassproxy isn't necessarily going to be working without some changes for these models.

snicker, 2025-01-20:
re-labeling issue