JuiceRescue/juicepassproxy #94: juicebox_commands: "Unable to send command to juicebox before current_max and current_max_charging values are set"

FalconFour, 2024-10-05:
Simple issue:
"current_max" and "current_max_charging" should default to the `current_rating` (m80 in my case - the max field value is 80, and sending an out-of-bounds value will not cause issues; the system has checks to not exceed its unit rating deep internally) when the variable is not otherwise set. A command response **must** be sent to **every** incoming telemetry string, so there can't be any reason it won't send a reply.

Seems to be caused by this:
```
    # Send a new message using values on mqtt entities
    async def send_cmd_message_to_juicebox(self, new_values):
       
       if self._mqtt_handler.get_entity("act_as_server").is_on():

          # Max current values must be define to send messages to device
          if self.is_mqtt_numeric_entity_defined("current_max") and self.is_mqtt_numeric_entity_defined("current_max_charging"):
              cmd_message = self.__build_cmd_message(new_values)
              if cmd_message:
                  _LOGGER.info(f"Sending command to juicebox {cmd_message} new_values={new_values}")
                  await self.send_data(cmd_message.encode('utf-8'), self._juicebox_addr)
          else:
              _LOGGER.warn("Unable to send command to juicebox before current_max and current_max_charging values are set") 
```
I'm having trouble following the process flow around to fix this - we would want to "else" that, if not defined, the value `current_rating` should be inserted into both values and then proceed normally.

Additionally, only `number.juicebox_max_charging_current` seems to be published to my HA dashboard, so there seems to be no way to satisfy this.

ivanfmartinez, 2024-10-05:
I selected to only send back the value that are ALREADY configured on the unit. Thats why I only send the command after the unit values are know.

Considering that are many devices with different currents, the basic code dont know what device will be communicating. The sequence that things happens now in juicepassproxy needs for example that the Entityes are created BEFORE the first message came from the juicebox. For this happen and set the max value to be used we need MAX_CURRENT. The default are now 48, in my case that my device supports 32A I set it on docker.

I used a safe approach to make it work on many devices. Can refuse to give some responses, but after getting all data from device messages it will start responding to the device.

FalconFour, 2024-10-11:
Fixed in latest commit 😄 Now it sets offline current after 6 minutes of not sending commands (so the box "hints" its present offline setting). Clever, and it works!

closed 2024-10-11 by FalconFour