JuiceRescue/juicepassproxy #95: juicebox_commands: Telemetry "max_current" should not override settings

FalconFour, 2024-10-05:
In the telemetry messages, it appears that the echoed max_current is influencing the controls set by the user - creating a sort of "eating its own tail" loop:
```
2024-10-04 23:36:40  INFO      [juicebox_mqtthandler] Number Callback (Max Charging Current): 30. User Data: None
2024-10-04 23:36:40  INFO      [ha_mqtt_discoverable.sensors] Setting Max Charging Current to 30.0 using hmd/number/JuiceBox/Max-Charging-Current/state
2024-10-04 23:36:40  INFO      [juicebox_mitm] command message = CMD52336A30M40C006S067!BQ7$ new_values=True new_version=False
2024-10-04 23:36:40  INFO      [juicebox_mitm] Sending command to juicebox CMD52336A30M40C006S067!BQ7$ new_values=True
2024-10-04 23:36:40  INFO      [ha_mqtt_discoverable.sensors] Setting Max Charging Current to 30.0 using hmd/number/JuiceBox/Max-Charging-Current/state
2024-10-04 23:36:44  INFO      [juicebox_udpcupdater] JuiceboxUDPCUpdater Check Starting
2024-10-04 23:36:44  DEBUG     [juicebox_udpcupdater] Active UDPC Stream: 2
2024-10-04 23:36:44  INFO      [juicebox_udpcupdater] UDPC IP incorrect, updating
2024-10-04 23:36:44  DEBUG     [juicebox_udpcupdater] Closing UDPC stream: 2
2024-10-04 23:36:45  INFO      [juicebox_mqtthandler] Number Callback (Max Current): 30. User Data: None
2024-10-04 23:36:45  INFO      [ha_mqtt_discoverable.sensors] Setting Max Current to 30.0 using hmd/number/JuiceBox/Max-Current/state
2024-10-04 23:36:45  INFO      [juicebox_mitm] command message = CMD52336A30M30C006S068!NF9$ new_values=True new_version=False
2024-10-04 23:36:45  INFO      [juicebox_mitm] Sending command to juicebox CMD52336A30M30C006S068!NF9$ new_values=True
2024-10-04 23:36:45  INFO      [ha_mqtt_discoverable.sensors] Setting Max Current to 30.0 using hmd/number/JuiceBox/Max-Current/state
2024-10-04 23:36:45  INFO      [juicebox_udpcupdater] UDPC IP Saved
2024-10-04 23:36:53  INFO      [juicebox_message] from_string 0817081001070432386218277503:v07,s0081,u08628,V2397,L0005160337,S0,T23,M32,m40,t09,i79,e-001,f6002,X0,Y0!DQ8:
2024-10-04 23:36:53  DEBUG     [juicebox_mqtthandler] From JuiceBox: b'0817081001070432386218277503:v07,s0081,u08628,V2397,L0005160337,S0,T23,M32,m40,t09,i79,e-001,f6002,X0,Y0!DQ8:'
2024-10-04 23:36:53  DEBUG     [juicebox_mqtthandler] Publish Basic Message: {'type': 'basic', 'current': 0, 'energy_session': 0, 'protocol_version': '07', 'unknown_s': '0081', 'loop_counter': '08628', 'voltage': 239.7, 'energy_lifetime': 5160337.0, 'status': 'Unplugged', 'temperature': 73.4, 'current_max_charging': 32.0, 'current_rating': 40.0, 'report_time': '09', 'interval': '79', 'unknown_e': '-001', 'frequency': 60.02, 'unknown_X': '0', 'unknown_Y': '0', 'power': 0.0, 'current_max': 40.0, 'data_from_juicebox': '0817081001070432386218277503:v07,s0081,u08628,V2397,L0005160337,S0,T23,M32,m40,t09,i79,e-001,f6002,X0,Y0!DQ8:'}
2024-10-04 23:36:53  INFO      [ha_mqtt_discoverable.sensors] Setting Current to 0 using hmd/sensor/JuiceBox/Current/state
2024-10-04 23:36:53  INFO      [ha_mqtt_discoverable.sensors] Setting Energy (Session) to 0 using hmd/sensor/JuiceBox/Energy--Session-/state
2024-10-04 23:36:53  INFO      [ha_mqtt_discoverable.sensors] Setting Voltage to 239.7 using hmd/sensor/JuiceBox/Voltage/state
2024-10-04 23:36:53  INFO      [ha_mqtt_discoverable.sensors] Setting Energy (Lifetime) to 5160337.0 using hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-10-04 23:36:53  INFO      [ha_mqtt_discoverable.sensors] Setting Status to Unplugged using hmd/sensor/JuiceBox/Status/state
2024-10-04 23:36:53  INFO      [ha_mqtt_discoverable.sensors] Setting Temperature to 73.4 using hmd/sensor/JuiceBox/Temperature/state
2024-10-04 23:36:53  INFO      [ha_mqtt_discoverable.sensors] Setting Max Charging Current to 32.0 using hmd/number/JuiceBox/Max-Charging-Current/state
2024-10-04 23:36:53  INFO      [ha_mqtt_discoverable.sensors] Setting Current Rating to 40.0 using hmd/sensor/JuiceBox/Current-Rating/state
2024-10-04 23:36:53  INFO      [ha_mqtt_discoverable.sensors] Setting Frequency to 60.02 using hmd/sensor/JuiceBox/Frequency/state
2024-10-04 23:36:53  INFO      [ha_mqtt_discoverable.sensors] Setting Power to 0.0 using hmd/sensor/JuiceBox/Power/state
2024-10-04 23:36:53  INFO      [ha_mqtt_discoverable.sensors] Setting Max Current to 40.0 using hmd/number/JuiceBox/Max-Current/state
2024-10-04 23:36:53  INFO      [juicebox_mitm] command message = CMD52336A30M30C006S069!KF9$ new_values=False new_version=False
2024-10-04 23:36:53  INFO      [juicebox_mitm] Sending command to juicebox CMD52336A30M30C006S069!KF9$ new_values=False
```
In the above: I set max current=30 and max charging current=30, but in the next telemetry message, it resets the values to those sent in the next telemetry string.

In my case, this is influenced a bit by the fact that my box doesn't seem to be getting the replies - yet TBD (Wireshark confirms the reply is properly sent, and there's no errors coming to the Zentri serial console or error messages returned by the box over UDP). But this shows a logical error in the code - these messages should be stored as separate, read-only validations of the station's operational mode, not changing a user-determined setting. It's more of a debug message than anything :) It says "I'm confirming these settings", not "I have changed my setting to this".

ivanfmartinez, 2024-10-05:
On v09  the value came from the device, and reflect the real value on the device.

v07  does not send the C variable and I have to use a different value. At this moment we dont really know if this have any effect on v07. A user must test this changing the value when online an testing the device at offline mode.

This is the code that updates the parameter : https://github.com/snicker/juicepassproxy/pull/69/files#diff-f0b56a62fb82577d98349d94af41e58b863885d740fcac8eb3b680bbe8a36e08R490

I have made some changes that are uncommited on README because I dont know what are the real behaviour on this v07 devices.

The original code was made considering the need for the TWO parameters on the CMD response, based on v09u tests. 

But if someone test it and confirm that the v07 device WORKS without sending the value on the CMD we can make it optional and dont send if the device was not sendint fo server.

The change that I made was just to make the juicepassproxy code send the response for juicebox without knowing the correct behaviour on v07.

The packet dumps that other users send show the ENELX server sending both parameters to the device, but if the device really use both only testing with real devices will confirm.

ivanfmartinez, 2024-10-05:
I agree that we can sepraate the command values from the used setting values, but using the same field permitted to use at begining the configurations MADE on my homeassistant and made on ENEL X to reflect the same value.

This behaviour is the same that the ENELX app used.

ivanfmartinez, 2024-10-05:
Also we need to know if all the fields in v07 reflect same values from v09u. 

During my tests I found definitions about the two fields that are reverse of the real behaviour from my device. The code on my branch works correctly on my v09u device. Both parameters are respected, and when I keep server without responding for more than 5 minutes the charger returns to the offline configuration.

niharmehta, 2024-10-05:
In the long past with HA, I made the mistake of trying to use a input number to update a value, and then the response of the value update the slider position which triggers it to try and again update the device value. Which caused a feedback type loop.  Eventually just gave up using the slider and just send a service call and then read it back as an entity.   

In regards to v07 protocol, I am happy to test, but not sure what exactly I should do to get the information you need.

ivanfmartinez, 2024-10-19:
Now the entities are separate

closed 2024-10-19 by ivanfmartinez