JuiceRescue/juicepassproxy #97: juicebox_commands branch: Act as Server errors in logs

wozz, 2024-10-05:
I'm seeing the following errors in the log:
```
2024-10-05 09:40:53  DEBUG     [juicebox_mqtthandler] Errors in last 60 min: 1
2024-10-05 09:40:53  WARNING   [juicebox_mqtthandler] Can't update attributes for Act as Server as MQTT isn't connected/started. (AttributeError: 'Switch' object has no attribute 'update_state')
```

I don't have the Act as Server switch available either

ivanfmartinez, 2024-10-05:
I have just tested from a build from my repository and works fine here.

Are you using the image from your repo definition in https://github.com/wozz/ha-addons/blob/main/juicepassproxy/Dockerfile ?

I made my build using the Dockerfile on the repository (except that my local build only work when adding extra slash / at copy command) : https://github.com/ivanfmartinez/juicepassproxy/blob/juicebox_commands/Dockerfile

snicker, 2025-01-20:
Closing for inactivity - please try v0.5.0 and report back if issue is not resolved.

closed 2025-01-20 by snicker