JuiceRescue/juicepassproxy #100: Current max bounds should be 80, not 48

FalconFour, 2024-10-06:
`juicepassproxy.py` line 48:
```
    max_current = config.get("MAX_CURRENT", 48)
```
Protocol max value is 80. Box will not operate at higher-than-rated values due to many levels of internal validation, so sending a higher value is not hazardous.

This can be a problem because the ZAP sends "80" when it doesn't have communication with the Atmel processor, and it throws script "out of bounds" errors while parsing.

ivanfmartinez, 2024-10-06:
The way that entities are created now requires the maximum value to be set before any message came from the device. considering that many devices are  32, 40 and 48. This default appear to be safe for most users because if we put 80 the slider created on HA will be too big and make difficult to select the value.

The correct solution will be a big refator that can be made later to only create MQTT entities after the first message come from the device with "Current Rating", this will make possible to specify the correct range.

In my mind the refactor is necessary also to make possible to support multiple juicebox devices on a single JPP instance.

The minority of users that have devices with bigger current can set the MAX_CURRENT parameter to 80 until a better refactor can get better solution.

FalconFour, 2024-10-07:
Fair enough :) Most are indeed 32, 40, and 48. The 75 was fairly uncommon but we've already seen a couple folks join Discord with 75-amp stations. May be worth some kind of patch there to keep it from inexplicably crashing for them.

ivanfmartinez, 2024-10-07:
They just have to set the value in the configuration file to 80, mine is 32

```
grep CURRE config/juicepassproxy.yaml
MAX_CURRENT: 32

```

ivanfmartinez, 2024-10-19:
@FalconFour have you tested this on latest  versions ? 

I have removed my MAX_CURRENT definition from the configuration, now without the MAX_CURRENT defined on the second JPP start it will use the last value received from the device current rating