JuiceRescue/juicepassproxy #104: Android App for Basic Control As Alternative to JPP+HASS

atc99, 2024-10-11:
I don't know if this is practical or in scope, but is possible to create a JPP Android app which allows displaying data and controlling the current settings of the JB without needing any network changes other than running the initial JB setup routine to connect to the app wifi network?

* App sets up a [local-only hotspot](https://developer.android.com/develop/connectivity/wifi/localonlyhotspot) which the JB can connect to.
* App/phone provides DHCP (may be a part of the local-only hotspot)
* App provides DNS to override resolution of the emotorwerks domain.
* App may provide NTP, if this is determined to be necessary for the JB at some point.
* App listens on UDP port 8042 to parse the commands and display the data being received in real-time.
* App provides the ability to control the Online/Offline max current settings and send the changes to the JB (this is probably the most important objective).
* App could have multiple tabs or a unit selection (by juicebox id) capability if multiple units connect to it simultaneously.
* Should this be successful, future updates could include logging and graphing of the data.

If this is possible, it could make it much less complicated for most users and a simpler "how-to" guide.  I have not leveraged these capabilities in an Android app.

ivanfmartinez, 2024-10-19:
@atc99 I think that if something like this is made will be valid just to control the offline current and online when the app is connected to the device. 

Maybe an ESP32 device can give better results, Juicebox conect to ESP32 and user can connect also and control  / view using web interface. This is simple and cheap for users that does not have any kind of home automation.

This is can be nice for some users but probably out of the scope of the current JPP. Maybe the code from JPP can be used on ESP32 with micropython (I never tested python on ESP32). Will be very nice if someone can try this.

atc99, 2024-10-19:
Non-rooted android devices aren't able to intercept the DNS resolution via a VPN since the mobile hotspot traffic side-steps the VPN and goes straight to the internet.  I tried this with [personal DNS filter (pDNSf)](https://www.zenz-solutions.de/personaldnsfilter-wp/).  In a second attempt, I had the router do the DNS override and point to an Android phone and an app was able to receive the UDP packets from the JB.  I'm not sure that these can be brought together as a complete solution using only the Android mobile hotspot on a non-rooted phone.  The intent was primarily for device control and possibly some limited real-time readouts and diagnostics more than long-term logging/graphing or automation.

I am closing this issue as it is unplanned for the current project.

closed 2024-10-19 by atc99