JuiceRescue/juicepassproxy #106: HASS: Hotspot + DNSMasq + JPP for MITM on Standalone HASS Unit

atc99, 2024-10-11:
Maybe it is possible to add a wifi adapter to a standalone Home Assistant (HASS) box and set up its own wifi hotpoint running HASSIO HotSpot and DNSMasq and then have the JuiceBox connect to the HASS wifi hotspot.  JPP would also need to be running on the HASS box and listening on port 8042 with --juicebox_id and --ignore_enelx (without --update_udpc) to avoid the telnet calls.  Configure JPP as appropriate.

I don't have a standalone Home Assistant (HASS) unit to try this solution out.

HASSIO HotSpot
https://github.com/joaofl/hassio-addons/tree/master/hassio-hotspot

```
{
    "ssid": "WIFI_NAME",
    "wpa_passphrase": "WIFI_PASSWORD",
    "channel": "0",
    "address": "192.168.2.1",
    "netmask": "255.255.255.0",
    "broadcast": "192.168.2.254"
    "interface": ""
    "interface_internet": "eth0"
    "allow_internet": false
    "dhcp_server": true
    "dhcp_start": "192.168.2.100",
    "dhcp_end": "192.168.2.200",
    "dhcp_dns": "<HASS_WIFI_IP i.e. 192.168.2.1>",
    "dhcp_subnet": "255.255.255.0",
    "dhcp_router": "192.168.2.1",
    "hide_ssid": false
}
```

DNSMasq Add-On
https://github.com/home-assistant/addons/blob/5fc9b3e5a4d9db805f65ad960e28a5516f37e456/dnsmasq/DOCS.md

Follow these steps to get the add-on installed on your system:

1. Navigate in your Home Assistant frontend to **Settings** -> **Add-ons** -> **Add-on store**.
2. Find the "Dnsmasq" add-on and click it.
3. Click on the "INSTALL" button.
4. Configure hosts to override jvb1.emotorwerks.com to resolve as the HASS wifi IP.
```
hosts:
  - host: jvb1.emotorwerks.com
    ip: <HASS_WIFI_IP i.e. 192.168.2.1>
```

ivanfmartinez, 2024-10-19:
https://github.com/JuiceRescue/juicepassproxy/issues/104

atc99, 2024-10-19:
If a solution like this could be put together then it should be added to the JPP documentation as a how-to or example configuration.

I am closing this issue as it is unplanned for the current project.

closed 2024-10-19 by atc99