JuiceRescue/juicepassproxy #107: New sensor state for last update or online/offline connection status

atc99, 2024-10-11:
Add a new sensor for last update received (datetime) and an online/offline Connection Status.  Naming is not fixed and is just a suggestion.  Adding these could enable support for other automation notifications or more advanced rules related to load-balancing.

* Last Update (Received)
  * Last time a command was received from the JB
* Last Update (Sent)
  * Last time a command was sent to the JB for a configuration change (current changes, etc)
* Connection Status
  * [Online,Offline] enum.
  * JPP sends offline for all JB's it has been processing commands for when shutting down.
  * JPP calculates this based on how long it has been since the JB was last seen.  Should be aligned with the 3-5 minute online/offline charging current setting.

atc99, 2024-10-16:
It looks like the [last_changed](https://community.home-assistant.io/t/how-do-i-use-state-last-changed-in-automation-condition/34465/3) attribute may cover the last update (received) and possibly the sent when using a template variable.

The connection status may also be able to be determined by the last_changed attribute or it could be an automation that updates a manually created sensor based on a last_changed attribute, but having JPP actively update it may be useful.

snicker, 2025-01-20:
Will leave issue open but recommend changing scope to just a connectivity sensor to be included in the future. 

in the meantime- you _could_ use an "ICMP Ping" sensor in Home Assistant if you are curious about the online state of your Juicebox, and follow @atc99 's recommendation to create a template sensor for the `last_changed` attribute.