JuiceRescue/juicepassproxy #112: updated Version including commands

BeneWilh, 2024-11-10:
Hi there,

It a great effort for everbody having a Juicbox and beeing willing in using it for solar powered charging. Since Enelx made the switch zu Juicepass steeing the Wallbox in EU is only possible in the App (very annoying as we all know). I'm already using juicepassproxy to read the values from my Juicebox 32 Pro (in EU) per MQTT and would love to be able to send also commands. Could anybody with higher capabilities in programming compile a new version? Thank you!

snicker, 2024-11-10:
@BeneWilh  progress on this functionality and version is being tracked here : https://github.com/JuiceRescue/juicepassproxy/pull/69

closed 2024-11-10 by snicker

BeneWilh, 2024-11-10:
well, my knowledge is not good enough to understand how i can start using the new branch (on a manual installation without docker).. maybe you could give some instructions for not native programmers. Thx