JuiceRescue/juicepassproxy #113: branch: juicebox_commands

BeneWilh, 2024-11-11:
I managed to get the branch "juicebox_commands" from @ivanfmartinez basically running..

Please see attached logging files for errors and further enhancement:

Logging_branch_jb_commands_log.txt => I had to disable logging to file
Logging_branch_jb_commands.txt => error this morning, yesterday it worked. seem to me like Juicebox got an error. ping is working as well as managing from the app.

Installation: Debian 12, manual install, branch: juicebox_commands

[Logging_branch_jb_commands_log.txt](https://github.com/user-attachments/files/17698258/Logging_branch_jb_commands_log.txt)
[Logging_branch_jb_commands.txt](https://github.com/user-attachments/files/17698263/Logging_branch_jb_commands.txt)

ivanfmartinez, 2024-11-11:
@BeneWilh the code paths that you had problem was not used on the docker image, now both should work with your environment.

Please use the version with latest commits and should work fine with default log on home directory.

But according to the log the JPP was unable to telnet to juicebox, you have to check that or set the juicebox_id from the command line. If that was just a temporary error on the device you can just retry.

BeneWilh, 2024-11-12:
@ivanfmartinez Looking now promising. Will test while charging next days.

[logging_juicebox_2024-11-12.txt](https://github.com/user-attachments/files/17722314/logging_juicebox_2024-11-12.txt)

closed 2024-11-12 by BeneWilh