JuiceRescue/juicepassproxy #114: OCPP Functionality?

kintaroju, 2024-11-13:
Hi All,

i am inquiring if anyone tried to do a HA, and use the "OCPP" functionality with their Juicebox Units by chance:

https://tridenstechnology.com/enel-x-way-migration/#h-how-to-migrate-from-enel-x-way
https://www.ampcontrol.io/post/navigating-the-juicebox-shutdown-how-to-migrate-and-switch-to-reliable-ocpp-software-with-ampcontrol

If this is viable solution, maybe we can use this and see about leveraging something existent? Also if anyone did make it work, if they can list what they can/cannot do that would be awesome! thanks!

snicker, 2025-01-20:
Recommended approach for now would be to build a secondary OCPP <-> MQTT bridge. This is some we could consider for a future implementation within juicepassproxy but we need expertise from a python dev familiar with OCPP implementations.

razvanphp, 2025-05-20:
> Recommended approach for now would be to build a secondary OCPP <-> MQTT bridge.

While working on [rabbitmq-web-ocpp plugin](https://github.com/vampirebyte/rabbitmq-web-ocpp/), I thought about this, being able to enrol the boxes to a traditional CSMS.

Would love to help test this.

Cheers,
R