JuiceRescue/juicepassproxy #115: JPP repeatedly crashing on invalid temperature

SLaks, 2024-11-15:
From the logs:

```

2024-11-15 16:16:48,023 INFO       Setting Power to 0 using hmd/sensor/Car-Charger/Power/state
2024-11-15 16:17:17,898 ERROR      Exception handling local data: could not convert string to float: '\\x0c\\xe1\\x89f`\\x81s\\x93r\\xa5\\x1b%/w/\\x11y\\xf2$>\\x81\\x1f\\xe2\\xb7\\xb4*\\x8a\\x10\\xc7'
Traceback (most recent call last):
  File "/juicepassproxy/juicepassproxy.py", line 335, in local_data_handler
    message = self.basic_message_try_parse(data)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/juicepassproxy/juicepassproxy.py", line 264, in basic_message_try_parse
    message["temperature"] = round(float(part.split("T")[1]) * 1.8 + 32, 2)
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: '\\x0c\\xe1\\x89f`\\x81s\\x93r\\xa5\\x1b%/w/\\x11y\\xf2$>\\x81\\x1f\\xe2\\xb7\\xb4*\\x8a\\x10\\xc7'
Traceback (most recent call last):
  File "/juicepassproxy/juicepassproxy.py", line 743, in <module>
    main()
  File "/juicepassproxy/juicepassproxy.py", line 735, in main
    pyproxy.udp_proxy(src, dst)
  File "/juicepassproxy/pyproxy/pyproxy.py", line 59, in udp_proxy
    proxy_socket.sendto(data, server_address)
TypeError: a bytes-like object is required, not 'NoneType'
--------------------------------
2024-11-15 16:17:20,124 INFO       Starting JuicePass Proxy
[...]
2024-11-15 16:17:20,126 INFO       Docker Environment Variables:
2024-11-15 16:17:20,144 INFO       UPDATE_UDPC: false
2024-11-15 16:17:20,146 INFO       DEBUG: false
2024-11-15 16:17:21,326 INFO       Starting JuicePass Proxy v0.2.2
2024-11-15 16:17:21,326 INFO       config_loc: /config/juicepassproxy.yaml
2024-11-15 16:17:21,525 INFO       enelx_server: juicenet-udp-prod3-usa.enelx.com
2024-11-15 16:17:21,525 INFO       enelx_port: 8047
2024-11-15 16:17:21,525 INFO       src: 172.16.100.2:8047
2024-11-15 16:17:21,535 INFO       dst: 158.47.1.128:8047
```

And the `Car Charger Status` entity in HA reads `unknown S\x13\xe3\x04\xbc\xec\x08\xb7` (the exact value changes every 2-3 hours).

For a few days before that, it was not reporting when I plugged in my car.

SLaks, 2024-11-15:
The Enel X Way app reports model 2JBO401RNA

SLaks, 2024-11-15:
I turned on `DEBUG` and got a different error:

```
2024-11-15 16:29:49,182 DEBUG      State topic unset, using default: hmd/sensor/Car-Charger/Power/state
2024-11-15 16:29:49,182 DEBUG      Writing '0' to hmd/sensor/Car-Charger/Power/state
2024-11-15 16:29:49,182 DEBUG      Publish result: (0, 3)
2024-11-15 16:29:53,668 DEBUG      remote: b'\xeev\x1c\x96\x03\xeee\x9e\xddPI\xab0\xb09A\xd3\xbd\x8a*\xff\xe6m\xd4t\xbb@\xb4Ac\xd0~\xd6\x12\xf0\x17{\xc1rr\x0cy#\xdb`\xf0\xa6\xd7\xab\r\x8b\x07\x08\x99\xddY\xc6;\xe7mEV6\xc5'
2024-11-15 16:30:19,001 DEBUG      local: b'0910042001260538236222625020:v09e\x00\x8f\x16\x00\x00\x00l\xfdz\xde\xb4\x0e\x92\x9b\xbf\xe7\xc5\xc4\xd91\x8bo\x14\xbfG\xa8\xa9\x18t\x84qk\x07<\xd1\xa7\x00\x98\xef\x0c*\xa4\x8aCD\x18\x07\xaa\xc3\n\xed\xfc\nl^\xf8p\xcf\r\xe87\xd1\x99\r\xffKYQS=\xff\xd8\xb3\xa6\xd0\xcby\x95\xbe\xde\x0e\xbf*\xcc\x95P#OR\xca\x1f\xc1\x04\xccf2\x1c\x84?\xc2I\xe64\x1b~<\x0c\xe9\x14\xaf\x85\xbe\xbdU\xd2\xab4\x81\x91\x88\t\xa8\xcb\xf9\xd3\x0c\xb3w\x1e\xdb\xf6\xb1\xa8d\xfc\xa0-9\x95\x03\xd9\n\x86\\\x03\x00\xe8\x12\xe7\xbd\x15\xd8l\x99\xd5\xa4P\xe9#'
2024-11-15 16:30:19,001 WARNING    Index error when handling local data, probably wrong number of items in list- nothing to worry about unless this happens a lot. (pop from empty list)
Traceback (most recent call last):
  File "/juicepassproxy/juicepassproxy.py", line 743, in <module>
      main()
      File "/juicepassproxy/juicepassproxy.py", line 735, in main
      pyproxy.udp_proxy(src, dst)
      File "/juicepassproxy/pyproxy/pyproxy.py", line 59, in udp_proxy
      proxy_socket.sendto(data, server_address)
    TypeError: a bytes-like object is required, not 'NoneType'
2024-11-15 16:30:19,031 DEBUG      Shutting down MQTT client
```

ivanfmartinez, 2024-11-15:
@SLaks your DEBUG shows that your device is using encrypted protocol. 

Encrypted currently cannnot be decoded. The option is to disable Juicebox Internet access and use the command branch to use only local control without ENEL X servers

https://github.com/JuiceRescue/juicepassproxy/pull/69

SLaks, 2024-11-15:
Fascinating...

Does that mean that JuiceBox pushed an update to my device _after the company was shut down_? (it was last working on November 7th)

What is the current status of that branch? Does it have current documentation? If I understand correctly, it uses the same setup as the base image, but requires option C (redirect UDP at the router level)?

ivanfmartinez, 2024-11-15:
A friend in Brazil also received yesterday that change, they are not updating the device but sending a different configuration.

The branch is working, all information is on the #69 . 

Should works like main with all options.

I dont use ENEL X servers for months.

SLaks, 2024-11-15:
I switched my Docker image to pull from `ivanfm/juicepassproxy:latest` (without changing any of my other config; I use DNS rebinding), and got

```
2024-11-15 18:44:45  DEBUG     [juicebox_mqtthandler] From JuiceBox: b'0910042001260538236222625020:v09e\x00\x8f\x16\x00\x00\x00w\xe3\xe9nG\x15\xeaHn\xc0.\x9c\x9d\xde\xac\xd3xw\xa0\xc1C\xdd\xfe\xbdo\x86S\xe0\xa5\xd4\xf8\x83\xb7*Q\xf7\x06\x11\x03\xe0\x16\xa2\xa1DJ\xe3\x1c)(\x0f\xef\x9c\xdcs\xac\x1c\xc2Eh\xecK\'\x07V\x0b}\xc9B[\x17O\x1cx\xa3\xf9H\x951\x04c>\xf0~\xe4\x10\xfd*\x88\xf1w\xcf\xb0\x1a^\x1cj\xf7\x0b\xa6\xe2\x99\xefAH\xfe\x13\xd9\x85#\x9b-\xc9s\xd5\x1fa2\xba3\xe8\xcc$qA\xc0\xd5B\x89\xbb\xdeK\x8e\x9a\xe8Vo"\xc4Q(9\x88\xe2\x9e\xa0\x02\x0c\xb5\x87a09P' decoded=None!None:
2024-11-15 18:44:45  ERROR     [__main__] A JuicePass Proxy task failed: TypeError: 'NoneType' object is not iterable
Traceback (most recent call last):
File "/juicepassproxy/juicepassproxy.py", line 585, in main
  await asyncio.gather(
        File "/juicepassproxy/juicebox_mitm.py", line 62, in start
  await self._connect()
    File "/juicepassproxy/juicebox_mitm.py", line 105, in _connect
  self._mitm_loop_task = await self._mitm_loop()
    ^^^^^^^^^^^^^^^^^^^^^^^
                           File "/juicepassproxy/juicebox_mitm.py", line 136, in _mitm_loop
  await self._main_mitm_handler(data, remote_addr)
    File "/juicepassproxy/juicebox_mitm.py", line 213, in _main_mitm_handler
  data = await self._local_mitm_handler(data, decoded_message)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
           File "/juicepassproxy/juicebox_mqtthandler.py", line 647, in local_mitm_handler
  message = decoded_message.to_simple_format()
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
              File "/juicepassproxy/juicebox_message.py", line 381, in to_simple_format
  for k in self.values:
      ^^^^^^^^^^^
             TypeError: 'NoneType' object is not iterable
2024-11-15 18:44:53  DEBUG     [__main__] jpp_task_list: [<Task finished name='mqtt_handler' coro=<JuiceboxMQTTHandler.start() done, defined at /juicepassproxy/juicebox_mqtthandler.py:450> result=None>, <Task finished name='mitm_handler' coro=<JuiceboxMITM.start() done, defined at /juicepassproxy/juicebox_mitm.py:58> exception=TypeError("'NoneType' object is not iterable")>]
2024-11-15 18:44:58  ERROR     [__main__] Restarting JuicePass Proxy Loop (10)
2024-11-15 18:44:58  INFO      [juicebox_mqtthandler] max_current: 48
2024-11-15 18:44:58  INFO      [juicebox_mqtthandler] Checking for initial_states on config
2024-11-15 18:44:58  INFO      [juicebox_mqtthandler] Starting JuiceboxMQTTHandler
2024-11-15 18:44:58  INFO      [juicebox_mitm] Starting JuiceboxMITM at 172.16.100.2:8047 reuse_port=True
2024-11-15 18:44:58  DEBUG     [juicebox_mitm] EnelX: 158.47.1.128:8047
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/-Car-Charger-/Status/config
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/-Car-Charger-/Status/state
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/-Car-Charger-/Current/config
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/-Car-Charger-/Current/state
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/-Car-Charger-/Current-Rating/config
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/-Car-Charger-/Current-Rating/state
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/-Car-Charger-/Max-Current-Offline-Device-/config
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/-Car-Charger-/Max-Current-Offline-Device-/state
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/number/-Car-Charger-/Max-Current-Offline-Wanted-/config
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] state_topic: hmd/number/-Car-Charger-/Max-Current-Offline-Wanted-/state
2024-11-15 18:44:58  WARNING   [juicebox_mqtthandler] Max Current(Offline/Wanted) has no initial_state
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/-Car-Charger-/Max-Current-Online-Device-/config
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/-Car-Charger-/Max-Current-Online-Device-/state
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/number/-Car-Charger-/Max-Current-Online-Wanted-/config
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] state_topic: hmd/number/-Car-Charger-/Max-Current-Online-Wanted-/state
2024-11-15 18:44:58  WARNING   [juicebox_mqtthandler] Max Current(Online/Wanted) has no initial_state
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/-Car-Charger-/Frequency/config
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/-Car-Charger-/Frequency/state
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/-Car-Charger-/Energy--Lifetime-/config
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/-Car-Charger-/Energy--Lifetime-/state
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] config_topic: homeassistant/sensor/-Car-Charger-/Energy--Session-/config
2024-11-15 18:44:58  INFO      [ha_mqtt_discoverable] state_topic: hmd/sensor/-Car-Charger-/Energy--Session-/state
[... cut ...]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
           File "/juicepassproxy/juicebox_mqtthandler.py", line 647, in local_mitm_handler
  message = decoded_message.to_simple_format()
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
              File "/juicepassproxy/juicebox_message.py", line 381, in to_simple_format
  for k in self.values:
      ^^^^^^^^^^^
             TypeError: 'NoneType' object is not iterable
```

Do I need to set `current_max_online` before it can launch at all?
I don't intend to control the max current at all.

ivanfmartinez, 2024-11-15:
The images in my docker repository are updated first message of #69  was updated with the correct address, I dont known if niharmeta or others are updated. 

The juicebox needs to be restarted without internet connection to work to return to unencrypted mode. 
If it still connect to ENEL X directory service it will switch to ENCRYPT mode.

I will check later if I can do better checking when device sends the encrypted messages to dont generate exceptions.

ivanfmartinez, 2024-11-15:
Just made a new commit with the procedure and workaround to dont get this errors when JPP receives encrypted messages : 

https://github.com/JuiceRescue/juicepassproxy/pull/69/commits/845c5a916deab5a7e860bc6bf2034ce2748224f0

Updated my docker images

niharmehta, 2024-11-16:
Updated my docker image as well  for x86 as well as arm users (Pi)

snicker, 2025-01-20:
closing as this should now be fixed in v0.5.0 after #69 was merged.

closed 2025-01-20 by snicker