JuiceRescue/juicepassproxy #116: Not a valid juicebox message...

BeneWilh, 2024-11-15:
@ivanfmartinez 

I got an error using the newest branch, please check.

"A JuicePass Proxy task failed: UnicodeDecodeError: 'utf-8' codec can't decode byte 0x9e in position 32: invalid start byte"

Details in the attached log. Thank you very much!

[Logging_branch_jb_commands_log 2024-11-14.txt](https://github.com/user-attachments/files/17780033/Logging_branch_jb_commands_log.2024-11-14.txt)

ivanfmartinez, 2024-11-15:
Same as #115 and #73 you need to restart the device without internet access to return it to unencrypted mode

ivanfmartinez, 2024-11-15:
@BeneWilh one thing interesting in your log is the messages have protocol version "v08", I never seen that before.

Will be nice if you add information about your device on the Wiki .

https://github.com/JuiceRescue/JuiceboxRescueWiki/wiki/Firmware-Versions

https://github.com/JuiceRescue/JuiceboxRescueWiki/wiki/Hardware-Versions

BeneWilh, 2024-11-16:
@ivanfmartinez I am owning a Juicebox 32Pro from 08/2018, all data were already available in the Wiki, I added v08.
With https://directory-api.emotorwerks.com/v1/profiles/juicebox/<serial> I'm getting the response

```
{
  "control_api_endpoint" : "https://juicenet-prod4.enelx.com",
  "device_https_endpoint" : "https://juicenet-prod4.enelx.com",
  "device_ocpp_endpoint" : "https://juicenet-prod4.enelx.com",
  "device_protocol" : "Both",
  "device_udp_endpoint" : "juicenet-udp-prod4-exw.enelx.com:9004",
  "encrypt" : "false"
}
```
which means - if I understand it correctly communication is without encryption.

IP/command/ver returns:

`{"id":1,"code":0,"flags":0,"response":"EMWERK-JB_1_1-1.4.0.28, 2021-04-27T20:39:50Z, ZentriOS-WZ-3.6.4.0\r\n"}`

I restarted the box without internet, reconnected to internet and the error still exists and v08 ist used. 

[Logging_branch_jb_commands_log 2024-11-16.txt](https://github.com/user-attachments/files/17787573/Logging_branch_jb_commands_log.2024-11-16.txt)

as changing amps/power via HomeAssistand was not working for me (also not with v07) its dificukt to exclude App usage (in europe the apps till works and at leat you can manually adjust charging, which is annoying). First time i see v08 in my logs is 2024-11-14 15:28:23; 2024-11-14 15:27:36 was the last time v07 is in the protocol

```
2024-11-14 15:27:13  INFO      [ha_mqtt_discoverable.sensors] Setting Power to 0 using hmd/sensor/JuiceBox/Power/state
2024-11-14 15:27:16  DEBUG     [juicebox_mqtthandler] From EnelX: b'CMD41527A20M20C006S574!4CH$'
2024-11-14 15:27:16  INFO      [ha_mqtt_discoverable.sensors] Setting Data from EnelX to CMD41527A20M20C006S574!4CH$ using hmd/sensor/JuiceBox/Data-from-EnelX/state
2024-11-14 15:27:35  INFO      [juicebox_udpcupdater] JuiceboxUDPCUpdater Check Starting
2024-11-14 15:27:35  DEBUG     [juicebox_udpcupdater] Active UDPC Stream: 3
2024-11-14 15:27:35  INFO      [juicebox_udpcupdater] UDPC IP correct
2024-11-14 15:27:36  INFO      [juicebox_message] from_string <<JuiceboxID>>:**v07**,s0538,u37300,V2308,L0006320697,S0,T10,M20,m32,t09,i56,e-001,f4998,X0,Y0!AT6:
2024-11-14 15:27:36  DEBUG     [juicebox_mqtthandler] From JuiceBox: b'<<JuiceboxID>>:**v07**,s0538,u37300,V2308,L0006320697,S0,T10,M20,m32,t09,i56,e-001,f4998,X0,Y0!AT6:' decoded=<<JuiceboxID>>:v07,s0538,u37300,V2308,L0006320697,S0,T10,M20,m32,t09,i56,e-001,f4998,X0,Y0!AT6:
2024-11-14 15:27:36  DEBUG     [juicebox_mqtthandler] decode/parsed message = {'type': 'basic', 'current': 0, 'energy_session': 0, 'serial': '<<JuiceboxID>>', 'protocol_version': '07', 'counter': '0538', 'loop_counter': '37300', 'voltage': 230.8, 'energy_lifetime': 6320697, 'status': 'Unplugged', 'temperature': 50.0, 'current_max_online': 20, 'current_rating': 32, 'report_time': '09', 'interval': 56, 'e': '-001', 'frequency': 49.98, 'X': '0', 'Y': '0', 'power': 0}
2024-11-14 15:27:36  DEBUG     [juicebox_mqtthandler] Publish Basic Message: {'type': 'basic', 'current': 0, 'energy_session': 0, 'serial': '<<JuiceboxID>>', '**protocol_version': '07'**, 'counter': '0538', 'loop_counter': '37300', 'voltage': 230.8, 'energy_lifetime': 6320697, 'status': 'Unplugged', 'temperature': 50.0, 'current_max_online': 20, 'current_rating': 32, 'report_time': '09', 'interval': 56, 'e': '-001', 'frequency': 49.98, 'X': '0', 'Y': '0', 'power': 0}
2024-11-14 15:27:36  INFO      [ha_mqtt_discoverable.sensors] Setting Current to 0 using hmd/sensor/JuiceBox/Current/state
2024-11-14 15:27:36  INFO      [ha_mqtt_discoverable.sensors] Setting Energy (Session) to 0 using hmd/sensor/JuiceBox/Energy--Session-/state
2024-11-14 15:27:36  INFO      [ha_mqtt_discoverable.sensors] Setting Voltage to 230.8 using hmd/sensor/JuiceBox/Voltage/state
2024-11-14 15:27:36  INFO      [ha_mqtt_discoverable.sensors] Setting Energy (Lifetime) to 6320697 using hmd/sensor/JuiceBox/Energy--Lifetime-/state
2024-11-14 15:27:36  INFO      [ha_mqtt_discoverable.sensors] Setting Status to Unplugged using hmd/sensor/JuiceBox/Status/state
2024-11-14 15:27:36  INFO      [ha_mqtt_discoverable.sensors] Setting Temperature to 50.0 using hmd/sensor/JuiceBox/Temperature/state
2024-11-14 15:27:36  INFO      [ha_mqtt_discoverable.sensors] Setting Max Current(Online/Device) to 20 using hmd/sensor/JuiceBox/Max-Current-Online-Device-/state
2024-11-14 15:27:36  INFO      [ha_mqtt_discoverable.sensors] Setting Current Rating to 32 using hmd/sensor/JuiceBox/Current-Rating/state
2024-11-14 15:27:36  INFO      [ha_mqtt_discoverable.sensors] Setting Frequency to 49.98 using hmd/sensor/JuiceBox/Frequency/state
2024-11-14 15:27:36  INFO      [ha_mqtt_discoverable.sensors] Setting Power to 0 using hmd/sensor/JuiceBox/Power/state
2024-11-14 15:27:41  DEBUG     [juicebox_mqtthandler] From EnelX: b'CMD41527A20M20C242S575!CV1$'
2024-11-14 15:27:41  INFO      [ha_mqtt_discoverable.sensors] Setting Data from EnelX to CMD41527A20M20C242S575!CV1$ using hmd/sensor/JuiceBox/Data-from-EnelX/state
2024-11-14 15:28:05  INFO      [juicebox_udpcupdater] JuiceboxUDPCUpdater Check Starting
2024-11-14 15:28:06  DEBUG     [juicebox_udpcupdater] Active UDPC Stream: 3
2024-11-14 15:28:06  INFO      [juicebox_udpcupdater] UDPC IP incorrect, updating
2024-11-14 15:28:06  DEBUG     [juicebox_udpcupdater] Closing UDPC stream: 3
2024-11-14 15:28:06  INFO      [juicebox_udpcupdater] UDPC IP Changed
[... cut ...] xe6\xe4O\xc2\xb8Ez,\x82U\xe0\x82\xab\xbe\xf3\xd3}\xb9\x9d7\x1a\xeeCL\xfe\x1c\x91)\x1d7>\xf8\xbc\xf4\xdb\x99[\x12iz\xf7\xb22\xaff\xbb!\xc7l\xbc(4*\xe1c\x1b\x8d\xe1\x85\x9b\x8c>\xfe\xc8\xe8\x9d\xfd\xea\xd5\x93\x98\xe6\xe2D\x85F\x11\x19\xca\xd7\x07\x03(\xc5\x96\x06\xbf\xb4Y\xc0''
Traceback (most recent call last):
  File "/home/juice/juicepassproxy/juicebox_message.py", line 152, in juicebox_message_from_bytes
    string = data.decode("utf-8")
             ^^^^^^^^^^^^^^^^^^^^
```

any suggestions?

ivanfmartinez, 2024-11-16:
@BeneWilh apparently the directory server is allways returning encrypt : false when requesting by browser and giving other results when juicebox requests.

The local control should work for v07 or v09 protocols. On v07 YOU MUST WAIT some minutes after starting juicepassproxy because JPP needs to know what is the offline current and that value will be sent when the device does not have responses from server for 5 minutes. https://github.com/JuiceRescue/juicepassproxy/pull/69/files#diff-01e6d9ffed056a02cae8d8a0ec5d476a64d017bf85c0d5a94bb23ca21f33f5aaR231

If you enable DEBUG on JPP and restart the juicebox you can see the debug messages that came from juicebox and this will show when the directory server indicates that juicebox should use encryption, you will see
```
YOUR SERIAL:DBG,NFO:DirSvc encrypt 1:
```

If you have specific problems with local control you should give more information on an specific issue.

You HAVE do block the access from your device to the directory server to keep unencrypted protocol and JPP will work decoding and passing data to enel x, and with local control.

I have just commited some adjusts for this v08 message dont get generic errors on the detection of v08 encrypted messages.

At this moment when your device is using encrypted protocol we cant decode the messages.

BeneWilh, 2024-11-17:
@ivanfmartinez 
- i blocked all internet traffic for Juicebox in Router
- i pulled your latest branch
- restarted the Juicebox
- restarted JPP

```
juice@juicepassproxy:~/juicepassproxy$ git checkout juicebox_commands
Bereits auf 'juicebox_commands'
Ihr Branch ist auf demselben Stand wie 'origin/juicebox_commands'.
juice@juicepassproxy:~/juicepassproxy$ git pull
remote: Enumerating objects: 7, done.
remote: Counting objects: 100% (7/7), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 7 (delta 5), reused 7 (delta 5), pack-reused 0 (from 0)
Entpacke Objekte: 100% (7/7), 1.11 KiB | 190.00 KiB/s, fertig.
Von https://github.com/ivanfmartinez/juicepassproxy
   845c5a9..6f00f9c  juicebox_commands -> origin/juicebox_commands
Aktualisiere 845c5a9..6f00f9c
Fast-forward
 juicebox_message.py     | 13 ++++++++-----
 juicebox_mqtthandler.py | 12 +++++++++---
 test_message.py         |  9 +++++++++
 3 files changed, 26 insertions(+), 8 deletions(-)
juice@juicepassproxy:~/juicepassproxy$
```
resulting in:

[Logging_branch_jb_commands_log 2024-11-17 1409.txt](https://github.com/user-attachments/files/17790571/Logging_branch_jb_commands_log.2024-11-17.1409.txt)

suggestions?

ivanfmartinez, 2024-11-17:
@BeneWilh  the device using encryption indicates that they still have any type of internet access or a different firmware that have different behaviour enabling the encryption.

All information that we have until now indicates that encryption is defined by the directory server.

Your router have a way to capture packets ? a full capture can indicate if it was connecting to any external  server or not before enabling encryption. If you have a full capture and dont want to send here with serial you can send by email (my id at gmail) or  t.me/ivanfmartinez or discord .

For logging only on JPP I think the best way is doing redirect on router (not telnet) and restart JPP before juicebox. This way we can see the juicebox debug messages on JPP. 

Your log from 15:17:48 to 15:17:49 are not enough to confirm that control is not working, because the updated values from juicebox comes from 1 or 2 status messages later.

I have removed the old code that are still giving exceptions on encrypted messages and pushed to my branch.

BeneWilh, 2024-11-17:
@ivanfmartinez 
in the logs you see that my box came back to v07 without encryption, so values where available.

- I stopped JPP
- git pull... several lines deleted
- switched off Juicebox
- restartet JPP

```
2024-11-17 20:10:41  WARNING   [__main__] Starting JuicePass Proxy v0.5.0 (Log Level: DEBUG, log_handlers=2)
2024-11-17 20:10:41  INFO      [__main__] log_loc: LOC/juicepassproxy.log
2024-11-17 20:10:41  INFO      [juicebox_config] config_loc: /root/.juicepassproxy/juicepassproxy.yaml
2024-11-17 20:10:41  INFO      [juicebox_config] Reading config from /root/.juicepassproxy/juicepassproxy.yaml
2024-11-17 20:10:41  INFO      [__main__] telnet port: 2000
2024-11-17 20:10:41  INFO      [__main__] telnet timeout: 30
2024-11-17 20:10:41  INFO      [__main__] ignore_enelx: False
2024-11-17 20:10:44  ERROR     [aiorun] Unhandled exception; stopping loop: 'Task exception was never retrieved'
Traceback (most recent call last):
  File "/home/juice/juicepassproxy/.venv/lib/python3.11/site-packages/aiorun.py", line 219, in new_coro
    await coro
  File "/home/juice/juicepassproxy/juicepassproxy.py", line 425, in main
    enelx_server_port = await get_enelx_server_port(
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/juice/juicepassproxy/juicepassproxy.py", line 127, in get_enelx_server_port
    async with JuiceboxTelnet(
  File "/home/juice/juicepassproxy/juicebox_telnet.py", line 20, in __aenter__
    if await self.open():
       ^^^^^^^^^^^^^^^^^
  File "/home/juice/juicepassproxy/juicebox_telnet.py", line 60, in open
    self.reader, self.writer = await telnetlib3.open_connection(
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/juice/juicepassproxy/.venv/lib/python3.11/site-packages/telnetlib3/client.py", line 352, in open_connection
    transport, protocol = await asyncio.get_event_loop().create_connection(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 1085, in create_connection
    raise exceptions[0]
  File "/usr/lib/python3.11/asyncio/base_events.py", line 1069, in create_connection
    sock = await self._connect_sock(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 973, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 634, in sock_connect
    return await fut
           ^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 674, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
OSError: [Errno 113] Connect call failed ('192.168.178.27', 2000)
2024-11-17 20:10:44  INFO      [aiorun] Entering shutdown phase.
2024-11-17 20:10:44  INFO      [aiorun] Waiting for executor shutdown.
2024-11-17 20:10:44  INFO      [aiorun] Shutting down async generators
2024-11-17 20:10:44  INFO      [aiorun] Closing the loop.
2024-11-17 20:10:44  INFO      [aiorun] Leaving. Bye!
2024-11-17 20:10:44  INFO      [aiorun] Reraising unhandled exception
```
then i waited for the JP to come online and restarted JPP... values are readable, ntp error exists still... changing amps does not work (no effect). Will send you the full logging. Thank you!

BeneWilh, 2024-11-23:
It seems that even all UDP Ports for the JB are blocked the change to v08 (now v08L) is happening.

`TODO: encrypted 'v08' - 'b'<<JB ID>>:v08L`

ivanfmartinez, 2024-11-23:
@BeneWilh 
Only  with a full capture of the wifi packets we can confirm who is responsible for the change.

the L  is the first byte after version , and not apper to be related to the version itself as other logs indicate different value.

As far I have seen just "v08" already indicate encrypted, and for v09 they put extra character to indicate.

BeneWilh, 2024-11-23:
> As far I have seen just "v08" already indicate encrypted, and for v09 they put extra character to indicate.

if you recall my other loggs, the "L" wasnt there, thats new. I blocked all UDP Ports on my Fritzbox and eventhoug the JB changed to v08L =>> From JuiceBox: b'<<JP ID>>:**v08L**\xa4\t\x00\x00\x0....

carrel-gr, 2024-11-23:
@BeneWilh have you tried blocking ALL ports, not just UDP ports?  I know that the command messages (ports 8047 & 8042) are UDP, but that comes after the directory service lookup.  I think the initial lookup is TCP.  If I look at my logs, I see my router blocking https traffic (TCP 443) from my JB to some internet host.  I suggest blocking EVERYTHING from the JB to the internet.  You then need to allow UDP 8042 (and maybe 8047) from the JB to JPP, and you have to allow some sort of DNS lookup by the JB, and the details of that depend on which JPP method you are using.

BeneWilh, 2024-11-23:
@carrel-gr on blocking all internet traffic for JB i had some failures on DNS for JB in the logging, so i though blocking al UDP Ports should be enough to prevent canging to encrypted... but I was wrong. Now i blocked all traffic for JB to the internet. JPP has full access... but I'm struggeling with commands & Enel X App... (using Ivan's latest brach as a manual installation)

carrel-gr, 2024-11-23:
> but I'm struggeling with commands & Enel X App...

Can you describe your struggle?  This isn't enough information.

Do you need to continue using the Enel X app?  I set the IGNORE_ENELX option and removed the app fro my phone.  Only HA now controls my JB.  I didn't want two masters, as that's bound to cause problems.  And I trust that HA will still be around tomorrow.  Can't say as much for enel x.

BeneWilh, 2024-11-23:
@carrel-gr  using [https://github.com/ivanfmartinez/juicepassproxy/tree/juicebox_commands]

I'm starting JPP with 

`python3 /home/juice/juicepassproxy/juicepassproxy.py --juicebox_host 192.168.178.27 --mqtt_host 192.168.178.18 --update_udpc --jpp_host 192.168.178.17 --experimental --debug --log_loc LOC --enelx_ip 158.47.3.128:9004 --ignore_enelx`

(also tried)

`python3 /home/juice/juicepassproxy/juicepassproxy.py --juicebox_host 192.168.178.27 --mqtt_host 192.168.178.18 --update_udpc --jpp_host 192.168.178.17 --experimental --debug --log_loc LOC --ignore_enelx`

I see all data in MQTT Browser and life values in HA & EnelX App including online / offline wanted (after 6 min). Changing the Amps in the app or HA is not changing anything in JB

[image]

so, you see: wanted 13 A => actual state 20A

```
2024-11-23 21:31:00  INFO      [juicebox_mqtthandler] Number Callback (Max Current(Online/Wanted)): 10. User Data: None
2024-11-23 21:31:00  INFO      [ha_mqtt_discoverable.sensors] Setting Max Current(Online/Wanted) to 10.0 using hmd/number/JuiceBox/Max-Current-Online-Wanted-/state
2024-11-23 21:31:00  ERROR     [juicebox_mitm] Must have both current_max(online|offline) defined to send command message
2024-11-23 21:31:00  INFO      [ha_mqtt_discoverable.sensors] Setting Max Current(Online/Wanted) to 10.0 using hmd/number/JuiceBox/Max-Current-Online-Wanted-/state
2024-11-23 21:31:00  INFO      [juicebox_mqtthandler] Number Callback (Max Current(Offline/Wanted)): 10. User Data: None
2024-11-23 21:31:00  INFO      [ha_mqtt_discoverable.sensors] Setting Max Current(Offline/Wanted) to 10.0 using hmd/number/JuiceBox/Max-Current-Offline-Wanted-/state
2024-11-23 21:31:00  INFO      [juicebox_mitm] command message = CMD62131A10M10C006S001!QC8$ new_values=True new_version=False
2024-11-23 21:31:00  INFO      [juicebox_mitm] Sending command to juicebox CMD62131A10M10C006S001!QC8$ new_values=True
2024-11-23 21:31:00  INFO      [juicebox_message] from_string 0817011001010463110000000000:v07,s0001,u39748,V2227,L0006470272,S2,T17,M20,m32,t09,i42,e-001,f5001,X0,Y0,E016377,A0201,p0996!PXF:
2024-11-23 21:31:00  DEBUG     [juicebox_mqtthandler] From JuiceBox: b'0817011001010463110000000000:v07,s0001,u39748,V2227,L0006470272,S2,T17,M20,m32,t09,i42,e-001,f5001,X0,Y0,E016377,A0201,p0996!PXF:' decoded=0817011001010463110000000000:v07,s0001,u39748,V2227,L0006470272,S2,T17,M20,m32,t09,i42,e-001,f5001,X0,Y0,E016377,A0201,p0996!PXF:
2024-11-23 21:31:00  DEBUG     [juicebox_mqtthandler] decode/parsed message = {'type': 'basic', 'current': 20.1, 'energy_session': 16377, 'serial': '0817011001010463110000000000', 'protocol_version': '07', 'counter': '0001', 'loop_counter': '39748', 'voltage': 222.7, 'energy_lifetime': 6470272, 'status': 'Charging', 'temperature': 62.6, 'current_max_online': 20, 'current_rating': 32, 'report_time': '09', 'interval': 42, 'e': '-001', 'frequency': 50.01, 'X': '0', 'Y': '0', 'power_factor': 0.996, 'power': 4476}
2024-11-23 21:31:00  DEBUG     [juicebox_mqtthandler] Publish Basic Message: {'type': 'basic', 'current': 20.1, 'energy_session': 16377, 'serial': '0817011001010463110000000000', 'protocol_version': '07', 'counter': '0001', 'loop_counter': '39748', 'voltage': 222.7, 'energy_lifetime': 6470272, 'status': 'Charging', 'temperature': 62.6, 'current_max_online': 20, 'current_rating': 32, 'report_time': '09', 'interval': 42, 'e': '-001', 'frequency': 50.01, 'X': '0', 'Y': '0', 'power_factor': 0.996, 'power': 4476}
2024-11-23 21:31:00  INFO      [ha_mqtt_discoverable.sensors] Setting Current to 20.1 using hmd/sensor/JuiceBox/Current/state
```

and charging amps are not changed even waiting for an other couple of messages... seems my JB is not accepting the change
`

carrel-gr, 2024-11-23:
Why set an enelx ip when using ignore_enelx?  If your enel x app is working, then enelx is also 'controlling' your JuiceBox.  I don't have any idea how to help in that situation.  I suggest you first get  things to a state where the enel x app says that your  JB is offline and only HA is controlling it.   Also do you need update_udpc?  Which "option" (A, B, or C) are you using?  Also I see your system is v07 and my experience here has been with v09.  You may need wiser people than I for v07 help if that is related to your issue.

BeneWilh, 2024-11-24:
@carrel-gr I tested already several options... with / without enelx, with IP and without. The problem is always the same... Homeassistant commands are not accepted by the JB. It migth be a issue with v07... if somebody has an idea what to test, happy to get help.
I am using option B withou Docker.

snicker, 2025-01-20:
@BeneWilh  are you still having issues? I'd recommend trying v0.5.0 and reporting back.

BeneWilh, 2025-01-26:
> [@BeneWilh](https://github.com/BeneWilh) are you still having issues? I'd recommend trying v0.5.0 and reporting back.

no change... I can read the values but no adjustments possible, seem they get ignored by my box

BeneWilh, 2025-01-26:
> > [@BeneWilh](https://github.com/BeneWilh) are you still having issues? I'd recommend trying v0.5.0 and reporting back.
> 
> no change... I can read the values but no adjustments possible, seem they get ignored by my box

@snicker Logging with V0.5 from today:

```
2025-01-26 17:14:46  INFO      [juicebox_mitm] command message = CMD01714A09M20C006S007!7MY$ new_values=False new_version=False
2025-01-26 17:14:46  INFO      [juicebox_mitm] Sending command to juicebox CMD01714A09M20C006S007!7MY$ new_values=False
2025-01-26 17:14:55  INFO      [juicebox_mqtthandler] Number Callback (Max Current(Online/Wanted)): 14. User Data: None
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Max Current(Online/Wanted) to 14.0 using hmd/number/JuiceBox/Max-Current-Online-Wanted-/state
2025-01-26 17:14:55  INFO      [juicebox_mitm] command message = CMD01714A14M20C006S008!SI8$ new_values=True new_version=False
2025-01-26 17:14:55  INFO      [juicebox_mitm] Sending command to juicebox CMD01714A14M20C006S008!SI8$ new_values=True
2025-01-26 17:14:55  INFO      [juicebox_message] from_string 08170XXXXXXXXXXXXXXXXXXXXX:v07,s0001,u46384,V2247,L0007719441,S2,T17,M20,m32,t09,i69,e-001,f4999,X0,Y0,E015208,A0200,p0996!IQG:
2025-01-26 17:14:55  DEBUG     [juicebox_mqtthandler] From JuiceBox: b'08170XXXXXXXXXXXXXXXXXXXXX:v07,s0001,u46384,V2247,L0007719441,S2,T17,M20,m32,t09,i69,e-001,f4999,X0,Y0,E015208,A0200,p0996!IQG:' decoded=08170XXXXXXXXXXXXXXXXXXXXX:v07,s0001,u46384,V2247,L0007719441,S2,T17,M20,m32,t09,i69,e-001,f4999,X0,Y0,E015208,A0200,p0996!IQG:
2025-01-26 17:14:55  DEBUG     [juicebox_mqtthandler] decode/parsed message = {'type': 'basic', 'current': 20.0, 'energy_session': 15208, 'serial': '08170XXXXXXXXXXXXXXXXXXXXX', 'protocol_version': '07', 'counter': '0001', 'loop_counter': '46384', 'voltage': 224.7, 'energy_lifetime': 7719441, 'status': 'Charging', 'temperature': 62.6, 'current_max_online': 20, 'current_rating': 32, 'report_time': '09', 'interval': 69, 'e': '-001', 'frequency': 49.99, 'X': '0', 'Y': '0', 'power_factor': 0.996, 'power': 4494}
2025-01-26 17:14:55  DEBUG     [juicebox_mqtthandler] Publish Basic Message: {'type': 'basic', 'current': 20.0, 'energy_session': 15208, 'serial': '08170XXXXXXXXXXXXXXXXXXXXX', 'protocol_version': '07', 'counter': '0001', 'loop_counter': '46384', 'voltage': 224.7, 'energy_lifetime': 7719441, 'status': 'Charging', 'temperature': 62.6, 'current_max_online': 20, 'current_rating': 32, 'report_time': '09', 'interval': 69, 'e': '-001', 'frequency': 49.99, 'X': '0', 'Y': '0', 'power_factor': 0.996, 'power': 4494}
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Current to 20.0 using hmd/sensor/JuiceBox/Current/state
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Energy (Session) to 15208 using hmd/sensor/JuiceBox/Energy--Session-/state
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Voltage to 224.7 using hmd/sensor/JuiceBox/Voltage/state
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Energy (Lifetime) to 7719441 using hmd/sensor/JuiceBox/Energy--Lifetime-/state
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Status to Charging using hmd/sensor/JuiceBox/Status/state
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Temperature to 62.6 using hmd/sensor/JuiceBox/Temperature/state
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Max Current(Online/Device) to 20 using hmd/sensor/JuiceBox/Max-Current-Online-Device-/state
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Current Rating to 32 using hmd/sensor/JuiceBox/Current-Rating/state
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Frequency to 49.99 using hmd/sensor/JuiceBox/Frequency/state
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Power Factor to 0.996 using hmd/sensor/JuiceBox/Power-Factor/state
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Power to 4494 using hmd/sensor/JuiceBox/Power/state
2025-01-26 17:14:55  INFO      [juicebox_mitm] command message = CMD01714A14M20C006S009!RI8$ new_values=False new_version=False
2025-01-26 17:14:55  INFO      [juicebox_mitm] Sending command to juicebox CMD01714A14M20C006S009!RI8$ new_values=False
2025-01-26 17:14:55  INFO      [ha_mqtt_discoverable.sensors] Setting Max Current(Online/Wanted) to 14.0 using hmd/number/JuiceBox/Max-Current-Online-Wanted-/state
2025-01-26 17:14:56  INFO      [juicebox_udpcupdater] JuiceboxUDPCUpdater Check Starting
2025-01-26 17:14:56  DEBUG     [juicebox_udpcupdater] Active UDPC Stream: 2
2025-01-26 17:14:56  INFO      [juicebox_udpcupdater] UDPC IP correct
2025-01-26 17:15:04  INFO      [juicebox_message] from_string 08170XXXXXXXXXXXXXXXXXXXXX:v07,s0001,u47284,V2247,L0007719453,S2,T17,M20,m32,t09,i70,e-001,f4998,X0,Y0,E015220,A0202,p0996!H93:
2025-01-26 17:15:04  DEBUG     [juicebox_mqtthandler] From JuiceBox: b'08170XXXXXXXXXXXXXXXXXXXXX:v07,s0001,u47284,V2247,L0007719453,S2,T17,M20,m32,t09,i70,e-001,f4998,X0,Y0,E015220,A0202,p0996!H93:' decoded=08170XXXXXXXXXXXXXXXXXXXXX:v07,s0001,u47284,V2247,L0007719453,S2,T17,M20,m32,t09,i70,e-001,f4998,X0,Y0,E015220,A0202,p0996!H93:
2025-01-26 17:15:04  DEBUG     [juicebox_mqtthandler] decode/parsed message = {'type': 'basic', 'current': 20.2, 'energy_session': 15220, 'serial': '08170XXXXXXXXXXXXXXXXXXXXX', 'protocol_version': '07', 'counter': '0001', 'loop_co [... cut ...]
2025-01-26 17:16:07  INFO      [ha_mqtt_discoverable.sensors] Setting Power Factor to 0.996 using hmd/sensor/JuiceBox/Power-Factor/state
2025-01-26 17:16:07  INFO      [ha_mqtt_discoverable.sensors] Setting Power to 4478 using hmd/sensor/JuiceBox/Power/state
2025-01-26 17:16:07  INFO      [juicebox_mitm] command message = CMD01716A14M20C006S015!LTV$ new_values=False new_version=False
2025-01-26 17:16:07  INFO      [juicebox_mitm] Sending command to juicebox CMD01716A14M20C006S015!LTV$ new_values=False
```