JuiceRescue/juicepassproxy #119: Minumum reported temperature?

metheos, 2024-11-27:
I blocked internet access on my juicebox 40 and it seems to be working great with juicepassproxy again.
I did notice that since the weather cooled off the reported temperature is stuck at 60.8 F
Is this normal? I'm guess the thing is only worried about getting too hot, but 16C seems a bit arbitrary 😅

carrel-gr, 2024-11-30:
FWIW, my temperature is changing throughout the day, tracking the ambient outdoors.  It also rises above ambient when charging.  So JPP is handling temp if the JB sends it.  Are you on an older version 1 JB or a newer version 2?  Any logs showing what your JB is sending to JPP?  Since you are seeing a valid number, but it never changes, it may be that the temperature sensor has died or is stuck.  "Have you tried turning it off and on again?"

metheos, 2024-11-30:
I'm not sure what version it is 🤔
Jpp is working fine reading and reporting data. The temperature also works fine above 60.8 F
It just won't go lower than that 😅

snicker, 2025-01-20:
Closing this issue as it appears to be isolated and may be due to hardware failure with an individual device.

closed 2025-01-20 by snicker