JuiceRescue/juicepassproxy #120: [__main__] A JuicePass Proxy task failed: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc9 in position 33:

BstNst99, 2024-12-16:
decode error because data from xnel server is not utf-8 format.

BstNst99, 2024-12-16:
I already fix by change all encode/decode in "juicebox_mqtthandler.py" from "utf-8" to "latin1" and fix error when log message by use str(message)

    async def _basic_message_publish(self, message):
        _LOGGER.debug(f"Publish {message.get('type').title()} Message: {str(message)}")

jjhuff, 2024-12-21:
I have an Gen1 and have been getting this: `v08n`
```
From JuiceBox: b'0817081002000455176119246503:v08n\xeb"\x00\x00\x00\x00\xa9\xcc~F\x05\xeb\xd0\xb3\x8f7\xf6\xe0\x03\x1d+\x05h"\xd2\x84e.\xdd>(\x80f\x1e\xf7\xect\xf8D\xa2\xc1l\xa5|\x8d{k2~W9\xdcc!\xe9\x00/\xcc\xf8X\xae\x04\xa7\xa0\xcd,\xbec\x9f\xcb\x85/t\x82\xdah?l\xbb\x8f^\xd7@J\xde\x81\xcb\x9c\xba\xa5]_\xe0w\xbfh|\x95\xb5\xea\xb8v\x96\x826\xcau\x13-\xb6\xe9\x06D\x84 1J\xcb\x8es\x12\x0e\x8f"{\xb8\xb5'
```

If I block the unit from the internet, it seem get stuck in a reboot loop

ivanfmartinez, 2024-12-22:
@BstNst99 the devices normally starts using encrypted mode when they have access to the Directory Server and the encryption keys, the sollution is to block the internet access and use only juicepassproxy.

@jjhuff  As far I have seen v08 means encrypted, and in your case the n is the first encrypted byte.

If you point the device to juicepassproxy using the router redirect you can see the debug boot logs on JPP.

The version in my branch can detect these encrypted messages, but for now we dont know how to decode them : https://github.com/JuiceRescue/juicepassproxy/pull/69#issue-2329314055

snicker, 2025-01-20:
v0.5.0 now contains the code to detect and log encrypted messages, but the current directive is to block internet access to the Juicebox.

Closing for now.

closed 2025-01-20 by snicker