JuiceRescue/juicepassproxy #123: Permissions Error when pulling image

daviddever, 2025-02-11:
Attempting to pull the image from `ghcr.io/juicerescue/juicepassproxy:latest` results in a permissions error, `Error Head "https://ghcr.io/v2/juicerescue/juicepassproxy/manifests/latest": denied` 

I was able to pull the image from `ghcr.io/snicker/juicepassproxy:latest`

I am guessing this is likely related to the build issue [122](https://github.com/JuiceRescue/juicepassproxy/issues/122) but wanted to bring it up separate just in case.

Thanks!

gimse, 2025-02-22:
Can somebody fix this?

snicker, 2025-03-08:
@gimse @daviddever please try again - seems that the permissions on the package repo were misconfigured when we moved to the new org.

snicker, 2025-03-08:
Confirmed `docker pull ghcr.io/juicerescue/juicepassproxy:edge` is working when unauthenticated with a user on discord - please re-open or create a new issue if still failing

closed 2025-03-08 by snicker