JuiceRescue/juicepassproxy #129: Auto-start charging without using the app

jcmunozfe, 2025-07-21:
The charger supports two alternatives: start the charge upon connection or start the charge from the app. At this point, the charger is configured with the second option. There is no exposed switch to change the configuration to start the charge upon connection.

So, at this point, it is possible to see the sensors and modify the Amps, but it is not possible to start a charging session.

Additionally, knowing the correct command to send using the experimental Send Command functionality to change this setting will be great.

Thanks for your work

ivanfmartinez, 2025-07-21:
As far I know the physical charger does not have this option. The option is on server side. 

If you set the wanted (online) current to ZERO it will not start charging until you set the value to a higher than 6 value.

Using this you can control the charger to start only when you want. But be aware that some cars dont like this and you have to do a more complex approach.

For my Bolt I leave the Default wanted current above 13, and after the car starts charging I pause (set to ZERO). If I leave it with value lower than 13 it consider as a wall charging and will allow only a maximum of 10A when charging starts.

jcmunozfe, 2025-07-21:
I made some additional failed tests, including waiting a long time for the offline mode. When I stop de JPP and start it again, the Max Current Offline and Online are restored to zero (even if the charger is not restarted).

Hopefully, someone who has the app working can provide the package dump when the configuration is changed from automatic start to charge controlled by the app. I can not, at this moment at least, since the app is not working for me.