JuiceRescue/juicepassproxy #131: "Voltie" takeover documentation refresh

jtolio, 2025-08-30:
Hello! I am, just now, finally trying to connect my Juicebox to the internet. I have Home Assistant running (for the first time) with mosquitto and juicepassproxy. Home assisstant shows statistics from the Juicebox. It took me a slightly longer time figuring out how to set this up because my Juicebox appears to have been upgraded to "voltie", e.g.:

```
EMWERK-JB201-1.0.46, Gecko_OS-STANDARD-4.2.7-11064, WGM160P
> list
list
! # Type  Info
# 1 FILE  godaddy_ca.pem-1.0.0 (1390, 0)
# 2 UDPC  dummy.voltie.us:8055 (8270)
```

What is JuiceRescue's position on Voltie? I couldn't find one.

I couldn't get the Voltie app to work, but I also can't get Home Assistant to change the charging state of my Juicebox either. Any Voltie-Juicebox specific tips or tricks on how to make HA control the charging state?

TheJoe1990, 2025-11-06:
> Oh, are you able to set the power level and have the setting sticky?
> […](#)

[image]

So, I got this connected less than 12 hours ago and I have never personally interacted with an electric car charger till this week.  When I move these sliders, the numbers on the right do update, but the numbers above them (Which I assume is coming from the charger?) do not seem to update.  Looks like maybe not then? Does that answer your question?

ivanfmartinez, 2025-11-07:
> I can confirm that the charger is not receiving instructions

I understand :
"Not receiving" as JPP does not send message to Juicebox, this will happen if IGNORE_ENELX is not enabled
"Not accepting" as Juicebox does not change behaviour based on what JPP sends

Best way to check is capturing packets to/from juicebox on the wifi access point or router where juicebox is connected.

The current behaviour is based on reverse enginner that some users made and I put the logic on JPP.

I remember that one user have the case where Juicebox apparently does not accept the messages, but we dont have more information about this to confirm if they changed something on protocol.

If Juicebox does not accept the messages we will have to get working messages from server that juicebox accept to compare with messages from JPP.

ivanfmartinez, 2025-11-12:
@apurohit8 I remember that if you redirect the traffic to JPP you can see on the container logs some log messages that came from device. Maybe your device is with an error state that we don´t know how to decode.

Considering that messages sent to device are not respected probably they have changed something in firmware, and without someone trying to capture packets from voltie and checking differences from our code nothing can be done.

There are reports of relay problems with juicebox, maybe something is wrong with your relay also.

will be nice to have an spare charger to test, but as I'm in Brazil the cost for sending and  importing taxes  can be greater than a new charger and I even dont know if the voltie servers will respond for  other countries. Its better to try to check with someone in USA that have time and can do the tests.

apurohit8, 2025-11-12:
@ivanfmartinez 

Attached is a copy of my current log file from the JB. Note that my EV isn't present (wife has it) so it's just sitting idle for now. I don't know if any of the info in the log file is useful, but I figured I'd offer it anyway.

[_juicepassproxy_logs.txt](https://github.com/user-attachments/files/23505772/_juicepassproxy_logs.txt)

I am more familiar with packet capture. When the EV is back, I can certainly pull some of the same logs when it's plugged in. I can also use Wireshark to capture packets if needed. I'll see if I can get that info for you soon.

As for the existing JB, I agree, shipping it to you in Brazil may not be feasible... too bad, because it would have been great to get it to you! But if you change your mind, let me know... we may be able to figure something out, if it's helpful! In the meantime, I think we can work on the packet capture if that gets you what you need. Let me know what I can do to help, or if you need something different than what I am planning to do.

apurohit8, 2025-11-14:
Unfortunately, I can't get the JB to connect to my laptop wifi - keeps saying the password is wrong (it sees it, but won't accept the connection password). While connected to "setup.com" on the JB, I can see it still shows the Enel logo on there, so I'm guessing my JB did not get updated to Voltie (I hear that they have a different icon?)

If this is true, then collecting data from my JB probably won't help this project, I don't think?

That being said, I do notice there is a "console" option in there. Is there a way I can read/write settings there while keeping the JB offline? Specifically, I want to see if the OFFLINE max current setting is set to 40A, and then I can retry plugging into the car. Are there commands I can use in the console here?

SpikedBlue, 2025-11-18:
I also seem to have the same issue.
- JPP tries to set the current, but always reads back 40.
- Plane old juicebox web interface left side has "connect", and "network". No icons of any kind other then network connection bars. (Running older black juicebox)
- I'm sure when I first got JPP working I had full control and the blue led (network) on the box was solid blue. Now it's slow blinging constantly. Not sure why.
- Manual Telnet gives the following:
-! # Type  Info
-# 0 FILE  webapp/index.html-1.4.0.24 (1995, 0)
-# 1 UDPC  jbv1.emotorwerks.com:8042 (855)

Did you still need a capture from the juicebox? I could try to get a capture.
Or have any suggestions to try?

nussdk, 2025-12-02:
i just installed this proxy today and catching up. I have the post voltie update on mine and seeing same issue where the home assistant values dont save on the device, reading works fine but cant seem to write. 

i dont think the new verison is encrypted. i can read some messages coming into the juicebox
running this version
`EMWERK-JB201-1.0.46, Gecko_OS-STANDARD-4.2.7-11064, WGM160P`

example packet coming in:

10517	393.821866	104.250.239.118	192.168.20.194	UDP	72	8055 → 12914 Len=30
and the packate bytes has this in it which looks like what JPP is sending???
```CMD21607A0000M030C006S125```

im not sure how to connect the charger to the real app as its disconinued (USA) to send commands from there? anyone know how to connect this, then i can capture more packets from the real server